Koenkk/zigbee2mqtt #13270: Xiaomi LLKZMK11LM OTA firmware update failed 'invalid image'

Chao-chao00, 2022-07-24:
### What happened?

Hi, i successfully updated the firmware of 3 of my Xiaomi LLKZMK11LM relays from 02-27-2019 to 9-22-2021 but for the 4th one the OTA firmware updates goes to 100% and then it says "Update failed with reason: 'invalid image'" so it stays at version from 02-27-2019.

I then tried using the Edge version of the Zigbee2MQTT Home Assistant addon but i didn't help and the same error occurred. I also tried using the Xiaoni Gateway 3 with AlexxIT Home Assistant Integration to do the OTA from his integration but the update goes to 100% and when i check the firmware version in Zigbee2MQTT it is still from 02-27-2019.

### What did you expect to happen?

I expected the 4th Xiaomi LLKZMK11LM relay to update to firmware from 9-22-2021 like the others did.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9175848/log.txt)

Romario1972, 2022-08-08:
I also have absolutely the same issue but fails the second one.

```
> Info 2022-08-08 17:35:27MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'main_room_chandelier' failed (Update failed with reason: 'invalid image')","meta":{"device":"main_room_chandelier","status":"update_failed"},"type":"ota_update"}'
Info 2022-08-08 17:35:27MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"main_room_chandelier"},"error":"Update of 'main_room_chandelier' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"gbjf3-1"}'
```

Chao-chao00, 2022-08-11:
I just want to add that the Xiaomi LLKZMK11LM relay that fails the firmware update was bought on Aug 19, 2020 and the 3 remaining relays that successfully update the firmware were bought on Apr 17, 2021,

I don't know if it is a coincidence that the older of the four relays fails the firmware update. Maybe they are from different hardware revision ?

closed 2022-09-18 by github-actions[bot]

NoAdO, 2023-01-02:
Same problem