Koenkk/zigbee2mqtt #13274: [New device support]:

raymonvdm, 2022-07-24:
### Link

https://www.ikea.com/nl/en/p/styrbar-remote-control-white-30488363/

### Database entry

{"id":34,"type":"Unknown","ieeeAddr":"0x84ba20fffe728e77","nwkAddr":65053,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1658684491996,"defaultSendRequestWhen":"immediate"}

### Comments

I think there is a new version of the Ikea Styrbar button

Koenkk, 2022-07-24:
The device has not been paired correctly, try repairing close to the coordinator (`"interviewCompleted":false`)

MattWestb, 2022-07-25:
`W2149` is standard marking for electric and is produces year 2021 and week 49.

raymonvdm, 2022-07-25:
The new Sonoff Zigbee 3 stick put me on the wrong feet, pairing it close to the stick solved the issue, but is also needed to repair a lot of other devices after fiddling around

closed 2022-07-26 by Koenkk