Koenkk/zigbee2mqtt #13275: [New device support]: Develco IOMZB-110

BenDavidson90, 2022-07-24:
### Link

https://www.develcoproducts.com/products/smart-relays/io-module/

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0x0015bc004200056a","nwkAddr":1663,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Unknown","modelId":"IOMZB-110","epList":[1,112,113,114,115,116,117,242],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"112":{"profId":260,"epId":112,"devId":12,"inClusterList":[0,3,15],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"IOMZB-110","manufacturerName":"Develco Products A/S","powerSource":128,"zclVersion":3,"dateCode":"2021-08-21 02:49"}}},"binds":[],"configuredReportings":[],"meta":{}},"113":{"profId":260,"epId":113,"devId":12,"inClusterList":[0,3,15],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"114":{"profId":260,"epId":114,"devId":12,"inClusterList":[0,3,15],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"115":{"profId":260,"epId":115,"devId":12,"inClusterList":[0,3,15],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"116":{"profId":260,"epId":116,"devId":2,"inClusterList":[0,3,4,5,6],"outClusterList":[3,15],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"117":{"profId":260,"epId":117,"devId":2,"inClusterList":[0,3,4,5,6],"outClusterList":[3,15],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2021-08-21 02:49","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1658686310145,"defaultSendRequestWhen":"immediate"}

### Comments

Device not supported

BenDavidson90, 2022-08-08:
Hi,

Since there is no update, here is some information I can provide:

# Description
I/O module 4 dry contact inputs + 2 NO/NC outputs

# Endpoints
- **112:** Input
- **113:** Input
- **114:** Input
- **115:** Input
- **116:** Output
- **117:** Output
- **242:** ?

# Other
I don't know if can it helps (I doubt...) but the device works on Jeedom with their Zigbee plugin based on Zigpy, here is some informations I was able to get from it
| **Name** | **Type** | **Generic type** | **Logical Id**   |
|----------|----------|------------------|------------------|
| Status 1 | Info     | ENERGY_STATE     | 116::6::0        |
| On 1     | Action   | ENERGY_ON        | 116::on_off::on  |
| Off 1    | Action   | ENERGY_OFF       | 116::on_off::off |
| Status 2 | Info     | ENERGY_STATE     | 117::6::0        |
| On 2     | Action   | ENERGY_ON        | 117::on_off::on  |
| Off 2    | Action   | ENERGY_OFF       | 117::on_off::off |
| Input 1  | Info     | ENERGY_STATE     | 112::15::85      |
| Input 2  | Info     | ENERGY_STATE     | 113::15::85      |
| Input 3  | Info     | ENERGY_STATE     | 114::15::85      |
| Input 4  | Info     | ENERGY_STATE     | 115::15::85      |

I use this module to know the status of my gate and without this module, many of my home automation scenarios are in trouble...

I tried to create an external converter but without success.

I would be very grateful for your help.

BenDavidson90, 2022-09-08:
Up to prevent GitHub from closing this issue.
Nobody can help me ? I tested several things but couldn't get it to work...

svhelge, 2022-09-20:
Deconz has added support for IOMZB-110 - see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5258 and the "driver" https://github.com/dresden-elektronik/deconz-rest-plugin/blob/1666a9e9a6b7b61e64b496cb29f09ef8ab6ab5bc/devices/develco/iomzb-110_switch_module.json

BenDavidson90, 2022-09-22:
Someone from the Jeedom community (MrGreen, thanks to him) is helping me and made a converter that makes the two outputs works:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['IOMZB-110'],
        model: 'IOMZB-110',
        vendor: 'Develco Products A/S',
        description: '2 Gang Relay',
        exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
        extend: extend.switch(),
        endpoint: (device) => {
            return {'l1': 116, 'l2': 117};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep1 = device.getEndpoint(1);
            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(ep2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep1);
            await reporting.onOff(ep2);
        },
};

module.exports = definition;
```

It's a very good start but it lacks the most important for me: the inputs. I hope this converter should be able to help a lot in figuring out how to make them work...

svhelge, 2022-10-10:
I have added support for IOMZB-110, see https://github.com/Koenkk/zigbee-herdsman-converters/pull/4736

BenDavidson90, 2022-10-10:
It works, thank you very much ! 😃

closed 2022-10-10 by BenDavidson90