Koenkk/zigbee2mqtt #13278: Lonsonho TS130F_dual - Moving Attribute wrong

roy81f, 2022-07-25:
### What happened?

"Moving" attribute triggered both on Left and Right states.

For example:
Curtain1 is Right,  Curtain2 is left,   when i press the Curtain1 (right) Button Up im getting on both Right and Left states moving up but only curtain1 is really moving as should.

### What did you expect to happen?

Moving attribute will change on the correct side only. (Right in this case)

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

ZZH

roy81f, 2022-07-25:
[image]

Koenkk, 2022-07-28:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    tuya_cover_options: {
        cluster: 'closuresWindowCovering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('tuyaMovingState')) {
                const value = msg.data['tuyaMovingState'];
                const movingLookup = {0: 'UP', 1: 'STOP', 2: 'DOWN'};
                result[postfixWithEndpointName('moving', msg, model, meta)] = movingLookup[value];
            }
            if (msg.data.hasOwnProperty('tuyaCalibration')) {
                const value = msg.data['tuyaCalibration'];
                const calibrationLookup = {0: 'ON', 1: 'OFF'};
                result[postfixWithEndpointName('calibration', msg, model, meta)] = calibrationLookup[value];

            }
            if (msg.data.hasOwnProperty('tuyaMotorReversal')) {
                const value = msg.data['tuyaMotorReversal'];
                const reversalLookup = {0: 'OFF', 1: 'ON'};
                result[postfixWithEndpointName('motor_reversal', msg, model, meta)] = reversalLookup[value];
            }
            if (msg.data.hasOwnProperty('moesCalibrationTime')) {
                const value = parseFloat(msg.data['moesCalibrationTime']) / 10.0;
                result[postfixWithEndpointName('calibration_time', msg, model, meta)] = value;
            }
            return result;
        },
    },
};

const definition =  {
    fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_j1xl73iw'}, {modelID: 'TS130F', manufacturerName: '_TZ3000_kmsbwdol'}],
    model: 'TS130F_dual',
    vendor: 'Lonsonho',
    description: 'Dual curtain/blind module',
    fromZigbee: [fz.cover_position_tilt, fzLocal.tuya_cover_options],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    exposes: [
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('left'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('right'),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time'),
        e.cover_position().withEndpoint('left'), exposes.binary('calibration', ea.ALL, 'ON', 'OFF')
            .withEndpoint('left'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF').withEndpoint('left'),
        e.cover_position().withEndpoint('right'), exposes.binary('calibration', ea.ALL, 'ON', 'OFF')
            .withEndpoint('right'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF').withEndpoint('right'),
    ],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- delete `data/state.json`
- start z2m, check if issue is fixed

roy81f, 2022-07-28:
hi Koen,
Thank you so much for quick response,  i had follow the instructions but still, it seems both sides (buttons) shared the same "Moving" attribute value.  

Best regards,
Roy

Koenkk, 2022-07-29:
Could you provide me the debug log when the device should report the `moving` attribute?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

roy81f, 2022-07-29:
hi, i pressed Left Up, Down and than Right Up and Down.

 2022-07-29 18:14:10Received Zigbee message from 'Master Bedroom Curtain and Roller Shutter', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"tuyaMovingState":0}' from endpoint 1 with groupID 0
Info 2022-07-29 18:14:10MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain and Roller Shutter', payload '{"calibration":"OFF","calibration_left":"OFF","calibration_right":"OFF","calibration_time":27.8,"linkquality":94,"motor_reversal":"OFF","motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving":"UP","position_left":100,"position_right":56}'
Debug 2022-07-29 18:14:11Received Zigbee message from 'Master Bedroom Curtain and Roller Shutter', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":4,"tuyaMovingState":1}' from endpoint 1 with groupID 0
Info 2022-07-29 18:14:11MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain and Roller Shutter', payload '{"calibration":"OFF","calibration_left":"OFF","calibration_right":"OFF","calibration_time":27.8,"linkquality":91,"motor_reversal":"OFF","motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving":"STOP","position_left":96,"position_right":56}'
Debug 2022-07-29 18:14:12Received Zigbee message from 'Master Bedroom Curtain and Roller Shutter', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":4,"tuyaMovingState":2}' from endpoint 1 with groupID 0
Info 2022-07-29 18:14:12MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain and Roller Shutter', payload '{"calibration":"OFF","calibration_left":"OFF","calibration_right":"OFF","calibration_time":27.8,"linkquality":94,"motor_reversal":"OFF","motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving":"DOWN","position_left":96,"position_right":56}'
Debug 2022-07-29 18:14:13Received Zigbee message from 'Master Bedroom Curtain and Roller Shutter', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"tuyaMovingState":1}' from endpoint 1 with groupID 0
Info 2022-07-29 18:14:13MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain and Roller Shutter', payload '{"calibration":"OFF","calibration_left":"OFF","calibration_right":"OFF","calibration_time":27.8,"linkquality":98,"motor_reversal":"OFF","motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving":"STOP","position_left":100,"position_right":56}'
Debug 2022-07-29 18:14:13Received Zigbee message from 'Master Bedroom Curtain and Roller Shutter', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":44,"tuyaMovingState":0}' from endpoint 2 with groupID 0
Info 2022-07-29 18:14:13MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain and Roller Shutter', payload '{"calibration":"OFF","calibration_left":"OFF","calibration_right":"OFF","calibration_time":27.8,"linkquality":94,"motor_reversal":"OFF","motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving":"UP","position_left":100,"position_right":56}'
Debug 2022-07-29 18:14:15Received Zigbee message from 'Master Bedroom Curtain and Roller Shutter', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":61,"tuyaMovingState":1}' from endpoint 2 with groupID 0
Info 2022-07-29 18:14:15MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain and Roller Shutter', payload '{"calibration":"OFF","calibration_left":"OFF","calibration_right":"OFF","calibration_time":27.8,"linkquality":91,"motor_reversal":"OFF","motor_reversal_left":"OFF","motor_reversal_right":"OFF","moving":"STOP","position_left":100,"position_right":39}'

Koenkk, 2022-07-30:
That looks as expected, I think the external converter has not been loaded. I've added `CUSTOM` at the end of the description. (https://github.com/Koenkk/zigbee2mqtt/issues/13278#issuecomment-1198570076). Can you check if that shows  up in the description of the device in the z2m frontend?

roy81f, 2022-07-30:
Yes, you are right, from some reason the external converter is not in use.

[image]

Koenkk, 2022-07-30:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13278#issuecomment-1198570076, please check again.

roy81f, 2022-07-30:
Perfect , thank you so much!

closed 2022-07-30 by roy81f

Koenkk, 2022-07-31:
Saw your message about the button, what exactly do you mean?

roy81f, 2022-07-31:
I meant if there any option to expose which button was pressed

Koenkk, 2022-07-31:
I'm not sure what buttons you mean, do you see any `Received zigbee message` in the debug log when clicking those buttons?

roy81f, 2022-07-31:
my mistake, its easy to find which button was pressed by using the new enities moving_left and moving_right.  Thank you for support!