Koenkk/zigbee2mqtt #13290: [New device support]: Aqara H1 Wired Dimmer Switch lumi.switch.rkna01

Deuteronomy93, 2022-07-26:
### Link

https://www.aqara.com/cn/smart-wall-switch-H1-with-neutral_overview

### Database entry

lumi.switch.rkna01

### Comments

Hi

I've added this device and created an external converter to process some of the information coming in however it looks like there are no converters available for the below functions that the device supports, listed below. I have tried all converters related to "aqaraOpple" without success.

- Twist left
- Twist right
- Twisting left whilst pressing
- Twisting right whilst pressing

To go through these easier I made an Excel file with the logs for each action, I found this better as I have a few new products, some that are in the process of trying to support and some that are already "supported" but still have debug logs. I can instead send a screenshot from Z2M if you prefer.

Any help would be much appreciated!

Many thanks

[Z2M H1 dimmer wired feedback.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/9186679/Z2M.H1.dimmer.wired.feedback.xlsx)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.rkna01'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'lumi.switch.rkna01', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara: Dimmer Switch H1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.xiaomi_on_off_action, fz.xiaomi_multistate_action, fz.xiaomi_power], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Deuteronomy93, 2022-07-29:
I've been playing around with this further, something to note is that for the press actions, I can put these into HA automations as triggers but not actions. When selecting the device as an action, the actions dropdown is greyed out.

ShadowFistOkami, 2022-07-31:
I could be wrong (I probably am), but the converter for the wireless version of this switch could work for you if you add your device's model ID https://www.zigbee2mqtt.io/devices/ZNXNKG02LM.html

```
    {
        zigbeeModel: ['lumi.remote.rkba01'],
        model: 'ZNXNKG02LM',
        vendor: 'Xiaomi',
        description: 'Aqara knob H1 (wireless)',
        meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
        exposes: [e.battery(), e.battery_voltage(),
            e.action(['single', 'double', 'hold', 'release', 'start_rotating', 'rotation', 'stop_rotating']),
            exposes.enum('operation_mode', ea.ALL, ['event', 'command']).withDescription('Button mode'),
            exposes.numeric('action_rotation_angle', ea.STATE).withUnit('*').withDescription('Rotation angle'),
            exposes.numeric('action_rotation_angle_speed', ea.STATE).withUnit('*').withDescription('Rotation angle speed'),
            exposes.numeric('action_rotation_percent', ea.STATE).withUnit('%').withDescription('Rotation percent'),
            exposes.numeric('action_rotation_percent_speed', ea.STATE).withUnit('%').withDescription('Rotation percent speed'),
            exposes.numeric('action_rotation_time', ea.STATE).withUnit('ms').withDescription('Rotation time'),
        ],
        fromZigbee: [fz.xiaomi_on_off_action, fz.xiaomi_multistate_action, fz.xiaomi_basic, fz.aqara_opple, fz.aqara_knob_rotation],
        toZigbee: [tz.aqara_opple_operation_mode],
        onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
    }
```

closed 2022-09-07 by github-actions[bot]