Koenkk/zigbee2mqtt #13291: Aqara "supported" devices with debug logs: FP1, Door & Window Sensor T1, Aqara Leak Sensor E1

Deuteronomy93, 2022-07-26:
### What happened?

In the debug logs there are multiple entries for these devices.

Sidenote: Regarding the adapter firmware it looks like the "Can be found in the frontend -> settings -> about -> coordinator revision" has been changed, I can confirm I got this from Koenkk's Github however and listed the information I have.

### What did you expect to happen?

I suppose this means that they have been added and supported with what are probably the triggered events etc, but their other reports have not been properly added?

One slight difference is regarding the Aqara Leak Sensor E1, someone has created a topic on this so I suppose it's in the process of being added? I copied the converter file from Otnow's response in this topic: https://github.com/Koenkk/zigbee2mqtt/issues/13252

However the debug logs still come in.

### How to reproduce it (minimal and precise)

These seem to be readings that are coming through passively without triggers (though I am not sure about the FP1) so honestly it would be just leaving them on for a short amount of time.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219

### Adapter

Sonoff B09KXTCMSC

### Debug log

Debug 2022-07-26 13:02:09Received Zigbee message from 'Aqara: Sensor_Door/Window P1 [1]', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,61,13,3,40,25,4,33,168,19,5,33,8,0,6,36,1,0,0,0,0,8,33,24,1,10,33,0,0,12,32,1,100,16,1,102,32,3,103,32,1,104,33,168,0,159,32,2,160,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2022-07-26 13:02:09lumi.magnet.ac01: Processed buffer into data {"1":3389,"3":25,"4":5032,"5":8,"6":1,"8":280,"10":0,"12":1,"100":1,"102":3,"103":1,"104":168,"159":2,"160":0}
Debug 2022-07-26 13:02:09lumi.magnet.ac01: unknown key 6 with value 1
Debug 2022-07-26 13:02:09lumi.magnet.ac01: unknown key 8 with value 280
Debug 2022-07-26 13:02:09lumi.magnet.ac01: unknown key 12 with value 1
Debug 2022-07-26 13:02:09lumi.magnet.ac01: unknown key 104 with value 168
Debug 2022-07-26 13:02:09lumi.magnet.ac01: Processed data into payload {"voltage":3389,"battery":100,"device_temperature":25,"power_outage_count":7,"state":"ON"}
Debug 2022-07-26 13:02:09lumi.magnet.ac01: Processed data into payload {"voltage":3389,"battery":100,"device_temperature":25,"power_outage_count":7,"state":"ON"}
Info 2022-07-26 13:02:09MQTT publish: topic 'zigbee2mqtt/Aqara: Sensor_Door/Window P1 [1]', payload '{"battery":100,"contact":false,"device_temperature":25,"linkquality":160,"power_outage_count":7,"state":"ON","voltage":3389}'
Debug 2022-07-26 13:02:13Saving state to file /config/zigbee2mqtt/state.json
Debug 2022-07-26 13:02:33Received Zigbee message from 'Aqara: Sensor_Presence FP1 [1]', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,30,5,33,1,0,8,33,54,1,10,33,0,0,12,32,20,16,32,1,18,32,0,101,32,0,102,32,3,103,32,0,104,32,0,105,32,1,106,32,1,107,32,3],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2022-07-26 13:02:33lumi.motion.ac01: Processed buffer into data {"3":30,"5":1,"8":310,"10":0,"12":20,"16":1,"18":0,"101":0,"102":3,"103":0,"104":0,"105":1,"106":1,"107":3}
Debug 2022-07-26 13:02:33lumi.motion.ac01: unknown key 8 with value 310
Debug 2022-07-26 13:02:33lumi.motion.ac01: unknown key 12 with value 20
Debug 2022-07-26 13:02:33lumi.motion.ac01: unknown key 16 with value 1
Debug 2022-07-26 13:02:33lumi.motion.ac01: unknown key 18 with value 0
Debug 2022-07-26 13:02:33lumi.motion.ac01: unknown key 104 with value 0
Debug 2022-07-26 13:02:33lumi.motion.ac01: Processed data into payload {"device_temperature":30,"power_outage_count":0,"presence":false,"motion_sensitivity":"high","monitoring_mode":"undirected","approach_distance":"medium"}
Debug 2022-07-26 13:02:33lumi.motion.ac01: Processed data into payload {"device_temperature":30,"power_outage_count":0,"presence":false,"motion_sensitivity":"high","monitoring_mode":"undirected","approach_distance":"medium"}
Info 2022-07-26 13:02:33MQTT publish: topic 'zigbee2mqtt/Aqara: Sensor_Presence FP1 [1]', payload '{"approach_distance":"medium","device_temperature":30,"linkquality":171,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":0,"presence":false,"presence_event":"leave","update":{"state":"idle"},"update_available":false}'
Debug 2022-07-26 13:10:30Received Zigbee message from 'Aqara: Sensor_Leak E1 [1]', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,160,11,3,40,25,4,33,0,0,5,33,1,0,8,33,8,0,10,33,0,0,12,32,30,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2022-07-26 13:10:30lumi.flood.acn001: Processed buffer into data {"1":2976,"3":25,"4":0,"5":1,"8":8,"10":0,"12":30,"100":0}
Debug 2022-07-26 13:10:30lumi.flood.acn001: unknown key 8 with value 8
Debug 2022-07-26 13:10:30lumi.flood.acn001: unknown key 12 with value 30
Debug 2022-07-26 13:10:30lumi.flood.acn001: Processed data into payload {"voltage":2976,"battery":36,"device_temperature":25,"power_outage_count":0,"state":"OFF"}
Debug 2022-07-26 13:10:30lumi.flood.acn001: Processed data into payload {"voltage":2976,"battery":36,"device_temperature":25,"power_outage_count":0,"state":"OFF"}

Deuteronomy93, 2022-07-26:
Also for the TVOC, noticing the below:
Debug 2022-07-26 15:31:24Received Zigbee message from 'Aqara: Sensor_TVOC [1]', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,25,12,3,40,24,4,33,168,19,5,33,7,0,6,36,1,0,0,0,0,8,33,24,1,10,33,0,0,12,32,1,100,41,188,9,101,33,251,17,102,33,180,0,103,32,2],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2022-07-26 15:31:24lumi.airmonitor.acn01: Processed buffer into data {"1":3097,"3":24,"4":5032,"5":7,"6":1,"8":280,"10":0,"12":1,"100":2492,"101":4603,"102":180,"103":2}
Debug 2022-07-26 15:31:24lumi.airmonitor.acn01: unknown key 6 with value 1
Debug 2022-07-26 15:31:24lumi.airmonitor.acn01: unknown key 8 with value 280
Debug 2022-07-26 15:31:24lumi.airmonitor.acn01: unknown key 12 with value 1
Debug 2022-07-26 15:31:24lumi.airmonitor.acn01: Processed data into payload {"voltage":3097,"battery":100,"device_temperature":24,"power_outage_count":6,"temperature":24.92,"humidity":46.03}
Debug 2022-07-26 15:31:24lumi.airmonitor.acn01: Processed data into payload {"voltage":3097,"battery":100,"device_temperature":24,"power_outage_count":6,"temperature":24.92,"humidity":46.03}

FlyingDomotic, 2022-08-12:
I've got the same issue, with 1.27.0, for the following items:
Device | Key | Value
lumi.vibration.aq1 | 6 | 0,917505
lumi.vibration.aq1 | 153 | 0
lumi.weather | 6 | 0,2
lumi.sensor_wleak.aq1 | appVersion | 6
lumi.sensor_wleak.aq1 | 6 | 0,1
lumi.sensor_magnet.aq2 | 6 | 0,1
lumi.sensor_motion.aq2 | 6 | 0,4
lumi.airmonitor.acn01 | 6 | 17
lumi.airmonitor.acn01 | 297 | 1

FlyingDomotic, 2022-09-13:
Still present in 1.27-2

FlyingDomotic, 2022-10-14:
Still present with 1.28-0

FlyingDomotic, 2022-11-15:
Still present in 1.28-0. Could someone provide me with values of missing keys, I can try to make the appropriate change.

FlyingDomotic, 2022-12-16:
Same proposal as last month...

closed 2023-01-25 by github-actions[bot]