Koenkk/zigbee2mqtt #13292: Devices fail spontaneously and stop reacting despite good LQI (timeout 10000)

makinitright, 2022-07-26:
### What happened?

I have a rather small Zigbee network and devices fail spontaneously, meaning that they stop reacting. And devices that appear to work fine still produce errors and changing colors for lights (hue oder tradfri) sporadically fail with large delays or unresponsiveness. 

Restarting the zigbee2mqtt docker (running on an unraid server) or dis- and reconnecting the conbee 2 stick (latest firmware) does not help at all.

They all have a solid LQI in my opinion. There is one device with N/A as it is powered off - so that is not part of the issue. The tradfri light that is failing at the moment has an LQI of 156.

[image]

Here is the current map generated a minute ago:

[image]

### What did you expect to happen?

Devices being stable within the network according to their LQI. 

### How to reproduce it (minimal and precise)

Just wait until a device fails. It can basically be any device. It is definitely not the range as other devices in the same room remain functional.

### Zigbee2MQTT version

1.24.0 (downgraded from 1.26.0 without any success regarding the issue)

### Adapter firmware version

0x26720700

### Adapter

ConBee2 (with extension cable)

### Debug log

`Error Publish 'set' 'state' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Failed to execute LQI for 'Ellie Deckenlicht'
Error Failed to execute LQI for 'Ankleide Deckenlicht'
Error Failed to execute LQI for 'Ellie Deckenlicht'
Error Failed to execute LQI for 'Ankleide Deckenlicht'
Error Publish 'set' 'state' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Publish 'set' 'state' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Publish 'set' 'state' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Publish 'set' 'color_temp' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 lightingColorCtrl.moveToColorTemp({"colortemp":378,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Publish 'set' 'brightness' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 genLevelCtrl.moveToLevelWithOnOff({"level":235,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Publish 'set' 'state' to 'Ellie Deckenlicht' failed: 'Error: Command 0xb4e3f9fffe216234/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Publish 'set' 'brightness' to 'Christophers Schranklicht' failed: 'Error: Command 0x001788010279a131/11 genLevelCtrl.moveToLevelWithOnOff({"level":61,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Error Failed to execute LQI for 'Wohnbereich Sideboard'
Error Failed to execute LQI for 'Christopher Deckenlicht'
Error Failed to execute LQI for 'Ellie Deckenlicht'
Error Failed to execute LQI for 'Ankleide Deckenlicht'`

makinitright, 2022-07-26:
Just powered everything down and waited 20 minutes before powering on again. Nothing changed except for another device failing…

GSzabados, 2022-07-27:
Tradfri bulbs have a tendency of crashing with older firmwares. (There are a bunch of known issues with them regarding crashing.) Try to update them to run the latest one and assess the situation again.
Hue bulbs are generally not good repeaters, and they work the best in their own environment. The Hue bridge gives a lot of neat features what neither ZHA or Z2M can.

makinitright, 2022-07-27:
Thanks for your input. Currently updating the bulbs. I used to run a Hue hub but wanted to have one consolidated environment - which I found in zigbee2mqtt - if not for the reporting issues above. 

I'll keep you updated!

GSzabados, 2022-07-27:
I have Hue, Wiser, SmartThings (not really used anymore) and Z2M and they run fairly well next to each other.

I kept the respective Hubs/Bridges for their capabilities what Z2M cannot offer. Like Hue with a Hue Sync box or the dynamic scenes, music visualisation, etc. Wiser works best with the TRVs and there is a custom component for it which is far more superior already than the Wiser app.
For everything else, there is Z2M. And works well with anything else what I put there. Indeed I have some issues time to time with IKEA bulbs, but that is down to the bulbs firmware.

closed 2022-09-04 by github-actions[bot]