Koenkk/zigbee2mqtt #13298: Kwikset Convert Oddity

nvx, 2022-07-26:
### What happened?

I have 3 Kwikset Convert zigbee deadbolts https://www.zigbee2mqtt.io/devices/66492-001.html across two different zigbee2mqtt setups.

The first two (one at each location) installed and setup fine without issues, but the third one (brought at the same time as the second) which I installed a couple of months later (real life got busy) seems to detect differently.

The first two show up as vendor / product of `Kwikset` / `SMARTCODE_CONVERT_GEN1`, while the third is detecting as `Kwikset` / `Smartcode` and shows as unsupported in zigbee2mqtt.

This seems similar to #3711, although the vendor shows Kwikset instead of Weiser. 

Setting up an external converter with the code simply copied from the `SMARTCODE_CONVERT_GEN1` with the zigbeeModel changed to `Smartcode` and all seems to work after repairing.

What's super strange is comparing units 2 and 3, they are identical in every way except for the zigbee address printed on the sticker. I even removed the zigbee board and there's a sticker on it indicating the firmware version/etc and that's identical between the two. The IEEE address even only differs by the last 3 hex nibbles so I imagine they rolled off the production line at a pretty similar time.

While adding `Smartcode` as an alias to `SMARTCODE_CONVERT_GEN1` would likely make it work out of the box, I'm a little hesitant to just submit suhc a PR since it looks like other models have shown up with the same symptom (#3711), perhaps like it's failing to detect what sort of device it should be? (I'm guessing the zigbee board is common between a few different versions of the lock?). Happy to submit such a PR if needed, but thought I'd open a discussion first.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

nvx, 2022-08-26:
Still relevant

nvx, 2022-09-26:
Still relevant. Happy to submit a PR for the fix that works for me, but curious to get some feedback first.

nvx, 2022-12-01:
Still happy to throw in a PR for this if that's the right option

Koenkk, 2022-12-01:
Please submit a PR

nvx, 2023-01-01:
Opened PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/5238

closed 2023-01-01 by Koenkk