Koenkk/zigbee2mqtt #13300: IKEA Tradfri LED2003G10

maxime-killinger, 2022-07-26:
### What happened?

No update seems to be available, the current version of my bulbs is 1.0.012 dated from 2020-11-11

### What did you expect to happen?

According to [the release notes](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) an update should be available

### How to reproduce it (minimal and precise)

 - Setup a new IKEA LED2003G10 bulb 
 - Try the ota update

### Zigbee2MQTT version

1.26.0-dev commit: 343ef7f

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

malaakso, 2022-07-28:
What makes you think there should be an update? Your bulb is "WS3.0", and the only entry seems to be:

TRÅDFRI N2, bulb design opal E27 - Globe & Tubular 470lm WS3.0 (V-1.0.012)

MattWestb, 2022-07-28:
IKEA CWS3 was getting one updated firmware but it was one lower version in the metadata so i was forcing updating and it was working OK for my system.
But IKEA was redrawing it some days later and have not releasing one updated version.
The release notes is changes and the update is deleted in it.
From reddit https://www.reddit.com/r/tradfri/comments/tk3gq3/recent_firmware_update_causing_issues_with/

One nice future was it changing CT the lower the brightness to warmer :-))

closed 2022-09-04 by github-actions[bot]