Koenkk/zigbee2mqtt #13304: Transition not working

Gensyn, 2022-07-26:
### What happened?

When turning on lights with brightness and transition, the lights turn on with the given brightness immediatly. I tried with Home Assistant and mosquitto and both give the same behavior. I tested this with IKEA Floalt and several Philips Hue bulbs (eg https://www.zigbee2mqtt.io/devices/9290020399.html), which all are supposed to support transition.

Example: mosquitto_pub -t "zigbee2mqtt/<name>" -m '{"state": "ON", "brightness": 200, "transition": 30}'

The light turns on at brightness 200. What I also think is noteworthy is that even though I have log_level set to debug, there is no log entry that an "ON" message was received. (There is also no entry for "OFF" messages)
The problem is the same when the light is already on and I just want to change the brightness with transition.

I also tried setting a transition just for those devices directly in z2m without any luck as well.

Am I doing something wrong? Noone else seems to have this problem ...

### What did you expect to happen?

The brightness fades in over time with the value given for transition.

### How to reproduce it (minimal and precise)

Turn on an IKEA Floalt or Philips Hue light bulb with transition.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

0x26490700

### Adapter

ConBee2/RaspBee2

Gensyn, 2022-07-29:
Nevermind. I tested this without being physically present at the lights and believed the Home Assistant UI when the brightness jumped directly to the target value. In person the transition works. Sorry.

closed 2022-07-29 by Gensyn

SkipperDo, 2022-11-27:
Same Problem here...
I just change from DECONZ to Zigbee2MQTT with Homeassistant & Mosquitto...

I Just change my automations in HA to match the new entities...
All my FLOALT Panel support transition but since the Change to Z2M the Transition don't work...
I tried a workaround in Z2M using the Settings (specific) in Z2M Frontend, and set the Transition value to 3. But nothing changes...

The FLOALT Panel directly "jump" to the target brightness... (in my case 75%)

Automation yaml:

`alias: Licht Gr. Flur
description: Tag & Nachtautomatisierung Gr. Flur, Es sei denn Nachtlicht ist aktiv.
trigger:
  - type: motion
    platform: device
    device_id: c63a81525de9f75bfb6f3b9b2cb63d1d
    entity_id: binary_sensor.bewegungsmelder_gr_flur_occupancy
    domain: binary_sensor
  - type: opened
    platform: device
    device_id: dc59b9a39a18d7394c3815b7139935a9
    entity_id: binary_sensor.entry_door_contact_2
    domain: binary_sensor
condition:
  - condition: state
    entity_id: automation.nachtlicht
    state: "off"
    enabled: false
action:
  - if:
      - condition: time
        after: "08:00:00"
        before: "21:30:00"
      - type: is_illuminance
        condition: device
        device_id: c63a81525de9f75bfb6f3b9b2cb63d1d
        entity_id: sensor.bewegungsmelder_gr_flur_illuminance_lux
        domain: sensor
        above: 0
        below: 50
        enabled: false
    then:
      - service: light.turn_on
        data:
          transition: 5
          kelvin: 2700
          brightness_pct: 75
        target:
          entity_id: light.floalt_gr_flur
      - delay:
          hours: 0
          minutes: 1
          seconds: 30
          milliseconds: 0
      - service: light.turn_off
        data:
          transition: 3
        target:
          entity_id: light.floalt_gr_flur
    else:
      - service: light.turn_on
        data:
          transition: 5
          kelvin: 2202
          brightness_pct: 1
        target:
          entity_id: light.floalt_gr_flur
      - delay:
          hours: 0
          minutes: 1
          seconds: 30
          milliseconds: 0
      - service: light.turn_off
        data:
          transition: 3
        target:
          entity_id: light.floalt_gr_flur
mode: restart`