Koenkk/zigbee2mqtt #13305: Default configuration doesn't start frontend

gregmac, 2022-07-26:
### What happened?

Installed zigbee2mqtt using the [Docker instructions](https://www.zigbee2mqtt.io/guide/installation/02_docker.html#running-the-container), but was unable to access web UI.

## Fix 

It seems like the [default configuration](https://github.com/Koenkk/zigbee2mqtt/blob/master/data/configuration.yaml) doesn't start the frontend.

Adding the following prior to starting the container fixes it:

```yaml
frontend:
  port: 8080
```

Seems like this should be the default configuration.

### What did you expect to happen?

Expected webUI to be accessible when accessing `http://...:8080`

### How to reproduce it (minimal and precise)

Follow instructions on https://www.zigbee2mqtt.io/guide/installation/02_docker.html:

1. wget https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/data/configuration.yaml -P data
2. ```sh
   $ docker run \
     --name zigbee2mqtt \
     --restart=unless-stopped \
     --device=/dev/ttyACM0 \
     -p 8080:8080 \
     -v $(pwd)/data:/app/data \
     -v /run/udev:/run/udev:ro \
     -e TZ=Europe/Amsterdam \
     koenkk/zigbee2mqtt
   ```
3. Try to load HTTP port 8080 (eg: `curl http://localhost:8080`)

   

### Zigbee2MQTT version

 1.26.0 commit: cd23aea

### Adapter firmware version

20211115

### Adapter

TI CC2531 USB stick

gregmac, 2022-07-26:
## Fix 

It seems like the [default configuration](https://github.com/Koenkk/zigbee2mqtt/blob/master/data/configuration.yaml) doesn't start the frontend.

Adding the following prior to starting the container fixes it:

```yaml
frontend:
  port: 8080
```

Seems like this should be the default configuration. Alternatively, the documentation needs to be modified.

Koenkk, 2022-07-27:
I prefer to document this instead, this makes the users aware that the frontend is running. Where would you expect this in the docs?

gregmac, 2022-07-27:
@Koenkk I would have looked for it in the ["Creating the initial configuration" section](https://www.zigbee2mqtt.io/guide/installation/02_docker.html#creating-the-initial-configuration), or the [linked "Configuring" page](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#configuring).

Maybe it can just link to the [Frontend page](https://www.zigbee2mqtt.io/guide/configuration/frontend.html)?

I got tripped up because following the docker instructions, the container successfully started and I thought "now what? is it working?". Seeing `-p 8080:8080` I assumed that was a web UI, and when it didn't load, my assumption was something broken and I wasn't sure if it was zigbee2mqtt or docker on this machine -- and so I didn't even continue on to the "Configuration" part of the docs. That parameter is pointless (and misleading) if the frontend is off.

I still think it'd be better to just change the default. My assumption is it's a tiny handful of users that don't want the UI, and all of them would have enough knowledge to change the configuration file. At the least, having `# frontend:` there but commented out would make it obvious -- I didn't even know that was an option at the time.

Koenkk, 2022-07-28:
I've added it to the docs now

> I still think it'd be better to just change the default. My assumption is it's a tiny handful of users that don't want the UI,

This can maybe be done later, I first need to find a solution how I can modify the configuration.yaml without breaking existing git-based setups.

closed 2022-09-04 by github-actions[bot]