Koenkk/zigbee2mqtt #13306: [New device support]: Lexman Wireless Slart Door Sensor with Vibration

Sonyoyo, 2022-07-26:
### Link

https://enki-home.com/produit/lot-de-3-detecteurs-douverture-connecte-lexman/

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0x5c0272fffeab2f0d","nwkAddr":32176,"manufId":4727,"manufName":"ADEO","powerSource":"Battery","modelId":"LDSENK08","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LDSENK08","manufacturerName":"ADEO","powerSource":3,"zclVersion":3,"appVersion":34,"stackVersion":6,"hwVersion":1,"dateCode":"20200519","swBuildId":"2.02"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c315d8","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c315d8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":34,"stackVersion":6,"hwVersion":1,"dateCode":"20200519","swBuildId":"2.02","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1658350595571,"defaultSendRequestWhen":"active"}

### Comments

Hello,
I am trying to integrate a new zigbee device: a door sensor with vibration from ADEO/Lexman.

After pairing, I created a converter (hereafter), using a [Tuya device](https://www.zigbee2mqtt.io/devices/TS0601_vibration_sensor.html) as example, but I am not able to make it to work. Opening/moving the device makes it generate messages such as: 

```
Received Zigbee message from '0x5c0272fffeab2f0d', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":52}' from endpoint 1 with groupID 0
```
However the message is followed by: 
```
No converter available for 'LDSENK08' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":52}'
```
Here `zonestatus` is `52` but may be anything between at least `48` and `55` (I think)...and I am not sure whether these numbers correspond to specific actions on the devices.

I looked at fromZigbee functions, searching for 'ssIasZone', 'zonestatus' and 'vibration'. However I do not understand how these pieces of codes work...neither I found zonestatus values corresponding to my case (48 -> 52).

Thanks in advance for any help. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK08'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LDSENK08', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Lexman Wireless Smart Door Window Sensor with Vibration', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [],
    toZigbee: [],
    exposes: [e.battery(), e.contact(), e.vibration()],
};

module.exports = definition;
```

Koenkk, 2022-07-28:
Can you check with: `fromZigbee: [fz.ias_contact_alarm_1],`

Sonyoyo, 2022-07-29:
Thanks for your help.
Now I have no more error messages. However, `battery` and `vibration` are set to `null` and do not change. I also tried by removing the device and pairing it again, with no improvement (done just once; should I retry?).

I did not know that there was also a tamper sensor on the device and now I better understand the `zonestatus` numbers (I guess). 
With `tamper: false`, `contact: true` and `battery_low: false`:
-  `50` means `vibration`
-  `48` means `no vibration`

However, the device status does not reflect this yet.

Koenkk, 2022-07-29:
Should work with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    LDSENK08: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                contact: !((zoneStatus & 1) > 0),
                vibration: (zoneStatus & 1<<1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    }
}

const definition = {
    zigbeeModel: ['LDSENK08'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LDSENK08', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Lexman Wireless Smart Door Window Sensor with Vibration', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fzLocal.LDSENK08, fz.battery],
    toZigbee: [],
    exposes: [e.battery_low(), e.contact(), e.vibration(), e.tamper(), e.battery()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;
```

Battery % might take some time to be reported.

Sonyoyo, 2022-08-02:
Thanks for your prompt and wonderful support.
Now I have all the variables available: contact, vibration, tamper, battery, battery_low.

That's perfect!

Koenkk, 2022-08-03:
Thanks, added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-03 by Koenkk