Koenkk/zigbee2mqtt #13307: Broken features on Woox R7051 Siren

Volan3D, 2022-07-26:
### What happened?

Hey. I was looking to add a simple siren to Home Assistant. Found that the [Woox R7051](https://www.zigbee2mqtt.io/devices/R7051.html#woox-r7051) seemed to be the one with the most supported features, but most of those seem to be broken.

First of all, it took hours to figure out how to trigger this alarm. Thought I was doing something wrong, but it's the code listed on the device page that does not work, at least not on my device. The only thing I can do with this siren is to turn it on and off with MQTT, and I can change the volume on the device page. Almost everything else doesn't work. Are the various modes supposed to sound different? I can not change "level" "strobe_level" "strobe false\true" "duty_cycle" or "duration". Buttons are greyed out and values does nothing.

To trigger this alarm, I need to publish this to topic: zigbee2mqtt/Siren Woox 1/set:

Payload to trigger alarm:
{   "warning": {   "mode": "burglar"   } }

And to stop alarm after a time delay (in the automation):
{   "warning": {   "mode": "stop"   } }

If I send the following, the siren will trigger, but changing any settings except for "mode" to stop, does nothing or makes the siren fail to trigger:

{"warning":{"mode":"burglar", "duration":"1", "level":"low", "strobe":"false", "strobe_duty_cycle":"2"}}
Is this supposed to be sent with the payload at all?

If I type it like this, it will fail to trigger (this is how it's written on the webpage above):
{"warning":{"mode":burglar, "duration":1, "level":low, "strobe":true, "strobe_duty_cycle":2}}

[image]

### What did you expect to happen?

Supported features should be working.

### How to reproduce it (minimal and precise)

You need the siren.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/9193672/debug.txt)

mimas71, 2022-07-30:
Hi, I'm in exactly the same situation, it would be great to control the siren with zigbee2mqtt

closed 2022-09-07 by github-actions[bot]

bzeiss, 2022-11-09:
Pity, same problem. I would really like if this worked properly.

bzeiss, 2022-11-09:
I think this device should at least be marked as broken on the devices page.

Volan3D, 2022-11-09:
> I think this device should at least be marked as broken on the devices page.

Yes.. Z2M might support the most items, but I'm starting to question the quality of support, on some items..

I have managed to get this siren to some use by creating automations with on\off and delays, to at least get beeps out of it. I have one for a single beep, one for double beep, and one for full alarm. Now I just call the helper that triggers the beeping. Saves me from putting in the code in other automations.

bzeiss, 2022-11-09:
I have done the same in my automations. So it basically works for what I get those devices, but I have ordered the woox gateway and a cc2531 stick + accesories for zigbee sniffing. It might be an easy enough device to get into creating support for devices.

Volan3D, 2022-11-10:
> I have done the same in my automations. So it basically works for what I get those devices, but I have ordered the woox gateway and a cc2531 stick + accesories for zigbee sniffing. It might be an easy enough device to get into creating support for devices.

Do you think you can fix this siren? Could this issue be a firmware problem, maybe it's too old, or newer then the one used to make it supported? I don't think it supports OTA, so I cant update if this is the case.

Btw, I have a wall mount for this siren on Printables, if you have a 3dprinter. Just search for 299479

bzeiss, 2022-11-10:
Who knows whether I can fix it. We'll have to see about that. I certainly want to fiddle around with it for a bit. I've had that thought about firmware deviations as well, but it's hard to figure out whether that is the case. Searching github of herdsman-converters for a little, it seems @mephistos is the original author of the support. Maybe he can add some insights.

bzeiss, 2022-11-11:
Ok, I have some news. I have received the woox hub today and paired the siren to it. The firmware version I have is according to the app 1.1.0 and there is no firmware update available. 

In the settings dialog, I can configure see/the following things, whatever some of them mean:
- alarm volume: muted, low, mid, high
- power supply state: "Supply normal", probably battery when unplugged
- battery: so far it displays only 0%. Maybe it stays there. Don't know.
- Alarm time: 0 or any other positive integer for seconds
- Alarm Set (greyed out)
- Battery Low: On
- Sire battery powered: On
- Alarm: On
- Siren Normal: On
- Sire external power supply: On

When I configure an alarm in an automation, I can do the following things:
- Alarm volume: muted, low, mid high
- Alarm switch: on, off
- brightness: dark, weak, middle, strong
- alarm time

The stuff that is there kind of works, but seems a bit buggy so far. When I activate an automation with volume low, it's not always low. Maybe the settings work in home assistant as well when triggered in specific manner maybe. There are _no_ different sounds to choose from.

mephistos, 2022-11-16:
Hello here,

Sorry but i'm busy so couldn't answer faster. I do my best to make work the siren on the first job but without the hub, so i cannot sniff the zigbee network or cannot check what option is available from the official hub.
But hey, i make it work, not with all options, but working : you can start and stop the alarm, before my contribution the alram was not reconize at all...

Because i don't have the official hub, and so i don't know at all the official options, i use the default siren component on Z2M...

I know not everything is working, but hey, it's an open source project, you can contribute to the documentation if you want a change, or to the code for make enhancement if you have the hub.

For me i cannot do more because no time and no hardeware, sorry.

This payload work for sure : 
`{"warning": {"mode": "emergency", "level": "very_high", "strobe_level":"very_high", "strobe": true, "strobe_duty_cycle": 1, "duration": 0}}`

The siren never stop or stop on this hardware default timer, i don't know, i've never let it yell more than few minutes....

And this payload work for stop the siren : 

`{"warning": {"mode": "stop", "level": "very_high", "strobe_level": "very_high", "strobe": true, "strobe_duty_cycle": 1, "duration": 1}}`

So please contribute to the project for making a best documentation (to complete what is working or not, and maybe exemple of payload), or a best code with all options provided by the offical hub :)

bzeiss, 2022-11-24:
So I have sniffed some stuff when using the woox siren with the woox app and have documented the parts that I thought would be relevant (I hope I didn't leave out any relevant parts, I'm new to this):
[woox-siren-sniffed-stuff.txt](https://github.com/Koenkk/zigbee2mqtt/files/10087142/woox-siren-sniffed-stuff.txt)

I still have to figure out how to test these results with the dev console, maybe someone else is quicker to integrate this (if possible).