Koenkk/zigbee2mqtt #13311: Moes Smart Knob: Not Supported?

iculookn, 2022-07-27:
### Link

https://www.aliexpress.com/item/1005003775694484.html?spm=a2g0o.order_list.0.0.18751802UKSNMh

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0xe0798dfffe884034","nwkAddr":50974,"manufId":4098,"manufName":"_TZ3000_qja6nq5z","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1658912926067,"defaultSendRequestWhen":"immediate"}

### Comments

It seems like this device should be supported, but I am getting "Not Supported"
Maybe the Manufacturer number is new?

Zigbee Model: TS004F
Zigbee Manufacturer: _TZ3000_qja6nq5z

https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html
https://community.home-assistant.io/t/zigbee2mqtt-tuya-moes-smart-knob-ers-10tzbvk-aa/41998

[image]
9

Koenkk, 2022-07-27:
`_TZ3000_qja6nq5z` is supported but make sure you are using the correct addon repo (check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0) and reinstall the edge addon to update to the latest version.

iculookn, 2022-07-28:
ok, that's strange. I originally had stable installed, and then this week I changed to EDGE via the addon store.
Now I followed the instructions as is listed in that link. removed the add-on, re-added and all works now.

THANK YOU

closed 2022-07-28 by Koenkk