Koenkk/zigbee2mqtt #13315: Docker: Using environment variable ZIGBEE2MQTT_CONFIG_MQTT_SERVER fails

mmattel, 2022-07-27:
### What happened?

Instead defining the MQTT settings in the configuration.yaml file, I tried to do this via environment variables in docker compose as described in the zigbee2mqtt manual. Here I identified a strange behaviour, possibly a bug.

Having the mqtt settings in the conf file, using mosquitto as broker works great, it fails when defining the server via an environment variable. Replace fqdn accordingly.

The config file contains (works):
```
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://fqdn:1883'
  # MQTT server authentication, uncomment if required:
  user: ****
  password: ****
```
The docker-compose environment definition (fails):
```
    environment:
      - ZIGBEE2MQTT_CONFIG_MQTT_SERVER='mqtt://fqdn:1883'
```
This is the error message from the log (shortened):
```
Connecting to MQTT server at "mqtt://fqdn:1883"
Zigbee2MQTT:error: MQTT failed to connect: Missing protocol
```

Note that using user/pass as env variable works.

### What did you expect to happen?

When starting up the container using the `ZIGBEE2MQTT_CONFIG_MQTT_SERVER` environment variable, the service should start in the same way as you would define it in the config file.

### How to reproduce it (minimal and precise)

Comment the mqtt.server setting in the config file and add the env variable having the same content. 

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

not relevant

### Adapter

ConBee2

### Debug log

info  2022-07-27 18:06:41: Starting Zigbee2MQTT version 1.26.0 (commit #cd23aea)
info  2022-07-27 18:06:41: Starting zigbee-herdsman (0.14.40)
info  2022-07-27 18:06:42: zigbee-herdsman started (resumed)
info  2022-07-27 18:06:42: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":120,"p>
info  2022-07-27 18:06:42: Currently 0 devices are joined:
warn  2022-07-27 18:06:42: `permit_join` set to  `true` in configuration.yaml.
warn  2022-07-27 18:06:42: Allowing new devices to join.
warn  2022-07-27 18:06:42: Set `permit_join` to `false` once you joined all devices.
info  2022-07-27 18:06:42: Zigbee: allowing new devices to join.
info  2022-07-27 18:06:43: Connecting to MQTT server at "mqtt://fqdn:1883"
error 2022-07-27 18:06:43: MQTT failed to connect: Missing protocol
error 2022-07-27 18:06:43: Exiting...
info  2022-07-27 18:06:43: Stopping zigbee-herdsman...
info  2022-07-27 18:06:45: Stopped zigbee-herdsman

Koenkk, 2022-07-28:
The following works fine for me: `ZIGBEE2MQTT_CONFIG_MQTT_SERVER="mqtt://localhost"  npm start `. Can you try changing `- ZIGBEE2MQTT_CONFIG_MQTT_SERVER='mqtt://fqdn:1883'` to `- ZIGBEE2MQTT_CONFIG_MQTT_SERVER="mqtt://fqdn:1883"`

mmattel, 2022-07-29:
I have tried the env with singe and double quotes before writing this issue, retested after your comment, the result always errors. I interpret in your example a local install, not via docker-compose, maybe there is a difference.

The interesting message is `Missing protocol` which means that the parser imho does not get the `mqtt://` part.

With more tests, I found the cause. I now can use the env variable but you MUST NOT put the url into any quotes. This is working `- ZIGBEE2MQTT_CONFIG_MQTT_SERVER=mqtt://fqdn:1883`

Maybe to check in the code if quotes create a difference in a local or in docker install.

mmattel, 2022-08-29:
not stale

closed 2022-10-07 by github-actions[bot]