Koenkk/zigbee2mqtt #13317: Centralite 3400 Keypad not arming/disarming

Drakmyth, 2022-07-28:
### What happened?

Attempting to arm a Centralite 3400 keypad by sending
```json
{
  "arm_mode": {
    "mode": "arm_all_zones"
  }
}
```
to `zigbee2mqtt/<FRIENDLY_NAME>/set` appears to have no effect on the keypad itself. This is the case regardless of the value of `mode`.

Worth noting that pairing the device and the process of submitting a pin from the keypad and using an automation to validate the pin and send the acknowledgement works just fine. It's just attempting to set the arm mode that does not work.

Also, my keypad is a model 3400 while the frontend identifies it as a 3400-D.

### What did you expect to happen?

Depending on the mode, the status key and indicator light should turn red to indicate armed, or the keypad should beep periodically if using a `*_delay` mode.

### How to reproduce it (minimal and precise)

1. Pair a Centralite 3400 to Z2M
2. Send `arm_mode` command to `zigbee2mqtt/<FRIENDLY_NAME>/set` topic
3. Observe that nothing happened

### Zigbee2MQTT version

1.26.0-dev commit: eda087a

### Adapter firmware version

0x26780700

### Adapter

ConBee2 Firmware 0x26780700

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9205396/log.txt)

Drakmyth, 2022-08-28:
Still seeing this behavior

Drakmyth, 2022-12-04:
Is this device just no longer supported? I recently updated to [1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [52e545f](https://github.com/Koenkk/zigbee2mqtt/commit/52e545f) hoping that maybe I would see improved functionality. Now everything except `linkquality` and `occupancy` only report `null`, `occupancy` is only ever `false`, and on the UI the battery level just shows a flashing red % sign. I have tried deleting and re-pairing the device multiple times and have tried changing the zigbee channel to multiple different values (initially 11, then 24 and 25) with no improvement.

On [this page](https://www.zigbee2mqtt.io/devices/3400-D.html#centralite-3400-d), following the `Arming/Disarming from the server` section, the only listed value that seems to have any impact is `disarm` which makes the top 3 buttons stop flashing red in sequence.

rindlerblabla, 2023-01-23:
```
service: mqtt.publish
data:
  topic: zigbee2mqtt/Keypaden/set
  payload: "{\"arm_mode\":{\"mode\":\"arm_all_zones\"}}"
  retain: true
```

This works just fine for me.

Drakmyth, 2023-01-24:
> ```
> service: mqtt.publish
> data:
>   topic: zigbee2mqtt/Keypaden/set
>   payload: "{\"arm_mode\":{\"mode\":\"arm_all_zones\"}}"
>   retain: true
> ```
> 
> This works just fine for me.

What model keypad do you have? Given the service/data format you provided I imagine you're interacting with the device via Home Assistant? I've found that if I press a zone button on the keypad and punch in a 4-digit number (any 4-digit number) I do receive a message on the `zigbee2mqtt/<FRIENDLY_NAME>/action` topic with the name of the zone button I pressed or `disarm` if I didn't press one, but the `action_code` field in the payload on `zigbee2mqtt/<FRIENDLY_NAME>` just says `null` so I'm not sure how I'm supposed to validate the code was correct.

Do you have any automations wired up to handle the handshake with the keypad, or do you only set the zone from the server?

rindlerblabla, 2023-01-24:
I am using the Xfinity keypad, recognised as URC4450BC0-X-R.

Using the Home assistant Alarmo HACS integration, with a four digit code. Am sure there are blueprints but I had it already set up before so I am still using different automations.

I am arming with following automation:

```
trigger:
  - platform: mqtt
    topic: zigbee2mqtt/Keypaden
condition:
  - condition: template
    value_template: "{{ trigger.payload_json.action == 'arm_all_zones' }}"
action:
  - service: alarm_control_panel.alarm_arm_away
    target:
      entity_id: alarm_control_panel.hus
    data:
      code: "{{ trigger.payload_json.action_code }}"
mode: single
```

closed 2023-03-03 by github-actions[bot]