Koenkk/zigbee2mqtt #13318: [Feature request]: Release a packaged version of zigbee2mqtt with nodejs in the package

rohankapoorcom, 2022-07-28:
### Is your feature request related to a problem? Please describe

Currently, installing Zigbee2MQTT (without Docker) requires installing Node.js and managing it's versions manually. This causes an additional maintenance hurdle when performing updates because you have to make sure that the system installed version of Node.js is still compatible with Zigbe22MQTT.

### Describe the solution you'd like

Utilizing [pkg](https://www.npmjs.com/package/pkg) to generate a packaged executable alongside the docker images will make it easy for users to install manually without worrying about Node.js versions.

For example ZWavejs2Mqtt provides this functionality which makes the [installation](https://zwave-js.github.io/zwavejs2mqtt/#/getting-started/other-methods?id=nodejs-or-pkg-version) as simple as downloading a zip file and unzipping it.

### Describe alternatives you've considered

Alternatively, the docker image could be used for easy installations, but some environments are not very well suited to docker.

### Additional context

Zwavejs2Mqtt uses this [script](https://github.com/zwave-js/zwavejs2mqtt/blob/master/package.sh) for packaging which is triggered by a Github Action as part of the release flow.

sjorge, 2022-07-28:
There was already an attempt https://github.com/Koenkk/zigbee2mqtt/pull/10958

rohankapoorcom, 2022-08-28:
Thanks @sjorge. I'll add a comment there to see what's needed to move that forward. I'm happy to assist.

closed 2022-10-06 by github-actions[bot]