Koenkk/zigbee2mqtt #13319: Gledopto GL-MC-001P keeps reconnecting

X52p, 2022-07-28:
### What happened?

I migrated from ZHA to Zigbee2Mqtt (Home Assistant Addon)
Most devices connected just fine, but my Gledopto GL-MC-001P doesn't.

I did the following steps:
1. Permit Join 
2. reset the device (using the guide from [0]) (Device blinked green afterwards)
3. device connects to the network (and stops blinking)
4. Zigbee2Mqtt shows that device connected and was interviewed successfully (device is also shown correctly with correct capabilities.
5. about a few seconds to up to a minute later, the message appears that the device disconnected from the network. (the device does not react to commands during the time it is listed as connected)

I tried this multiple times and placed the device in different spots around the house (including near the Zigbee Dongle)

The device worked before with the same Adapter. The only thing I did was Disable ZHA, install Zigbee2Mqtt and reconnect the device.

I found this issue which might be related: #11202

[0]: https://www.zigbee2mqtt.io/devices/GL-MC-001P.html

### What did you expect to happen?

I expected that the device paired, connected to the network and stayed connected.

### How to reproduce it (minimal and precise)

1. Permit Join 
2. reset the device (using the guide from [0])
3. wait 2 seconds
4. device disconects itself

### Zigbee2MQTT version

 1.26.0

### Adapter firmware version

20211217

### Adapter

SMARTLIGHT CC2652P Zigbee USB Adapter SLZB-02

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9208443/log.txt)

steveb166, 2022-08-08:
Can't help you here - but I'm suffering with almost exactly the same issue. Mine wasn't caused the same way but has the same symptom. 

Long story short - I had to re-add most of my zigbee devices in a failed SD card to SSD migration. This is the only device I cannot get to re-join (well, stay joined).

I have tried with both ZHA and Z2M with no luck.

adgenet, 2022-08-16:
I have just migrated coordinators from a ConbeeII to a Sonoff Dongle-P and encountered this exact issue.
As alluded to in the OP, I suspect the cause to be a repeat of [#11202](https://github.com/Koenkk/zigbee2mqtt/issues/11202) . Has Gledopto provided anybody with updated firmware for this controller?

X52p, 2022-09-16:
The issue still exists. I would be glad if anyone has suggestions what to try, or how to start debugging this.

X52p, 2022-10-17:
The issue still exists, and I have no Idea how to fix this. Suggestions are welcome.

closed 2022-11-26 by github-actions[bot]