Koenkk/zigbee2mqtt #13322: Moes / Tuya TS0013: Unable to control 3 gang independently

lucddm, 2022-07-29:
### What happened?

I have successfully paired the Moes 3 gang Zigbee switch. However, it is impossible to turn ON or OFF 1 gang at a time. For example, if I try to turn ON the gang 1 (left) via market or via the Zigbee2MQTT front-end all the gangs will turn ON and the same thing happens when turning it OFF. 

Please note that the gang does not mater. The same thing happens if you try with the gang 2 and 3. 
I have tried 2 different Zigbee adapter (CC2531 and EZSPV8), and to create an external converter and the same thing happened. 

[image]

https://user-images.githubusercontent.com/108508482/181698448-be07949c-676b-46bb-aef6-b006560d08aa.mp4

### What did you expect to happen?

When I turn ON or OFF the gang 1, I expect that only the gang 1 would respond (turn ON or OFF). 

### How to reproduce it (minimal and precise)

I can reproduce it by simply turning ON or OFF one of the 3 gangs.

### Zigbee2MQTT version

 1.25.1

### Adapter firmware version

20180507

### Adapter

CC2531

Koenkk, 2022-07-29:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0013'],
    model: 'TS0013',
    vendor: 'TuYa',
    description: 'Smart light switch - 3 gang without neutral wire',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right')],
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    whiteLabel: [{vendor: 'TUYATEC', model: 'GDKES-03TZXD'}],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        try {
            for (const ID of [1, 2, 3]) {
                const endpoint = device.getEndpoint(ID);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            }
        } catch (e) {
            // Fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/4872
        }
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

lucddm, 2022-08-01:
Hi @Koenkk, thanks for your reply. I just tried it and it works! Thank you. 

My understanding is that tuya switches need to be asked basic information in 'genBasic' in order to the work properly, is that right? 

I have modified the driver in order to set 2 variables ('power_on_behavior' and 'backlight_mode'). Hope this can help someone needing to set these variables. 

```js 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0013'],
    model: 'TS0013',
    vendor: 'Moes',
    description: 'Smart light switch - 3 gang without neutral wire',
    extend: extend.switch(),
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
    fromZigbee: extend.switch().fromZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), 
	exposes.enum('power_on_behavior', ea.ALL, Object.values(tuya.moesSwitch.powerOnBehavior)),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']).withDescription('Indicator light status: LOW: Off | MEDIUM: On| HIGH: Inverted')],
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    whiteLabel: [{vendor: 'TUYATEC', model: 'GDKES-03TZXD'}],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        try {
            for (const ID of [1, 2, 3]) {
                const endpoint = device.getEndpoint(ID);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            }
        } catch (e) {
            // Fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/4872
        }
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;
```

Koenkk, 2022-08-01:
Included, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-01 by Koenkk

brio11, 2023-07-13:
If you are running on Zigbee2mqtt 1.18.1-1 I do recommend to update it to newer one, here is 'how to':
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation

azmat007, 2023-07-25:
hi
I did according to what was said
My enviroment is ZHa on home assistant using Zigbee conbee 2

The tuya model is TS00013

I did as you stated above it didnt work but then I realized its for Zigbee2mqtt

DO you have any way to do it on ZHA??

i tried the ZHA toolkit and the magic spell code too added that in configuration.yaml

It didnt work.  Please let us know there are many who are looking for the similar answer. 

Thankyou so much

thepaulf, 2024-05-31:
I have the same problem on zigbee2mqtt version [1.37.1] (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.1) 
Toggling one through Zigbee toggles them all, and Power-on behavior and backlight mode are also not working

grabbou, 2024-07-14:
This issue still persist. To fix, you must define external coordinator as follows:
```js
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const definition = {
    zigbeeModel: ['TS0013'],
    model: 'TS0013',
    vendor: '_TZ3000_qewo8dlz',
    description: 'Smart light switch - 3 gang without neutral wire',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3}}), onOff({"powerOnBehavior":false,"endpointNames":["2","3","1"]}), identify()],
    exposes: [e.switch().withEndpoint('1'), e.switch().withEndpoint('2'), e.switch().withEndpoint('3')],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        try {
            for (const ID of [1, 2, 3]) {
                const endpoint = device.getEndpoint(ID);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            }
        } catch (e) {
            // Fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/4872
        }
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;
```
It's weird Moes gets reported as Tuya, but I suspect something with configuration that makes it stop working.

thiagoanegreiros, 2024-11-04:
I still have the same issue even with @grabbou solution
=(

thiagoanegreiros, 2024-11-04:
Sorry guys... it was my zigbee2mqqt that was not updated. I needed to uninstall it and add the new repo on the store to install the updated version.

So double check the HA zigbee2mqqt version if anyone gets this

miguelmontanod, 2026-02-15:
> Hi [@Koenkk](https://github.com/Koenkk), thanks for your reply. I just tried it and it works! Thank you.
> 
> My understanding is that tuya switches need to be asked basic information in 'genBasic' in order to the work properly, is that right?
> 
> I have modified the driver in order to set 2 variables ('power_on_behavior' and 'backlight_mode'). Hope this can help someone needing to set these variables.
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['TS0013'],
>     model: 'TS0013',
>     vendor: 'Moes',
>     description: 'Smart light switch - 3 gang without neutral wire',
>     extend: extend.switch(),
>     toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
>     fromZigbee: extend.switch().fromZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
>     exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), 
> 	exposes.enum('power_on_behavior', ea.ALL, Object.values(tuya.moesSwitch.powerOnBehavior)),
>         exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']).withDescription('Indicator light status: LOW: Off | MEDIUM: On| HIGH: Inverted')],
>     endpoint: (device) => {
>         return {'left': 1, 'center': 2, 'right': 3};
>     },
>     whiteLabel: [{vendor: 'TUYATEC', model: 'GDKES-03TZXD'}],
>     meta: {multiEndpoint: true},
>     configure: async (device, coordinatorEndpoint, logger) => {
>         await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
>         try {
>             for (const ID of [1, 2, 3]) {
>                 const endpoint = device.getEndpoint(ID);
>                 await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
>             }
>         } catch (e) {
>             // Fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/4872
>         }
>         device.powerSource = 'Mains (single phase)';
>         device.save();
>     },
> };
> 
> module.exports = definition;

Your solution works! Thank you very much!