Koenkk/zigbee2mqtt #13323: Tuya Multi-Control Linkage Associtation

olili, 2022-07-29:
### Is your feature request related to a problem? Please describe

Tuya is offering a feature called "Multi-Control Linkage / Associtation" for switch devices.  
It is mainly a multi-way controlling of one relay-switch. 

Details and API can be found here:
https://developer.tuya.com/en/docs/app-development/devicemulticontrol?id=Ka6ki8l92gduk

### Describe the solution you'd like

Does anybody see a chance to activate this feature by Z2M? 

### Describe alternatives you've considered

Exisiting zigbee binding mechanism are not working as no output clusters are defined for binding. 

### Additional context

Setup: latest z2m version, C2652P Sonoff-stick, latest z2m-firmware

olili, 2022-08-29:
unstale

ciB89, 2022-09-11:
+1 for this.

olili, 2022-10-13:
I bought a TuYa coordinator and configured the feature withe TuYa app:

I think "multi-association" is not done by a zigbee binding as the feature is not working in case of switched.-off router. 

So I assume that it is realzized within the coordinator firmware. 

Maybe "multi-association" would also be a awesome z2m feature for one of the next release?

O.

Koenkk, 2022-10-18:
If this is done on the gateway site, you can make such feature by creating e.g. an automation in Home Assistant. Having such automation system within z2m is out-of-scope.

Aleborg, 2022-11-15:
Really need this as well, bought plenty of TS110E ( 1 and 2 gangs, [Link to aliexpress](https://www.aliexpress.com/item/1005004580552856.html) ) just for this reason. 
Had them multi-controlled in Tuya but have been starting to migrate everything into Zigbee2MQTT, was really suprised when I noticed that this wasn't possible :(

olili, 2022-12-31:
unstale

sstefanowski, 2023-01-02:
I also think that having "Multi-control" in Z2M could be a nice feature... especially considering the fact that Tuya has a big piece of the Zigbee device market and many switches are relying on this in 2-Way setups.

I have also installed many Zigbee (Tuya) switches at home and... in several cases I have 2-Way circuits. I implemented automation in HA trying to keep the switches in sync but, there are two major issues a) significant lag and b) possible feedback loops.
You can try to reduce the risk of a feedback loop by implementing special conditions in HA automation but still, when you press on-off-on-off several times quickly the whole system gets crazy as the physical state of the switch cannot be quickly synced because significant lag. 
So in my opinion, if any Zigbee devices could be "logically grouped" in Z2M and exposed to the MQTT client as a single switch this could be really nice. It would reduce a lag second it would be MQTT-client-firendly.

MattWestb, 2023-01-09:
If working with Zigbee light its one must using `Zigbee Controllers` that can binding to devices or groups that is making them working also then the coordinator / host  system is not working.
All other ways is host bases and is getting lags and is **not** working if the host system is having problems.

tuyas local linkage is local in the  ZBGW = its not sending the command to the cloud and getting command back and sending it to the light so i think the right name shall being "cloud less linking".

WojtaszekMarek, 2023-02-13:
stale

olili, 2023-03-16:
unstale. 

Still hoping that some more people need multi-way switch feature wihout developping complex HA scripts for supressing/handling feedback loops.

If yes, please "like" this comment.

olili, 2023-04-16:
another question: is it possible to implement such a feature as a Z2M extension?

dalbre, 2023-05-12:
I would also strongly second this. 2-way-switches

closed 2023-06-20 by github-actions[bot]

piwi3910, 2024-04-04:
any news on this

sstefanowski, 2024-04-04:
I guess nothing new. 
In meantime I created automation on HA to synchronize state of stair switches. Of course there is at lease one second lag between switches and sometimes switches go crazy when you press quickly on/off several times. 

This is my code of HA automation. Anyone has better implementation of such automation?

```
alias: (LS) Light Synchro Hall 1
description: ""
trigger:
  - platform: state
    entity_id:
      - light.bseed_hall_11
      - light.bseed_hall_12
    to: null
condition:
  - and:
      - condition: template
        value_template: "{{ trigger.to_state.state in ['off', 'on'] }}"
      - condition: template
        value_template: "{{ states('light.bseed_hall_11') != states('light.bseed_hall_12') }}"
action:
  - service: homeassistant.turn_{{ trigger.to_state.state }}
    target:
      entity_id: |-
        {% if trigger.entity_id !='light.bseed_hall_11' %}
          light.bseed_hall_11
        {% else %}
          light.bseed_hall_12
        {% endif %}
mode: single
```

rizosso, 2024-06-25:
Hi there, I’m currently experiencing a similar problem with my Bseed light switches. There is no finding available in zigbee2mqtt. Automations do not always work so I would like to No, how the people here are handling this issue?

mayongze, 2024-07-03:
@olili https://github.com/Anonym-tsk/zigbee2mqtt-extensions

ARoussakov, 2025-04-20:
Also looking the this option. Found just this: https://smarthome.yoavmor.com/home-assistant/synchronizing-two-switches-in-home-assistant/