Koenkk/zigbee2mqtt #13328: Tuya TS0201 Reporting Humidity off by factor of 10

dfiore1230, 2022-07-29:
### What happened?

Connected a Tuya TS0201 and watch reporting. Temperature is correct, Humidity displays 47.3% (on screen) but reports 4.73 via MQTT

log/2022-07-21.23-42-25/log.txt:info  2022-07-29 11:28:22: MQTT publish: topic 'zigbee2mqtt/Primary Bathroom Sensor', payload '{"battery":100,"humidity":4.73,"linkquality":79,"temperature":22.33,"voltage":3000}'

[image]

### What did you expect to happen?

Reporting matches display

### How to reproduce it (minimal and precise)

Add device to network, observe logs, front-end and device display

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

0x26680700

### Adapter

ConbeeII

Koenkk, 2022-07-30:
Can you provide the data/database.db entry of your device?

kirovilya, 2022-07-31:
Use humidity10 converter https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L512

dfiore1230, 2022-08-01:
Here is the database.db entry

`{"id":25,"type":"EndDevice","ieeeAddr":"0xa4c138c788411dca","nwkAddr":30887,"manufId":4417,"manufName":"_TZ3000_itnrsufe","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,57346,0],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011z\u0000\u0000\u0000\u0012","65506":51,"65508":1,"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_itnrsufe","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2251}},"msRelativeHumidity":{"attributes":{"measuredValue":501}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":192,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":126974596},"lastSeen":1659357364608,"defaultSendRequestWhen":"immediate"}`

I wouldnt know where to start using humidity10, but reading the code quickly, it seems to me to be dividing by 10, where as this needs to be multiplied by 10.

kirovilya, 2022-08-01:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L3023
Here I have the same device as you and the required converter is used there. wait for today's release and everything will work

dfiore1230, 2022-08-01:
Many thanks.

closed 2022-08-01 by Koenkk