Koenkk/zigbee2mqtt #13329: Negative Power values into the millions | Develco SPLZB-134 Power Plug (Type G)

sphinxicus, 2022-07-29:
### What happened?

Develco SPLZB-134 Power Plug (Type G)  is showing negative values for power when checking history charts in Home assistant.

The negative values that happen without any time period pattern are always the same and always come in clusters.

Power always shows the following value in the zigbee2mqtt logs -8388608 followed by -32768 

Voltage values are also incorrect at the same time and always show value of 655.35 when they should be showing circa the 230 range.

### What did you expect to happen?

not to receive negative values as i am not feeding 8 million W into the grid from my fridge.

### How to reproduce it (minimal and precise)

not possible to reproduce on demand but happens a few times a day - I enabled herdsman debug logging as well as debug in the zigbee2mqtt UI but i do not know where the herdsman debug log is located.

Environment Details

Version 	core-2022.7.7
Installation Type 	Home Assistant OS
Development 	false
Supervisor 	true
Docker 	true
Host Operating System 	Home Assistant OS 8.4
Update Channel 	stable
Supervisor Version 	supervisor-2022.07.0
Agent Version 	1.2.1
Docker Version 	20.10.14

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220219

### Adapter

CC2652P SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

```
debug 2022-07-29 11:37:15: Received Zigbee message from 'Kitchen power Point PM', type 'read', cluster 'genTime', data '["timeStatus","time","timeZone","dstStart","dstEnd","dstShift"]' from endpoint 2 with groupID 0
info  2022-07-29 11:37:15: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.01,"device_temperature":33,"energy":6.31,"last_seen":"2022-07-29T10:37:15.075Z","linkquality":91,"power":2,"state":"ON","voltage":234.84}'
debug 2022-07-29 11:37:15: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,6310],"instantaneousDemand":-8388608}' from endpoint 2 with groupID 0
info  2022-07-29 11:37:15: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.01,"device_temperature":33,"energy":6.31,"last_seen":"2022-07-29T10:37:15.135Z","linkquality":91,"power":-8388608,"state":"ON","voltage":234.84}'
debug 2022-07-29 11:37:15: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":-32768,"rmsCurrent":65535,"rmsVoltage":65535}' from endpoint 2 with groupID 0
info  2022-07-29 11:37:15: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":65.54,"device_temperature":33,"energy":6.31,"last_seen":"2022-07-29T10:37:15.147Z","linkquality":94,"power":-32768,"state":"ON","voltage":655.35}'
debug 2022-07-29 11:37:15: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
debug 2022-07-29 11:37:15: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":-32768}' from endpoint 2 with groupID 0
info  2022-07-29 11:37:15: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":65.54,"device_temperature":33,"energy":6.31,"last_seen":"2022-07-29T10:37:15.160Z","linkquality":91,"power":-32768,"state":"ON","voltage":655.35}'
info  2022-07-29 11:37:15: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":65.54,"device_temperature":-32768,"energy":6.31,"last_seen":"2022-07-29T10:37:15.160Z","linkquality":91,"power":-32768,"state":"ON","voltage":655.35}'
```

Koenkk, 2022-07-30:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // SPLZB-134 reports strange values sometimes 
    // https://github.com/Koenkk/zigbee2mqtt/issues/13329
    SPLZB134_electrical_measurement: {
        ...fz.electrical_measurement.cluster,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.rmsVoltage !== 0xFFFF && msg.data.rmsCurrent !== 0xFFFF && msg.data.activePower !== -0x8000) {
                return fz.electrical_measurement.convert(model, msg, publish, options, meta);
            }
        },
    },
    SPLZB134_device_temperature: {
        ...fz.device_temperature.cluster,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.currentTemperature !== -0x8000) {
                return fz.device_temperature.convert(model, msg, publish, options, meta);
            }
        },
    }
}

const definition = {
    zigbeeModel: ['SPLZB-134'],
    model: 'SPLZB-134',
    vendor: 'Develco',
    description: 'Power plug (type G)',
    fromZigbee: [fz.on_off, fzLocal.SPLZB134_electrical_measurement, fz.metering, fzLocal.SPLZB134_device_temperature],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.device_temperature()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering', 'genDeviceTempCfg']);
        await reporting.onOff(endpoint);
        await reporting.deviceTemperature(endpoint, {change: 2}); // Device temperature reports with 2 degree change
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.activePower(endpoint, {change: 10}); // Power reports with every 10W change
        await reporting.rmsCurrent(endpoint, {change: 20}); // Current reports with every 20mA change
        await reporting.rmsVoltage(endpoint, {min: constants.repInterval.MINUTES_5, change: 400}); // Limit reports to every 5m, or 4V
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint, {change: [0, 20]}); // Limit reports to once every 5m, or 0.02kWh
        await reporting.instantaneousDemand(endpoint, {min: constants.repInterval.MINUTES_5, change: 10});
    },
    endpoint: (device) => {
        return {default: 2};
    },
};

module.exports = definition;
```
- start z2m, check if issue is fixed

sphinxicus, 2022-08-01:
Just confirming that the below steps to add this are correct;

Create file named SPLZB_134.js in config/zigbee2mqtt/
add the above congifig to that file and save
in z2m UI click settings > external converters
add SPLZB_134.js > click submit
restart z2m

i have done the above steps and z2m started successfully - I will moniotor for the next few days

Thanks!

Koenkk, 2022-08-01:
Great, let me know then I will integrate the changes.

sphinxicus, 2022-08-03:
Hi @Koenkk 

Since implementing the external converter detailed above, I have not had any further instances of the negative values mentioned in my original report. Typically, i did have two clear days where this issue did not occur prior to implementing the external converter.

So either i wouldnt have seen the issue, or what i assume is some sort of "filter" in the above code to pass valid values and drop the decimal equivilent of the hex values (0xFFFF or -0x8000)  is doing its job.

I have not had any side effects from running the above code so either way i guess you could call it a success :)

sphinxicus, 2022-08-03:
I spoke too soon. I just spotted the same issue again when checking the history chart for the plug.

`debug 2022-08-03 21:25:04: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,19249],"instantaneousDemand":-8388608}' from endpoint 2 with groupID 0
info  2022-08-03 21:25:04: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":1.02,"device_temperature":34,"energy":19.25,"last_seen":"2022-08-03T20:25:04.985Z","linkquality":91,"power":-8388608,"state":"ON","voltage":237.97}'
debug 2022-08-03 21:25:04: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":-32768,"rmsCurrent":65535,"rmsVoltage":65535}' from endpoint 2 with groupID 0
debug 2022-08-03 21:25:04: No converter available for 'SPLZB-134' with cluster 'haElectricalMeasurement' and type 'attributeReport' and data '{"activePower":-32768,"rmsCurrent":65535,"rmsVoltage":65535}'
info  2022-08-03 21:25:04: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":1.02,"device_temperature":34,"energy":19.25,"last_seen":"2022-08-03T20:25:04.994Z","linkquality":91,"power":-8388608,"state":"ON","voltage":237.97}'
debug 2022-08-03 21:25:05: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
info  2022-08-03 21:25:05: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":1.02,"device_temperature":34,"energy":19.25,"last_seen":"2022-08-03T20:25:05.090Z","linkquality":91,"power":-8388608,"state":"ON","voltage":237.97}'
debug 2022-08-03 21:25:05: Received Zigbee message from 'Kitchen power Point PM', type 'read', cluster 'genTime', data '["timeStatus","time","timeZone","dstStart","dstEnd","dstShift"]' from endpoint 2 with groupID 0
info  2022-08-03 21:25:05: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":1.02,"device_temperature":34,"energy":19.25,"last_seen":"2022-08-03T20:25:05.243Z","linkquality":91,"power":-8388608,"state":"ON","voltage":237.97}'
debug 2022-08-03 21:25:05: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,19249],"instantaneousDemand":114}' from endpoint 2 with groupID 0
info  2022-08-03 21:25:05: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":1.02,"device_temperature":34,"energy":19.25,"last_seen":"2022-08-03T20:25:05.538Z","linkquality":91,"power":114,"state":"ON","voltage":237.97}'
debug 2022-08-03 21:25:05: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":114,"rmsCurrent":624,"rmsVoltage":24024}' from endpoint 2 with groupID 0
debug 2022-08-03 21:25:05: No converter available for 'SPLZB-134' with cluster 'haElectricalMeasurement' and type 'attributeReport' and data '{"activePower":114,"rmsCurrent":624,"rmsVoltage":24024}'`

Koenkk, 2022-08-04:
This should fix it:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // SPLZB-134 reports strange values sometimes 
    // https://github.com/Koenkk/zigbee2mqtt/issues/13329
    SPLZB134_electrical_measurement: {
        ...fz.electrical_measurement,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.rmsVoltage !== 0xFFFF && msg.data.rmsCurrent !== 0xFFFF && msg.data.activePower !== -0x8000) {
                return fz.electrical_measurement.convert(model, msg, publish, options, meta);
            }
        },
    },
    SPLZB134_device_temperature: {
        ...fz.device_temperature,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.currentTemperature !== -0x8000) {
                return fz.device_temperature.convert(model, msg, publish, options, meta);
            }
        },
    },
    SPLZB134_metering: {
        ...fz.metering,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.instantaneousDemand !== -0x8000) {
                return fz.metering.convert(model, msg, publish, options, meta);
            }
        },
    },
};

const definition = {
    zigbeeModel: ['SPLZB-134'],
    model: 'SPLZB-134',
    vendor: 'Develco',
    description: 'Power plug (type G)',
    fromZigbee: [fz.on_off, fzLocal.SPLZB134_electrical_measurement, fz.SPLZB134_metering, fzLocal.SPLZB134_device_temperature],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.device_temperature()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering', 'genDeviceTempCfg']);
        await reporting.onOff(endpoint);
        await reporting.deviceTemperature(endpoint, {change: 2}); // Device temperature reports with 2 degree change
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.activePower(endpoint, {change: 10}); // Power reports with every 10W change
        await reporting.rmsCurrent(endpoint, {change: 20}); // Current reports with every 20mA change
        await reporting.rmsVoltage(endpoint, {min: constants.repInterval.MINUTES_5, change: 400}); // Limit reports to every 5m, or 4V
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint, {change: [0, 20]}); // Limit reports to once every 5m, or 0.02kWh
        await reporting.instantaneousDemand(endpoint, {min: constants.repInterval.MINUTES_5, change: 10});
    },
    endpoint: (device) => {
        return {default: 2};
    },
};

module.exports = definition;
```

sphinxicus, 2022-08-04:
Thanks. I'll edit my SPLZB_134.js, restart z2m and I'll test for a few extra days this time.

sphinxicus, 2022-08-04:
> This should fix it:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = [...]

I tried implementing this but z2m fails to start with the following error:

`TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)
[17:37:37] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js`

Reverting back to the original external converter code works again

Koenkk, 2022-08-04:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/13329#issuecomment-1205405341

sphinxicus, 2022-08-05:
z2m is still throwing this error in the logs even with the updated code

```
[14:12:48] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)
```

sphinxicus, 2022-08-05:
Still seeing that negative value.

Started z2m with the latest external connecter code at 14:17

```
debug 2022-08-05 14:17:08: Loaded state from file /config/zigbee2mqtt/state.json
info  2022-08-05 14:17:08: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-05.14-17-08' filename: log.txt
debug 2022-08-05 14:17:08: Removing old log directory '/config/zigbee2mqtt/log/2022-08-05.14-16-16'
info  2022-08-05 14:17:09: Starting Zigbee2MQTT version 1.26.0 (commit #unknown)
info  2022-08-05 14:17:09: Starting zigbee-herdsman (0.14.40)
```

Same negative value captured again shortly after restart at 14:27:14

[image]

```
debug 2022-08-05 14:27:14: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,22799],"instantaneousDemand":-8388608}' from endpoint 2 with groupID 0
info  2022-08-05 14:27:14: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.71,"device_temperature":34,"energy":22.8,"last_seen":"2022-08-05T13:27:14.079Z","linkquality":91,"power":-8388608,"state":"ON","voltage":238.57}'
debug 2022-08-05 14:27:14: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":-32768,"rmsCurrent":65535,"rmsVoltage":65535}' from endpoint 2 with groupID 0
info  2022-08-05 14:27:14: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.71,"device_temperature":34,"energy":22.8,"last_seen":"2022-08-05T13:27:14.085Z","linkquality":94,"power":-8388608,"state":"ON","voltage":238.57}'
debug 2022-08-05 14:27:25: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":130,"rmsCurrent":736,"rmsVoltage":23829}' from endpoint 2 with groupID 0
info  2022-08-05 14:27:25: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.74,"device_temperature":34,"energy":22.8,"last_seen":"2022-08-05T13:27:25.403Z","linkquality":91,"power":130,"state":"ON","voltage":238.29}'
info  2022-08-05 14:27:35: MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"battery_low":false,"check_meter":false,"current_summation":1000,"energy":351.73,"interface_mode":"electricity","last_seen":"2022-08-05T13:27:35.711Z","linkquality":94,"power":1639,"pulse_configuration":1000}'
debug 2022-08-05 14:27:36: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,22801],"instantaneousDemand":133}' from endpoint 2 with groupID 0
info  2022-08-05 14:27:36: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.74,"device_temperature":34,"energy":22.8,"last_seen":"2022-08-05T13:27:36.296Z","linkquality":94,"power":133,"state":"ON","voltage":238.29}'
debug 2022-08-05 14:27:36: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":133,"rmsCurrent":752,"rmsVoltage":23669}' from endpoint 2 with groupID 0
info  2022-08-05 14:27:36: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.75,"device_temperature":34,"energy":22.8,"last_seen":"2022-08-05T13:27:36.308Z","linkquality":94,"power":133,"state":"ON","voltage":236.69}'
debug 2022-08-05 14:27:41: Received Zigbee message from 'Kitchen power Point PM', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":133,"rmsCurrent":757,"rmsVoltage":23697}' from endpoint 2 with groupID 0
info  2022-08-05 14:27:41: MQTT publish: topic 'zigbee2mqtt/Kitchen power Point PM', payload '{"current":0.76,"device_temperature":34,"energy":22.8,"last_seen":"2022-08-05T13:27:41.340Z","linkquality":94,"power":133,"state":"ON","voltage":236.97}'

```

Koenkk, 2022-08-06:
I found the issues, added to little zeros on line 34:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // SPLZB-134 reports strange values sometimes 
    // https://github.com/Koenkk/zigbee2mqtt/issues/13329
    SPLZB134_electrical_measurement: {
        ...fz.electrical_measurement,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.rmsVoltage !== 0xFFFF && msg.data.rmsCurrent !== 0xFFFF && msg.data.activePower !== -0x8000) {
                return fz.electrical_measurement.convert(model, msg, publish, options, meta);
            }
        },
    },
    SPLZB134_device_temperature: {
        ...fz.device_temperature,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.currentTemperature !== -0x8000) {
                return fz.device_temperature.convert(model, msg, publish, options, meta);
            }
        },
    },
    SPLZB134_metering: {
        ...fz.metering,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.instantaneousDemand !== -0x800000) {
                return fz.metering.convert(model, msg, publish, options, meta);
            }
        },
    },
};

const definition = {
    zigbeeModel: ['SPLZB-134'],
    model: 'SPLZB-134',
    vendor: 'Develco',
    description: 'Power plug (type G)',
    fromZigbee: [fz.on_off, fzLocal.SPLZB134_electrical_measurement, fzLocal.SPLZB134_metering, fzLocal.SPLZB134_device_temperature],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.device_temperature()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering', 'genDeviceTempCfg']);
        await reporting.onOff(endpoint);
        await reporting.deviceTemperature(endpoint, {change: 2}); // Device temperature reports with 2 degree change
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.activePower(endpoint, {change: 10}); // Power reports with every 10W change
        await reporting.rmsCurrent(endpoint, {change: 20}); // Current reports with every 20mA change
        await reporting.rmsVoltage(endpoint, {min: constants.repInterval.MINUTES_5, change: 400}); // Limit reports to every 5m, or 4V
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint, {change: [0, 20]}); // Limit reports to once every 5m, or 0.02kWh
        await reporting.instantaneousDemand(endpoint, {min: constants.repInterval.MINUTES_5, change: 10});
    },
    endpoint: (device) => {
        return {default: 2};
    },
};

module.exports = definition;
```

sphinxicus, 2022-08-06:
thanks! I did spot that none of the hex numbers converted to decimal -8388608 but i didn't question it as i was guessing at the function of the code :)

I have apllied the above and restarted z2m without issues. Will test for a few days and revert. Thanks again.

sphinxicus, 2022-08-12:
I've ran this config for almost a week now and have no further issues to report.

Thanks!

Koenkk, 2022-08-13:
Great, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-13 by Koenkk

middelink, 2022-09-16:
FTR, It seems the SPLZB-131 and SPLZB-132 suffer from the same problem. https://github.com/Koenkk/zigbee-herdsman-converters/pull/4616 fixes SPLZB-132.

Koenkk, 2022-09-19:
@middelink pushed the same fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

PVi1, 2022-11-18:
Hi all,
the similar fix is needed for AQSZB-110.
From time to time it returns temperature of -32768 or voc 65535.

I have tried to create external converter with the following code, but unfortunatelly zigbee2mqtt docker won't start anymore.
The error is:
`/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)
Using '/app/data' as data directory
`
Converter looks like this:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // AQSZB-110 reports strange values sometimes
    AQSZB110_temperature: {
        ...fz.temperature,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.measuredValue !== -0x8000) {
               return fz.temperature.convert(model, msg, publish, options, meta);
            }
        },
    },
    AQSZB110_voc: {
        ...fz.develco_voc,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.measuredValue !== 0xFFFF) {
                return fz.develco_voc.convert(model, msg, publish, options, meta);
            }
        },
    },
};

const definition = {
	zigbeeModel: ['AQSZB-110'],
        model: 'AQSZB-110',
        vendor: 'Develco',
        description: 'Air quality sensor',
        fromZigbee: [fz.develco_voc_battery, fzLocal.AQSZB110_voc, fzLocal.AQSZB110_temperature, fz.humidity],
        toZigbee: [],
        exposes: [
            e.voc(), e.temperature(), e.humidity(),
            e.battery(), e.battery_low(),
            exposes.enum('air_quality', ea.STATE, [
                'excellent', 'good', 'moderate',
                'poor', 'unhealthy', 'out_of_range',
                'unknown']).withDescription('Measured air quality'),
        ],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(38);
            const options = {manufacturerCode: 0x1015};
            await reporting.bind(endpoint, coordinatorEndpoint,
                ['develcoSpecificAirQuality', 'msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg']);
            await endpoint.configureReporting('develcoSpecificAirQuality', [{attribute: 'measuredValue', minimumReportInterval: 60,
                maximumReportInterval: 3600, reportableChange: 10}], options);
            await reporting.temperature(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 10});
            await reporting.humidity(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 300});
            await reporting.batteryVoltage(endpoint, {min: constants.repInterval.HOUR, max: 43200, change: 100});
        },
};

module.exports = definition;
`

Can you spot the error?
As a last resort I went ahead and modified the develco.js converter directly in zigbee2mqtt docker container and it works fine.

Can you add these fixes to develco.js please?

Thanks!

Koenkk, 2022-11-20:
Done! (for all develco devices)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)