Koenkk/zigbee2mqtt #13330: [New device support]: Vimar 03981 connected relay module

stefa168, 2022-07-29:
### Link

https://www.vimar.com/en/int/catalog/product/index/code/03981

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0xf4ce360a9db9369e","nwkAddr":26319,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"On_Off_Switch_Module_v1.0","epList":[10,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":0,"inClusterList":[6,0,3,7,5,4],"outClusterList":[6,3],"clusters":{"genBasic":{"attributes":{"modelId":"On_Off_Switch_Module_v1.0","manufacturerName":"Vimar","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200501","swBuildId":""}},"genOnOff":{"attributes":{"onOff":1}},"genIdentify":{"attributes":{"identifyTime":0}},"genOnOffSwitchCfg":{"attributes":{"switchType":0}},"genScenes":{"attributes":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200501","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1659114685886,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, today I received this Vimar 03981 module.

I have managed to get the relay to switch with an external converter, however I'm not sure how should I implement the other converters.

For example, by looking at the clusters list it seems like the device supports `genIdentify` both for output and input, however I'm not sure how should I implement the converter for the `toZigbee` entry. Same thing for the other clusters...

Any help would be greatly appreciated! Thanks!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['On_Off_Switch_Module_v1.0'],
    model: '03981',
    vendor: 'Vimar',
    description: 'IoT connected relay module',
    fromZigbee: [fz.on_off, fz.identify],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;
```

Koenkk, 2022-07-30:
Added it, thanks.

> For example, by looking at the clusters list it seems like the device supports genIdentify both for output and input, however I'm not sure how should I implement the converter for the toZigbee entry. Same thing for the other clusters...

Since this is just a simple on/off device, we only want to expose the on/off functionality so your converter looks good.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

stefa168, 2022-07-31:
> Since this is just a simple on/off device, we only want to expose the on/off functionality so your converter looks good.

Ok I see, I was asking because from the device documentation it seems like it is possibile to set the switch to work in two modes (monostable or bistable, with bistable being the default) and set the time it is on, however it looks like these values aren't exposed by the device...

Thanks again!

closed 2022-07-31 by stefa168