Koenkk/zigbee2mqtt #13333: Battery is always 100%

SzemiSun, 2022-07-30:
### What happened?

The battery value of my xiaomi sensors with batteries always shows 100%, although the battery_voltage should be less

### What did you expect to happen?

Displays the correct value as a percentage

battery voltage is 3055 in my case, the voltage range is 2850 to 3200 according to source code:

this means that 
    const normalised = (3055 - 2850) / (3200 - 2850);
    return Math.round(normalised * 100);

the result should be 58, but it's actually 100, see below the log.

### How to reproduce it (minimal and precise)

I put a submerged battery in the sensor

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

Silicon_Labs_slae.sh_cc2652rb_stick

### Debug log

debug 2022-07-30 11:25:58: Received Zigbee message from 'ht_nappali', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3055,"10":27811,"100":2660,"101":6110,"4":5032,"5":1073,"6":[0,1]}}' from endpoint 1 with groupID 0
debug 2022-07-30 11:25:58: lumi.sens,lumi.sensor_ht: unknown key 6 with value 0,1
debug 2022-07-30 11:25:58: lumi.sens,lumi.sensor_ht: Processed data into payload {"voltage":3055,"battery":100,"power_outage_count":1072,"temperature":26.6,"humidity":61.1}
debug 2022-07-30 11:25:58: lumi.sens,lumi.sensor_ht: Processed data into payload {"voltage":3055,"battery":100,"power_outage_count":1072,"temperature":26.6,"humidity":61.1}
info  2022-07-30 11:25:58: MQTT publish: topic 'zigbee2mqtt/ht_nappali', payload '{"battery":100,"humidity":61.1,"last_seen":"2022-07-30T11:25:58+02:00","linkquality":58,"power_outage_count":1072,"temperature":26.6,"voltage":3055}'
debug 2022-07-30 11:25:58: Received Zigbee message from 'ht_nappali', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sensor_ht"}' from endpoint 1 with groupID 0
debug 2022-07-30 11:25:58: lumi.sens,lumi.sensor_ht: Processed data into payload {}
info  2022-07-30 11:25:58: MQTT publish: topic 'zigbee2mqtt/ht_nappali', payload '{"battery":100,"humidity":61.1,"last_seen":"2022-07-30T11:25:58+02:00","linkquality":54,"power_outage_count":1072,"temperature":26.6,"voltage":3055}'
debug 2022-07-30 11:26:03: Received Zigbee message from 'ht_nappali', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2674}' from endpoint 1 with groupID 0
info  2022-07-30 11:26:03: MQTT publish: topic 'zigbee2mqtt/ht_nappali', payload '{"battery":100,"humidity":61.1,"last_seen":"2022-07-30T11:26:03+02:00","linkquality":58,"power_outage_count":1072,"temperature":26.7,"voltage":3055}'
debug 2022-07-30 11:26:03: Received Zigbee message from 'ht_nappali', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6821}' from endpoint 1 with groupID 0
info  2022-07-30 11:26:03: MQTT publish: topic 'zigbee2mqtt/ht_nappali', payload '{"battery":100,"humidity":68.2,"last_seen":"2022-07-30T11:26:03+02:00","linkquality":58,"power_outage_count":1072,"temperature":26.7,"voltage":3055}'
debug 2022-07-30 11:26:24: Received Zigbee message from 'ht_nappali', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2689}' from endpoint 1 with groupID 0
info  2022-07-30 11:26:24: MQTT publish: topic 'zigbee2mqtt/ht_nappali', payload '{"battery":100,"humidity":68.2,"last_seen":"2022-07-30T11:26:24+02:00","linkquality":54,"power_outage_count":1072,"temperature":26.9,"voltage":3055}'
debug 2022-07-30 11:26:24: Received Zigbee message from 'ht_nappali', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6146}' from endpoint 1 with groupID 0
info  2022-07-30 11:26:24: MQTT publish: topic 'zigbee2mqtt/ht_nappali', payload '{"battery":100,"humidity":61.5,"last_seen":"2022-07-30T11:26:24+02:00","linkquality":58,"power_outage_count":1072,"temperature":26.9,"voltage":3055}'
debug

Koenkk, 2022-07-30:
> battery voltage is 3055 in my case, the voltage range is 2850 to 3200 according to source code:

This calculation is not in 1.26.0, but only in the dev branch (and will be in the next release on 1 August)

SzemiSun, 2022-07-30:
Ohh, You are right, I missed that, sorry

closed 2022-07-30 by Koenkk