Koenkk/zigbee2mqtt #13335: IKEA E2001/E2002 STYRBAR remote controller - incorrect long arrows press recognition

JackTalisker, 2022-07-30:
### What happened?

Long press left or right arrow fires the "on" payload instead of arrow_left_hold and arrow_right_hold payloads

### What did you expect to happen?

To have the arrow_left_hold and arrow_right_hold payloads

### How to reproduce it (minimal and precise)

long press left or right arrow

### Zigbee2MQTT version

1.26.0 commit: cd23aea

### Adapter firmware version

20200708 - 1.0.024

### Adapter

IKEA E2001/E2002

### Debug log

```
Debug 2022-07-30 14:46:05Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 14:46:05MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-30T12:46:05.223Z","linkquality":204,"update":{"state":null},"update_available":null}'
Debug 2022-07-30 14:46:05Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 14:46:05MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-30T12:46:05.242Z","linkquality":208,"update":{"state":null},"update_available":null}'
Debug 2022-07-30 14:46:05Received Zigbee message from 'styr2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 14:46:05MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":"on","battery":90,"last_seen":"2022-07-30T12:46:05.723Z","linkquality":204,"update":{"state":null},"update_available":null}'
Info 2022-07-30 14:46:05MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":"","battery":90,"last_seen":"2022-07-30T12:46:05.723Z","linkquality":204,"update":{"state":null},"update_available":null}'
Info 2022-07-30 14:46:05MQTT publish: topic 'zigbee2mqtt/styr2/action', payload 'on'
Debug 2022-07-30 14:46:05Received Zigbee message from 'styr2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 14:46:05MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-30T12:46:05.739Z","linkquality":248,"update":{"state":null},"update_available":null}'
Debug 2022-07-30 14:46:06Received Zigbee message from 'styr2', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-30 14:46:06MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-30T12:46:06.267Z","linkquality":208,"update":{"state":null},"update_available":null}'
Debug 2022-07-30 14:46:06Received Zigbee message from 'styr2', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-30 14:46:06MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-30T12:46:06.286Z","linkquality":208,"update":{"state":null},"update_available":null}'

Debug 2022-07-30 14:59:26Received Zigbee message from 'styr1', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 14:59:26MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T12:59:26.109Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 14:59:26Received Zigbee message from 'styr1', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 14:59:26MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T12:59:26.147Z","linkquality":244,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 14:59:26Received Zigbee message from 'styr1', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 14:59:26MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"on","battery":85,"last_seen":"2022-07-30T12:59:26.556Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 14:59:26MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"","battery":85,"last_seen":"2022-07-30T12:59:26.556Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 14:59:26MQTT publish: topic 'zigbee2mqtt/styr1/action', payload 'on'
Debug 2022-07-30 14:59:26Received Zigbee message from 'styr1', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 14:59:26MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T12:59:26.591Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
```

JackTalisker, 2022-07-30:
This is the same issue as #9867 since it was flagged staled and then closed.

Koenkk, 2022-07-30:
What `Received Zigbee message from` is from the left vs right press?

JackTalisker, 2022-07-30:
```
Left arrow
Debug 2022-07-30 16:31:50Received Zigbee message from 'styr1', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 16:31:50MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:31:50.751Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 16:31:50Received Zigbee message from 'styr1', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 16:31:50MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:31:50.777Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 16:31:51Received Zigbee message from 'styr1', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 16:31:51MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"on","battery":85,"last_seen":"2022-07-30T14:31:51.206Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:31:51MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"","battery":85,"last_seen":"2022-07-30T14:31:51.206Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:31:51MQTT publish: topic 'zigbee2mqtt/styr1/action', payload 'on'
Debug 2022-07-30 16:31:51Received Zigbee message from 'styr1', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 16:31:51MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:31:51.231Z","linkquality":164,"update":{"state":"idle"},"update_available":false}'

Right arrow
Debug 2022-07-30 16:32:44Received Zigbee message from 'styr1', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 16:32:44MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:32:43.816Z","linkquality":164,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 16:32:44Received Zigbee message from 'styr1', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-30 16:32:44MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:32:43.839Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 16:32:44Received Zigbee message from 'styr1', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 16:32:44MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"on","battery":85,"last_seen":"2022-07-30T14:32:44.264Z","linkquality":164,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:32:44MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"","battery":85,"last_seen":"2022-07-30T14:32:44.264Z","linkquality":164,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:32:44MQTT publish: topic 'zigbee2mqtt/styr1/action', payload 'on'
Debug 2022-07-30 16:32:44Received Zigbee message from 'styr1', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-30 16:32:44MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:32:44.284Z","linkquality":164,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 16:32:45Received Zigbee message from 'styr1', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-30 16:32:45MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:32:44.832Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-30 16:32:45Received Zigbee message from 'styr1', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-30 16:32:45MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:32:44.969Z","linkquality":168,"update":
```

JackTalisker, 2022-07-30:
Don't know if could be helpful but the short arrow press (which works fawlessly) behave as follows:

```
Left short press
Debug 2022-07-30 16:38:49Received Zigbee message from 'styr1', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
Info 2022-07-30 16:38:49MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"arrow_left_click","battery":85,"last_seen":"2022-07-30T14:38:49.469Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:38:49MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"","battery":85,"last_seen":"2022-07-30T14:38:49.469Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:38:49MQTT publish: topic 'zigbee2mqtt/styr1/action', payload 'arrow_left_click'
Debug 2022-07-30 16:38:49Received Zigbee message from 'styr1', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
Info 2022-07-30 16:38:49MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:38:49.501Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'

Right short press
Debug 2022-07-30 16:39:19Received Zigbee message from 'styr1', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
Info 2022-07-30 16:39:19MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"arrow_right_click","battery":85,"last_seen":"2022-07-30T14:39:19.063Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:39:19MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":"","battery":85,"last_seen":"2022-07-30T14:39:19.063Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-30 16:39:19MQTT publish: topic 'zigbee2mqtt/styr1/action', payload 'arrow_right_click'
Debug 2022-07-30 16:39:19Received Zigbee message from 'styr1', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
Info 2022-07-30 16:39:19MQTT publish: topic 'zigbee2mqtt/styr1', payload '{"action":null,"battery":85,"last_seen":"2022-07-30T14:39:19.096Z","linkquality":168,"update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2022-07-31:
@JackTalisker for the right hold I see different output as for the left hold.

Left: commandTradfriArrowRelease -> commandTradfriArrowRelease -> commandOn -> commandOn
Right: commandTradfriArrowRelease -> commandTradfriArrowRelease -> commandOn -> commandOn -> **commandTradfriArrowSingle -> commandTradfriArrowSingle**

Can you confirm that you triggered identical actions for both buttons?

JackTalisker, 2022-07-31:
Messages change on button pressing duration.

These are the messages for left & right arrows for 0.5, 1 and 2 seconds:

```
Left arrow for approx 0.5 second
Debug 2022-07-31 12:20:00Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:20:00MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:20:00.938Z","linkquality":180,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:20:00Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:20:00MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:20:00.958Z","linkquality":180,"update":{"state":"idle"},"update_available":false}'

Left arrow for approx 1 second
Debug 2022-07-31 12:09:59Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:09:59MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:09:59.881Z","linkquality":192,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:09:59Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:09:59MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:09:59.922Z","linkquality":192,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:10:00Received Zigbee message from 'styr2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:00MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":"on","battery":90,"last_seen":"2022-07-31T10:10:00.375Z","linkquality":192,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-31 12:10:00MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":"","battery":90,"last_seen":"2022-07-31T10:10:00.375Z","linkquality":192,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-31 12:10:00MQTT publish: topic 'zigbee2mqtt/styr2/action', payload 'on'
Debug 2022-07-31 12:10:00Received Zigbee message from 'styr2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:00MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:10:00.399Z","linkquality":192,"update":{"state":"idle"},"update_available":false}'

Left arrow for approx 2 seconds
Debug 2022-07-31 12:10:15Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Debug 2022-07-31 12:10:15Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:15MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:10:15.200Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-31 12:10:15MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:10:15.200Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:10:15Received Zigbee message from 'styr2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:15MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":"on","battery":90,"last_seen":"2022-07-31T10:10:15.608Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-31 12:10:15MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":"","battery":90,"last_seen":"2022-07-31T10:10:15.608Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'
Info 2022-07-31 12:10:15MQTT publish: topic 'zigbee2mqtt/styr2/action', payload 'on'
Debug 2022-07-31 12:10:15Received Zigbee message from 'styr2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:15MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:10:15.637Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:10:16Received Zigbee message from 'styr2', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:16MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:10:16.152Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:10:16Received Zigbee message from 'styr2', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:10:16MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:10:16.169Z","linkquality":184,"update":{"state":"idle"},"update_available":false}'

Right arrow for approx 0.5 second
Debug 2022-07-31 12:13:44Received Zigbee message from 'styr2', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
[... cut ...]
Info 2022-07-31 12:14:08MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:14:08.758Z","linkquality":188,"update":{"state":"idle"},"update_available":false}'
Debug 2022-07-31 12:14:08Received Zigbee message from 'styr2', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Info 2022-07-31 12:14:08MQTT publish: topic 'zigbee2mqtt/styr2', payload '{"action":null,"battery":90,"last_seen":"2022-07-31T10:14:08.800Z","linkquality":188,"update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2022-08-01:
Could you provide me the herdsman debug log of the `Right arrow for approx 2 seconds`?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

JackTalisker, 2022-08-01:
Here it is.
The device is "styr1" line 898 to 911
[z2m_herdsman.txt](https://github.com/Koenkk/zigbee2mqtt/files/9235984/z2m_herdsman.txt)

Koenkk, 2022-08-01:
I don't see the herdsman debug logging (not that this is logged to STDOUT only). 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

JackTalisker, 2022-08-02:
I added the _DEBUG=zigbee-herdsman*_ viariable to my docker-compose.yaml file then I retrieved the log with the _docker logs zigbee2mqtt > log.txt 2>&1_ command.
What should I see in the log to understand that the zigbee.hredsman debug is activated?

My docker-compose.yaml file z2m section looks like this:
```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
    - "/home/pi/zigbee2mqtt:/app/data"
    devices:
    #- "192.168.0.77:8888"
    - /dev/null:/dev/ttyACM0
    environment:
    - TZ=Europe/Rome
    - DEBUG=zigbee-herdsman*
    network_mode: host
    restart: always
```

Koenkk, 2022-08-02:
Did you do a `docker compose up` after editing the docker-compose file?

JackTalisker, 2022-08-03:
> Did you do a `docker compose up` after editing the docker-compose file?

Yes of course but I was working with the wrong config file.
Now the following should be the right log:

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9249013/log.txt)

Koenkk, 2022-08-03:
@JackTalisker hmm, I don't think we can distinguish the left/right button press. will buy a styr remote at ikea to debug further.

JackTalisker, 2022-08-03:
@Koenkk 
here is the right arrow log
[log_right.txt](https://github.com/Koenkk/zigbee2mqtt/files/9253085/log_right.txt)

Maarten112, 2022-08-04:
I discovered the same issue yesterday, altough I get the arrow_left_hold payload. However, There's always an ON action payload before an ' arrow_left_hold' action payload.

I'm not super familiair with how all zigbee things works in detail, but it seems like this is not a zigbee2mqtt issue but the way how the styrbar remote sents out their signals?

Here you can see a small log part of the behaviour I get when long pressing the left arrow and releasing it at the end:

`
Info 2022-08-04 17:55:02MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)', payload '{"action":"on","battery":null,"linkquality":70,"update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:02MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)', payload '{"action":"","battery":null,"linkquality":70,"update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:02MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)/action', payload 'on'
Info 2022-08-04 17:55:02MQTT publish: topic 'zigbee2mqtt/test', payload '{"brightness":146,"color_mode":"color_temp","color_temp":398,"state":"ON"}'
Info 2022-08-04 17:55:02MQTT publish: topic 'zigbee2mqtt/0x04cd15fffec3d4a2 (ikea lamp)', payload '{"brightness":146,"color_mode":"color_temp","color_temp":398,"linkquality":70,"power_on_behavior":"on","state":"ON","update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:03MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)', payload '{"action":"arrow_left_hold","battery":null,"linkquality":70,"update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:03MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)', payload '{"action":"","battery":null,"linkquality":70,"update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:03MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)/action', payload 'arrow_left_hold'
Info 2022-08-04 17:55:04MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)', payload '{"action":"arrow_left_release","action_duration":0.508,"battery":null,"duration":0.508,"linkquality":70,"update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:04MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)', payload '{"action":"","battery":null,"linkquality":70,"update":{"state":null},"update_available":null}'
Info 2022-08-04 17:55:04MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe80df33 (ikea schakelaar)/action', payload 'arrow_left_release'`

Koenkk, 2022-08-04:
Bought the styrbar and got the same result as @Maarten112 

This should be fixed with the following converter

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const fzLocal = {
    command_on: {
        cluster: 'genOnOff',
        type: 'commandOn',
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg)) return;
            const arrowReleaseAgo = Date.now() - globalStore.getValue(msg.endpoint, 'arrow_release', 0);
            console.log(arrowReleaseAgo);
            if (arrowReleaseAgo > 700) {
                return {action: 'on'};
            }
        },
    },
    ikea_arrow_release: {
        cluster: 'genScenes',
        type: 'commandTradfriArrowRelease',
        options: [exposes.options.legacy()],
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg)) return;
            globalStore.putValue(msg.endpoint, 'arrow_release', Date.now());
            const direction = globalStore.getValue(msg.endpoint, 'direction');
            if (direction) {
                globalStore.clearValue(msg.endpoint, 'direction');
                const duration = msg.data.value / 1000;
                const result = {action: `arrow_${direction}_release`, duration, action_duration: duration};
                if (!utils.isLegacyEnabled(options)) delete result.duration;
                return result;
            }
        },
    },
};

const definition = {
    zigbeeModel: ['Remote Control N2'],
    model: 'E2001/E2002',
    vendor: 'IKEA',
    description: 'STYRBAR remote control N2 cUSTOM',
    fromZigbee: [fz.battery, fzLocal.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.ikea_arrow_click,
        fz.ikea_arrow_hold, fzLocal.ikea_arrow_release],
    exposes: [e.battery(), e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'arrow_left_click', 'arrow_right_click', 'arrow_left_hold',
        'arrow_right_hold', 'arrow_left_release', 'arrow_right_release'])],
    toZigbee: [],
    ota: ota.tradfri,
    meta: {battery: {dontDividePercentage: true}},
    configure: async (device, coordinatorEndpoint, logger) => {
        // Binding genOnOff is not required to make device send events.
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

@JackTalisker can you also check if it fixes the issue for you? If not provide me the debug log of a: hold right button for 5 seconds; release

git-developer, 2022-08-04:
This converter offers a workaround for the problem within z2m - which is great and all that can be done here, so thank you, Koen!

When the remote is directly bound to a device (without going through z2m), the problem remains, of course. I'm still curious if someone is able to develop a custom firmware that fixes the remote instead of trying to cope with the impact of the buggy remote firmware.

TheCyberfreak, 2022-08-04:
@Koenkk for me it works with the customized converter!

JackTalisker, 2022-08-05:
> @JackTalisker can you also check if it fixes the issue for you? If not provide me the debug log of a: hold right button for 5 seconds; release

I found  that pressing the arrows for more than 2 seconds I get the same behaviour. After a very long time (at least 3 seconds) the payload _'arrow_xxxx_hold'_ appears.

So fixed! The external converter works. The unexpected payload 'on' is gone.

Thank you very much @Koenkk

Maarten112, 2022-08-05:
@Koenkk Works for me! Great!

Some sidenotes for people who will find this when searching for the same issue:

**when using zigbee2mqtt as a homeassistant addon:**
1) add ext_converter.js to /usr/share/hassio/homeassistant/zigbee2mqtt
2) open the zigbee2mqtt webgui and navigate to settings --> external converters. In the input field, add ' ext_converter.js ' and click on submit. 
3) restart the zigbee2mqtt addon and see Koen's magic working :) 

**when the styrbar remote is also binded to a single endpoint (bulb) or group**
Since the styrbar remote will always sent out an ' on ' paypload when longpressing < or > arrowbuttons, your endpoint(s) will turn on. @Koenkk's solution only filters the ' on ' payload on a software base, but the hardware (styrbar remote) will always sent it as a rf command anyway. 

Long pressing the < and > buttons will work fine in zigbee2mqtt (you will only see an ' arrow_left_hold' command, no need to filter it yourself anymore!), but be informed that your endpoints will turn on anyway when binded to the styrbar remote. 

If your situation can't use this behaviour, don't bind the styrbar to your bulbs and let your domotica controller software do all the logic. 

I hope this makes sense.

Koenkk, 2022-08-05:
Great I pushed the fix.

From an IKEA perspective the on command makes a lot of sense. The arrows are used to change the color/colortemp of a bulb, by sending an on command first Ikea ensures the bulbs are on when changing the colortemp/color. Otherwise nothing would happen when holding this button while the bulb is off.

closed 2022-08-05 by Koenkk

mauritsivs, 2022-11-14:
> Great I pushed the fix.

so does this mean i should not need that extra .js file to make this work? 

I recently added this device but its still not functioning with the long press left/right arrow

festoino, 2022-11-14:
> 1. ext_converter.js

It gives me this error:

Zigbee2MQTT:error 2022-11-14 14:55:15: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'meta')}

namadori, 2022-12-30:
I am having the same issue with error "Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'meta')}". Removing the custom converter restored the functionality. Maybe some updates of Z2M broke this code?

cristiang777, 2023-09-24:
I am dealing with the same error after adding the ext_converter.js. It worked for about 2 weeks without issues but now the STYRBAR's up/on button short press throws this error:

```
023-09-24 18:33:52Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'meta')}
Debug 2023-09-24 18:33:52TypeError: Cannot read properties of undefined (reading 'meta') at Object.hasAlreadyProcessedMessage (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:70:15) at Object.convert (/app/data/extension/externally-loaded.js:19:23) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:108:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:527:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:738:18) at EZSPAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
```

schaijik, 2023-12-02:
Did anyone ever find a permanent fix for this issue? Or should I just bin this thing and get some night's sleep :-D

lebaguette900, 2024-01-08:
Hi,

I too am pulling my hair on this. I got the same error as namadori and festoino. Any news?

After many hours of troubleshooting I am close to joining schaijik and chuck this remote in the trash. If I did, any recommendations for a solid alternative remote?

Thanks for everyone's efforts.

martin3000, 2024-08-28:
My IKEA E2002 sends the command sequence [release 0, on, press 2, hold] if I long press left or right. So yes, it sends "on", but before it sends "release 0". So you know that "release 0"->"on" can be ignored.