Koenkk/zigbee2mqtt #13337: Moes BRT-100-TRV Error: Invalid value for number *_max_temperature: 45 (range 15.0 - 35.0)

AdamoIV, 2022-07-30:
### What happened?

`Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:211
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 22:11:16 (1 occurrences)
Last logged: 22:11:16

Invalid value for number.loungeradiator_left_max_temperature: 45 (range 15.0 - 35.0)`

### What did you expect to happen?

No error.

### How to reproduce it (minimal and precise)

No user action, seems to happen on it's own. Over time the log file fills up with the same error.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220219

### Adapter

zStack3x0 [SONOFF Zigbee 3.0 Dongle Plus]

### Debug log

info  2022-07-30 22:11:16: MQTT publish: topic 'zigbee2mqtt/LoungeRadiator_Left', payload '{"battery":80,"boost_heating":"OFF","boost_heating_countdown":0,"boost_heating_countdown_time_set":300,"child_lock":"UNLOCK","current_heating_setpoint":16,"eco_mode":"OFF","eco_temperature":20,"last_seen":"2022-07-30T22:07:42+01:00","linkquality":94,"local_temperature":20,"local_temperature_calibration":-3,"max_temperature":45,"min_temperature":5,"position":50,"preset":"manual","programming_mode":"06:00/20°C  11:30/21°C  13:30/22°C  17:30/23°C  06:00/24°C  12:00/23°C  14:30/22°C  17:30/21°C  06:00/19°C  12:30/20°C  14:30/21°C  18:30/20°C","update":{"state":null},"update_available":null,"valve_state":"CLOSED","window":"CLOSED","window_detection":"OFF"}'

AdamoIV, 2022-07-30:
Seems similar to this issue [#9971](https://github.com/Koenkk/zigbee2mqtt/issues/9971)?

Koenkk, 2022-07-31:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-31 by Koenkk