Koenkk/zigbee2mqtt #13339: Comments not preserved in configuration.yaml

incontestableness, 2022-07-30:
### What happened?

I created a configuration.yaml with information about settings I chose as well as options I wasn't ready to try out yet, but after starting zigbee2mqtt, my configuration was overwritten and reformatted with two-space indentation.

### What did you expect to happen?

I expected that my configuration.yaml would not be rewritten, removing all of my comments and four-space indentation.

### How to reproduce it (minimal and precise)

Create a configuration.yaml with comments, then start zigbee2mqtt.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

N/A

### Adapter

N/A

Koenkk, 2022-07-31:
This is indeed not possible. The used YAML library doesn't preserve comments (and it turns out this is very difficult to do)

incontestableness, 2022-08-01:
Maybe add a notice to the docs alongside mentions of "GENERATE"?

closed 2022-09-09 by github-actions[bot]