Koenkk/zigbee2mqtt #13342: MCCGQ11LM not always reporting close/open

johannes-z, 2022-07-31:
### What happened?

Sometimes the [MCCGQ11LM](https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#xiaomi-mccgq11lm) doesn't trigger when opening/closing the window/door they're attached to. 

LQI is good for all contacts, even the one failing to report, around 150. In the network map they report 0 however. What's new is that previously paired devices don't show a link in the Map anymore, only when connected to Routers:
[image]

I cannot see any log entries for failed attempts.

### What did you expect to happen?

Open/Close should be reported reliably every time.

### How to reproduce it (minimal and precise)

It seems to happen randomly an sporadic. There is no exact way of me reproducing it. Sometimes it works, sometimes it doesn't.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220507

### Adapter

Sonoff 3.0 USB Dongle zStack3x0

### Debug log

Debug Logging shows `contact: false`, but in reality it's closed.
```
info  2022-07-31 09:24:46: MQTT publish: topic 'zigbee2mqtt/E03 WDS', payload '{"battery":100,"contact":false,"device_temperature":28,"linkquality":160,"power_outage_count":8,"temperature":26,"voltage":3015}'
```

closed 2022-09-22 by github-actions[bot]

n3roGit, 2022-12-15:
i have the same

fliespl, 2023-04-30:
Happens on my end as well. Will give 14lm a shot

majkers, 2024-04-15:
Can we reopen this one? I submitted something similar https://github.com/Koenkk/zigbee2mqtt/issues/22210