Koenkk/zigbee2mqtt #13346: Z2M fails after closing the putty session

manfilof, 2022-07-31:
### What happened?

Good night,

There is a problem.
I have a z2m installed on a raspberry. To run the z2m I use Putty. 
After "npm start" everything works perfectly.

But as soon as I close the putty window with session on my laptop (of something is going with the internet that breaks the connection between my laptop and raspberry) z2m stops working.

Feeling myselft stupid because I believed that everything I perform via Putty does on raspberry side and even I close the putty window the z2m service should continue running.

Google did not help me with my issue.

Thank you in advance.

### What did you expect to happen?

Z2M should continue running after the closing Putty session.

### How to reproduce it (minimal and precise)

To close the Putty window.

### Zigbee2MQTT version

1.24.0 commit: a73caa3e

### Adapter firmware version

 20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

there is nothing in the log, since this is not about the z2m-device issue.
BTW, I closed the putty that broke the z2m service and there is nothing in the log

debug 2022-07-31 22:09:00: Received Zigbee message from 'Bathroom relay', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:09:19: Received Zigbee message from 'Plug Humidifier', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":1,"149":43.226016998291016,"152":21.090$
info  2022-07-31 22:09:19: MQTT publish: topic 'zigbee2mqtt/Plug Humidifier', payload '{"consumption":43.23,"energy":43.23,"linkquality":96,"power":21.09,"state":"ON","temperature"$
debug 2022-07-31 22:09:33: Received Zigbee message from 'Plug living room electronics', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:09:39: Received Zigbee message from 'Plug Humidifier', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:10:00: Received Zigbee message from 'Bathroom relay', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:10:28: Received Zigbee message from 'Plug living room electronics', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:10:33: Received Zigbee message from 'Plug Humidifier', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:10:37: Received Zigbee message from 'Bathroom relay', type 'raw', cluster 'genBasic', data '{"data":[28,95,17,104,10,1,255,66,68,3,40,36,5,33,159,13,8,33,47,18,$
debug 2022-07-31 22:11:00: Received Zigbee message from 'Bathroom relay', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:11:12: Received Zigbee message from '0x2c1165fffe4f03b3', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with g$
debug 2022-07-31 22:11:18: Received Zigbee message from 'Plug living room electronics', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2022-07-31 22:11:21: Received Zigbee message from 'Kitchen flood sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":$
info  2022-07-31 22:11:21: MQTT publish: topic 'zigbee2mqtt/Kitchen flood sensor', payload '{"battery":100,"battery_low":false,"linkquality":54,"tamper":false,"water_leak":false}'
debug 2022-07-31 22:11:23: Received Zigbee message from 'Kitchen flood sensor', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":100}' from endpoi$
info  2022-07-31 22:11:23: MQTT publish: topic 'zigbee2mqtt/Kitchen flood sensor', payload '{"battery":100,"battery_low":false,"linkquality":51,"tamper":false,"water_leak":false}'
debug 2022-07-31 22:11:23: Received Zigbee message from 'Plug Humidifier', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0

vTomTom, 2022-08-01:
Follow official guide to run as a service/daemon, do not run within user session which will close all programms if you close it - when it should run forever.
This will also start zigbee2mqtt if you reboot your rasp-pi with no manually intervention.

https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl

Koenkk, 2022-08-01:
You should indeed run it as a daemon, the link from @vTomTom shows you how.

closed 2022-08-01 by Koenkk