Koenkk/zigbee2mqtt #13361: Longpress not available EnOcean PTM 215Z

BodingClockchian, 2022-08-01:
### What happened?

Hi,

I recently switched from Deconz to Zigbee2MQTT using Hassio.
With Deconz I was able to intercept the long press action on the Friend of Hue Switch (PTM 215ZE)
Seems that Zigbee2MQTT only supports the following options: press_1, release_1, press_2, release_2, press_3, release_3, press_4, release_4, press_1_and_3, release_1_and_3, press_2_and_4, release_2_and_4, press_energy_bar. 
Not sure what press_energy_bar does.

Could it be possible to add the long press options as well? I used to control the brightness of my lights with this option.

Also after update 1.27.0-1 the device page is now linking to https://www.zigbee2mqtt.io/devices/HK-LN-DIM-A.html and not the correct one https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

0x26720700

### Adapter

Conbee II

Danit2, 2022-08-02:
The PTM 215ZE do not have this function.
In Deconz the software make this. 
So Z2M don't have this function.

I do this with Node-Red and it works perfect.

BodingClockchian, 2022-08-02:
> The PTM 215ZE do not have this function. In Deconz the software make this. So Z2M don't have this function.
> 
> I do this with Node-Red and it works perfect.

How did you managed to get this working in Node-red?

Danit2, 2022-08-02:
Like this:
[image]

Here you have the flow. You need the [`node-red-contrib-button-events`](https://flows.nodered.org/node/node-red-contrib-button-events) 
```
[{"id":"db818a961b46e485","type":"template","z":"e1e4f87e252f30d4","name":"Off","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"state\": \"OFF\"}","output":"str","x":790,"y":80,"wires":[["fb237a56521980d1"]]},{"id":"619b7a55435aa591","type":"template","z":"e1e4f87e252f30d4","name":"Dimm Stop","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"brightness_move\": 0}","output":"json","x":810,"y":200,"wires":[["fb237a56521980d1"]]},{"id":"a4b390aab6729752","type":"template","z":"e1e4f87e252f30d4","name":"Dimm Up","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"brightness_move\": 30}","output":"json","x":800,"y":120,"wires":[["fb237a56521980d1"]]},{"id":"9359acdb486e27c6","type":"template","z":"e1e4f87e252f30d4","name":"Dimm Down","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"brightness_move\": -30}","output":"json","x":810,"y":160,"wires":[["fb237a56521980d1"]]},{"id":"3e73646046a501ad","type":"template","z":"e1e4f87e252f30d4","name":"On","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"state\": \"ON\"}","output":"str","x":790,"y":40,"wires":[["fb237a56521980d1"]]},{"id":"64c6eb8d3d17f85f","type":"mqtt in","z":"e1e4f87e252f30d4","name":"Feller Switch","topic":"zigbee2mqtt/Feller_Switch","qos":"2","datatype":"json","broker":"909f37f7.fa4128","nl":false,"rap":true,"rh":0,"inputs":0,"x":90,"y":240,"wires":[["09e513edc5eb5c46"]]},{"id":"09e513edc5eb5c46","type":"switch","z":"e1e4f87e252f30d4","name":"Button press","property":"payload.action","propertyType":"msg","rules":[{"t":"eq","v":"press_1","vt":"str"},{"t":"eq","v":"release_1","vt":"str"},{"t":"eq","v":"press_2","vt":"str"},{"t":"eq","v":"release_2","vt":"str"},{"t":"eq","v":"press_3","vt":"str"},{"t":"eq","v":"release_3","vt":"str"},{"t":"eq","v":"press_4","vt":"str"},{"t":"eq","v":"release_4","vt":"str"},{"t":"eq","v":"press_1_and_3","vt":"str"},{"t":"eq","v":"press_2_and_4","vt":"str"},{"t":"eq","v":"release_1_and_3","vt":"str"},{"t":"eq","v":"release_2_and_4","vt":"str"}],"checkall":"true","repair":false,"outputs":12,"x":270,"y":240,"wires":[["7dd22f99ae2a51be"],["7dd22f99ae2a51be"],["8534068aa833f268"],["8534068aa833f268"],["7051beca696b08c7"],["7051beca696b08c7"],["383557b98a3d4bb4"],["383557b98a3d4bb4"],[],[],[],[]]},{"id":"fb237a56521980d1","type":"mqtt out","z":"e1e4f87e252f30d4","name":"Küchenlampe","topic":"zigbee2mqtt/DALI_Device_Korridor/l1/set","qos":"","retain":"","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"909f37f7.fa4128","x":1020,"y":120,"wires":[]},{"id":"7dd22f99ae2a51be","type":"button-events","z":"e1e4f87e252f30d4","name":"Button 1","outputs":3,"inputField":"payload.action","inputFieldType":"msg","outputField":"payload.action","outputFieldType":"msg","downValue":"press_1","downValueType":"str","upValue":"release_1","upValueType":"str","idleValue":"1","clickedInterval":"0","pressedInterval":"500","debounceInterval":"10","events":[{"type":"clicked"},{"type":"pressed"},{"type":"released"}],"x":520,"y":80,"wires":[["3e73646046a501ad"],["a4b390aab6729752"],["619b7a55435aa591"]]},{"id":"8534068aa833f268","type":"button-events","z":"e1e4f87e252f30d4","name":"Button 2","outputs":3,"inputField":"payload.action","inputFieldType":"msg","outputField":"payload.action","outputFieldType":"msg","downValue":"press_2","downValueType":"str","upValue":"release_2","upValueType":"str","idleValue":"1","clickedInterval":"0","pressedInterval":"500","debounceInterval":"10","events":[{"type":"clicked"},{"type":"pressed"},{"type":"released"}],"x":520,"y":160,"wires":[["db818a961b46e485"],["9359acdb486e27c6"],["619b7a55435aa591"]]},{"id":"7051beca696b08c7","type":"button-events","z":"e1e4f87e252f30d4","name":"Button 3","outputs":3,"inputField":"payload.action","inputFieldType":"msg","outputField":"payload.action","outputFieldType":"msg","downValue":"press_3","downValueType":"str","upValue":"release_3","upValueType":"str","idleValue":"1","clickedInterval":"0","pressedInterval":"500","debounceInterval":"10","events":[{"type":"clicked"},{"type":"pressed"},{"type":"released"}],"x":520,"y":300,"wires":[["9a517026c0fda22e"],["357f6508287fe5d4"],["143191829eb2757a"]]},{"id":"383557b98a3d4bb4","type":"button-events","z":"e1e4f87e252f30d4","name":"Button 4","outputs":3,"inputField":"payload.action","inputFieldType":"msg","outputField":"payload.action","outputFieldType":"msg","downValue":"press_4","downValueType":"str","upValue":"release_4","upValueType":"str","idleValue":"1","clickedInterval":"0","pressedInterval":"500","debounceInterval":"10","events":[{"type":"clicked"},{"type":"pressed"},{"type":"released"}],"x":520,"y":380,"wires":[["1def92efacf571ed"],["65b53875825cecc1"],["143191829eb2757a"]]},{"id":"1def92efacf571ed","type":"template","z":"e1e4f87e252f30d4","name":"Off","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"state\": \"OFF\"}","output":"str","x":790,"y":300,"wires":[["18d0373c3ef21b50"]]},{"id":"143191829eb2757a","type":"template","z":"e1e4f87e252f30d4","name":"Dimm Stop","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"{\"brightness_move\": 0}","output" [... cut ...] ","userProps":"","correl":"","expiry":"","broker":"909f37f7.fa4128","x":1030,"y":340,"wires":[]},{"id":"909f37f7.fa4128","type":"mqtt-broker","name":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"protocolVersion":"4","keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willPayload":"","willMsg":{},"sessionExpiry":""}]
```

Hope it helps.

BodingClockchian, 2022-08-02:
Ah of course. This is working perfectly. Thank you.

BodingClockchian, 2022-08-03:
Using Danit's example in Node-red solved this issue for me.

closed 2022-08-03 by BodingClockchian