Koenkk/zigbee2mqtt #13365: Sengled E11-N1EA bulbs no longer show RGB or other controls

bverkron, 2022-08-01:
### What happened?

After upgrading to version 1.27.0-1 of the HA add-on from version 1.26.0-1 my Sengled bulbs not longer have any controls other than basic on-off.

### What did you expect to happen?

Still have all the RGB and other controls I had in 1.26.0-1

### How to reproduce it (minimal and precise)

Run 1.26.0 with a Sengled E11-N1EA bulb paired. You should have full functionality of the bulb. i.e. temperature, color, etc.
Upgrade to 1.27.0. You should now only have basic controls like on-off.

I also tried the latest Edge version 1.27.0-dev commit: [c2faa38](https://github.com/Koenkk/zigbee2mqtt/commit/c2faa38). Same problem.

HA info:
Version | core-2022.7.7
Host Operating System | Home Assistant OS 8.4
Supervisor Version | supervisor-2022.07.0

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

Coordinator revision 20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102N

### Debug log

[zigbee2mqtt_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9237549/zigbee2mqtt_log.txt) (Redacted private info)

bverkron, 2022-08-01:
Perhaps related to #13302 as it was a change specifically for these bulbs?

DHandspikerWade, 2022-08-02:
Also experiencing this issue. Confirmed RBG works on 1.26.0 but switching to the 1.27.0 tagged docker image removes the functionality. 

Tested via MQTT message on the `zigbee2mqtt/sengled_bulb_1/set` topic.

```
{
  "color": {
    "x": 0.99,
    "y": 0.99
  }
}
```

bverkron, 2022-08-02:
What brand / model of bulbs are you using @DHandspikerWade?

DHandspikerWade, 2022-08-02:
Bulbs are labeled as "Sengled Elements Color Plus" and Zigbee2MQTT reports the model as E11-N1EA

ajlunis, 2022-08-02:
Also having this issue

bverkron, 2022-08-02:
Same bulbs or different @ajlunis?

ajlunis, 2022-08-02:
@bverkron Sorry for such little detail, same bulbs.

Koenkk, 2022-08-02:
I made a stupid mistake in https://github.com/Koenkk/zigbee-herdsman-converters/commit/37fee09d5a0bc150a9c36d1cf7ee4c33261fcce3, fixed it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

For those who want to stay on the stable release, you can use the following external converter until the next release (1 September).

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['E11-N1EA'],
    model: 'E11-N1EA',
    vendor: 'Sengled',
    description: 'Element plus color (A19)',
    fromZigbee: extend.light_onoff_brightness_colortemp_color().fromZigbee.concat([fz.metering]),
    toZigbee: extend.light_onoff_brightness_colortemp_color().toZigbee,
    ota: ota.zigbeeOTA,
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness_colortemp_color().configure(device, coordinatorEndpoint, logger);
        device.powerSource = 'Mains (single phase)';
        device.save();

        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.instantaneousDemand(endpoint);
    },
    exposes: extend.light_onoff_brightness_colortemp_color().exposes.concat([e.power(), e.energy()]),
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, issue should be fixed

closed 2022-08-02 by Koenkk

bverkron, 2022-08-02:
Thanks!

I reverted the title to re-include the bulb model since we confirmed it's specific to that model. Might make it easier for others to find the bug.

russellmoran99, 2024-04-08:
I changed my zigbee channel today and had to repair all of my devices.  I no longer have color options.  I have followed the instructions provided by @Koenkk and created the ext_converter.js.  I am still not able to show the colors.  Any helps appreciated.

bat1939, 2024-04-08:
I would also like to report that the issue with these bulbs are back.

russellmoran99, 2024-04-08:
> I would also like to report that the issue with these bulbs are back.

I hate to say that I’m glad. I thought it was just me.  If you find a fix, please share.

loafbutts, 2024-04-12:
This seems to be fixed in the dev branch, as mentioned in issue #22136. In Home Assistant, I switched from the regular Zigbee2MQTT add-on to Zigbee2MQTT Edge, and all of the color controls for my Sengled bulbs came back.