Koenkk/zigbee2mqtt #13370: Wrong devices battery level after last update (1.27.0-1)

zenguru84, 2022-08-02:
### What happened?

before the update the device (aqara tem/humid/press sensor) reported 3015mV and 100% battery level.
after the update the device is reporting 3015mV and 47% battery level.

### What did you expect to happen?

i would expect the battery level to be shown correctly.
as far as i know the battery level should be displayed under 100% only if the voltage drops below 3000mV.

### How to reproduce it (minimal and precise)

just update z2m from 1.26.0 to 1.27.0

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20220219

### Adapter

sonoff zigbee 3.0 dongle plus

ErnestStaug, 2022-08-02:
Same happened to me, all devices suddenly dripped to 30-50%, before all were 100%

sjorge, 2022-08-02:
See the long discussion here: https://github.com/Koenkk/zigbee2mqtt/issues/8499

zenguru84, 2022-08-02:
> See the long discussion here: #8499

i have checked with a brand new battery. still on 47%. i don't think the two topics are related. maybe i'm wrong..

sjorge, 2022-08-02:
Yesterdays release switched back to the old way the battery % was calculated. If you downgrade and it shows back to the old value it's going to be because the calculation was reverted.

znanywawa, 2022-08-02:
> Yesterdays release switched back to the old way the battery % was calculated. If you downgrade and it shows back to the old value it's going to be because the calculation was reverted.

Yes, but still, issue seems to be weird if new battery is calculated like below 50% or (like in my case) 1%

closed 2022-08-02 by Koenkk