Koenkk/zigbee2mqtt #13372: IKEA E1766 won't open/close the blind after firmware upgrade

Geoff571, 2022-08-02:
### What happened?

I've upgrade the firmware on the E1766 to 2.3.079 and while I can still bind it to the group with the blind, it no longer triggers the blind. The actions from the button are being sent to Z2M as I can see them momentarily appear in the device state info.

The Z2M docs for the E1766 remote indicate that binding to the group no longer works and that the remote shouild be bound directly to the device. However whenever I try that through the frontend I get the error below:

```
Error 2022-08-02 13:18:19Nothing to bind from 'Office - Blind remote' to 'Office - Blind'
Info 2022-08-02 13:18:19MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"from":"Office - Blind remote","to":"Office - Blind"},"type":"device_bind_failed"}'
Info 2022-08-02 13:18:19MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Office - Blind remote/1","to":"Office - Blind/1"},"error":"Nothing to bind","status":"error","transaction":"pjg4s-2"}'
Error 2022-08-02 13:18:19Nothing to bind
```
How can I get the remote working directly with the blind again? 

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

 20210120

### Adapter

Electrolama zzh!

Geoff571, 2022-08-25:
Since this issue doesn't seem to attract any attention, does anyone know how to bind the remote directly to the blind?

MattWestb, 2022-08-25:
Retest the controller (4* paring button) and wait until is stop blinking (its have timing out and forming its own network).
Long press the paring button near the blind and it shall touch linking the bling to it standalone network without coordinator.

Geoff571, 2022-08-26:
I've given that a quick try but it hasn't worked. I'll have another go on Monday and also double check the IKEA instructions for rebinding 

Thanks
🙂

EuleMitKeule, 2022-08-28:
I experience the same problem, but also all my bound remotes don't publish action messages while bound to a device.

jficz, 2022-08-29:
Same here with CC2652rb - as per the docs with latest firmware doesn't work with Fyrtur when bound to group and I cannot bind it to the blinds directly - "Nothing to bind".

jficz, 2022-08-30:
Works: need to bind the remote to the `Closure` cluster instead of "onOff".

Not sure if this is how it's supposed to work though (I'm very new in all this)

EuleMitKeule, 2022-08-30:
It started working for me again, maybe because I switched to the latest-dev docker image. I bound Closures to the blinds, but I did so before when it didn't work.

jficz, 2022-08-30:
I'm on NixOS, z2m v1.26.0.

closed 2022-10-09 by github-actions[bot]

ouinouin, 2023-06-10:
hi, same here  i have 2 sets of frytur to do further testing , im alreadiy binding directly to the dievice as its told in the 1766 documentation.