Koenkk/zigbee2mqtt #13377: Support for profalux MOT-C1Z10C and its associated remote MAI-ZTS

Sigfrodr, 2022-08-02:
### Link

https://www.profalux.com/produits/volets-roulants/visio/

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c211bc","nwkAddr":0,"manufId":0,"epList":[242,47,13,12,110,11,10,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"} {"id":2,"type":"Router","ieeeAddr":"0x20918a000010dd07","nwkAddr":61422,"manufId":4368,"manufName":"Profalux","powerSource":"Mains (single phase)","modelId":"MOT-C1Z10C\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,10,21,256,64544,64545],"outClusterList":[3,10,64544],"clusters":{"genBasic":{"attributes":{"modelId":"MOT-C1Z10C\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":"Profalux","powerSource":1,"zclVersion":3,"appVersion":24,"hwVersion":29,"dateCode":"20000101","swBuildId":"Release"}},"genLevelCtrl":{"attributes":{"currentLevel":250}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c211bc","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":514,"inClusterList":[0,1,2,3,4,5,9,10,21,258,2821,64514,64544,64545],"outClusterList":[3,10,25,32,64544],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":24,"hwVersion":29,"dateCode":"20000101","swBuildId":"Release","zclVersion":3,"interviewCompleted":true,"meta":{"configured":734529022},"lastSeen":1659463873932,"defaultSendRequestWhen":"immediate"} {"id":3,"type":"EndDevice","ieeeAddr":"0x20918a000010dd08","nwkAddr":56300,"manufId":4368,"manufName":"Profalux","powerSource":"Battery","modelId":"MAI-ZTS","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":513,"inClusterList":[0,3,21,64514,64544],"outClusterList":[3,4,5,6,8,256,64544,64545],"clusters":{"genBasic":{"attributes":{"modelId":"MAI-ZTS","manufacturerName":"Profalux","powerSource":3,"zclVersion":3,"appVersion":25,"hwVersion":29,"dateCode":"Jun 10 2021","swBuildId":"Release"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":515,"inClusterList":[0,1,2,3,9,21,32,64514,64544],"outClusterList":[3,4,5,25,258,64544,64545],"clusters":{"genBasic":{"attributes":{}},"genPollCtrl":{"attributes":{"checkinInterval":1200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c211bc","endpointID":242}],"configuredReportings":[],"meta":{}}},"appVersion":25,"hwVersion":29,"dateCode":"Jun 10 2021","swBuildId":"Release","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659463607169,"defaultSendRequestWhen":"fastpoll"}root@openmediavault:/srv/dev-disk-by-label-data/appdata/zigbee2mqtt# 

### Comments

Needed to make a custom converter for this model and its associated remote

What has changed vs defaut profalux.js:

 zigbeeModel: ['MOT-C1Z10C\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'], for cover
 zigbeeModel: ['MAI-ZTS'], for remote

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['MAI-ZTS'],
        fingerprint: [{manufId: 4368, endpoints: [{ID: 1, profileID: 260, deviceID: 513, inputClusters: [0, 3, 21],
            outputClusters: [3, 4, 5, 6, 8, 256, 64544, 64545]}]}],
        model: 'NB102',
        vendor: 'Profalux',
        description: 'Cover remote',
        fromZigbee: [],
        toZigbee: [],
        exposes: [],
    },
    {
        zigbeeModel: ['MOT-C1Z10C\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        fingerprint: [{manufId: 4368, endpoints: [{ID: 1, profileID: 260, deviceID: 512,
            inputClusters: [0, 3, 4, 5, 6, 8, 10, 21, 256, 64544, 64545], outputClusters: [3, 64544]}]}],
        model: 'NSAV061',
        vendor: 'Profalux',
        description: 'Cover',
        fromZigbee: [fz.cover_position_via_brightness, fz.cover_state_via_onoff],
        toZigbee: [tz.cover_via_brightness],
        exposes: [e.cover_position().setAccess('state', ea.ALL)],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl']);
            await reporting.brightness(endpoint);
        },
    },
];
```

Sigfrodr, 2022-09-05:
Not stale

Sigfrodr, 2022-10-06:
not stale

dim6003, 2022-11-04:
User of Remote model MAI-ZTG here, don't if compatible from ZigBee point of view with the MAI-ZTS, but interested anyway, can provide  data if needed.

dim6003, 2022-12-07:
not stale

dim6003, 2023-01-08:
not stale

dim6003, 2023-02-08:
..

Jeepmg, 2023-03-08:
Why is this shutter and its remote still not integrated into the profalux.js file of herdsman-converters?

dim6003, 2023-04-08:
What can be done to make this go forward?

sven337, 2023-04-08:
The shutter actually works as far as I can tell. The remote appears as "unsupported" but the Zigbee logs show some traffic that shouldn't be too hard to figure out. I guess this needs a contribution.

sven337, 2023-04-18:
Interesting: my cover is paired, reports its position properly, but I can't set its position from zigbee2mqtt (it's like it was ignored).
Open/close/stop similarly don't work. Does anybody else see that ?

Iroamstaff, 2023-05-14:
Sven337 j'ai l'intégration z2m ha sur raspberry pi il m'a suffit de redémarrer le raspberry pi pour que tout fonctionne correctement j'ai eu ce souci sur chaques volets jumelée

sven337, 2023-05-15:
It's not that - see issue #17674

closed 2023-06-23 by github-actions[bot]