Koenkk/zigbee2mqtt #13379: Woox R7054

wisienoza, 2022-08-02:
### What happened?

Hello,
I have a problem with a Woxx 7054 remote control. it appears as supported on the zigbee2mqtt device list (https://www.zigbee2mqtt.io/devices/TS0215A_remote.html#tuya-ts0215a_remote) but unfortunately I am unable to pair it. When I add the device to Home Assistant, the device is recognized as unsupported. I have zigbee2mqtt installed as an addon.
[image]

I will be very grateful for your help.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

Koenkk, 2022-08-03:
Please provide the data/database.db entry of your device.

wisienoza, 2022-08-03:
database.db below:

{"id":1,"type":"Coordinator","ieeeAddr":"0x00212effff082229","nwkAddr":0,"manufId":4405,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[0,10,25],"outClusterList":[1,32,1280],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":100,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Router","ieeeAddr":"0x00158d0007092cd1","nwkAddr":22681,"manufId":4447,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.light.aqcn02","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,4,3,5,10,258,13,19,6,1,1030,8,768,1027,1029,1026],"outClusterList":[25,10,13,258,19,6,1,1030,8,768],"clusters":{"genBasic":{"attributes":{"65281":{"3":41,"5":75,"8":288,"9":1539,"10":0,"100":0,"101":35,"102":370},"modelId":"lumi.light.aqcn02","appVersion":32,"dateCode":"09-01-2020"}},"genOnOff":{"attributes":{"61440":117440534,"onOff":0}},"genLevelCtrl":{"attributes":{"61440":5806336,"currentLevel":35}},"lightingColorCtrl":{"attributes":{"colorTemperature":370,"tuyaRgbMode":5806336,"colorMode":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":2,"hwVersion":1,"dateCode":"09-01-2020","swBuildId":"1.32\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1659546157298,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0007bccc1f","nwkAddr":7183,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3012},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":76},{"elmType":32,"elmVal":78}],"modelId":"lumi.sensor_switch","appVersion":10}},"genOnOff":{"attributes":{"32768":128,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":10,"interviewCompleted":true,"meta":{},"lastSeen":1659544002482,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0007e68653","nwkAddr":62736,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"1":3005,"4":5032,"5":8,"6":[0,1],"10":0,"100":2488,"101":5360,"102":100427},"modelId":"lumi.weather","appVersion":5}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2490}},"msRelativeHumidity":{"attributes":{"measuredValue":5448}},"msPressureMeasurement":{"attributes":{"measuredValue":1004,"scale":-1,"scaledValue":10043}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":5,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1659545315287,"defaultSendRequestWhen":"immediate"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x804b50fffe04702e","nwkAddr":57443,"manufId":4098,"manufName":"_TZE200_d0yu2xgi","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0011","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_d0yu2xgi","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-201703211},"lastSeen":1659543812401,"defaultSendRequestWhen":"immediate"}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d0007e639e1","nwkAddr":22975,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,65535,6],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"attributes":{"65281":{"1":3025,"3":29,"4":5032,"5":14,"6":[0,7],"10":0,"100":0},"modelId":"lumi.sensor_magnet.aq2","appVersion":3,"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20161128","swBuildId":"3000-0001"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":2,"hwVersion":2,"dateCode":"20161128","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{"configured":-1496948516},"lastSeen":1659545571575,"defaultSendRequestWhen":"immediate"}
{"id":7,"type":"EndDevice","ieeeAddr":"0x00158d0007e2ac06","nwkAddr":64009,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,65535,1030,1024,1280,1,3],"outClusterList":[0,25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":33,"4":5032,"5":10,"6":[0,12],"10":0,"11":18,"100":0},"modelId":"lumi.sensor_motion.aq2","appVersion":5,"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170627","swBuildId":"3000-0001"}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":8}},"msOccupancySensing":{"attributes":{"occupancy":1}},"ssIasZon [... cut ...]
{"id":19,"type":"EndDevice","ieeeAddr":"0xa4c138c99f3c2925","nwkAddr":26209,"manufId":4417,"manufName":"_TZ3000_0zrccfgx","powerSource":"Battery","modelId":"TS0215A","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[1,3,1280,0],"outClusterList":[1281,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff082229","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659531038557,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-08-04:
The one with manufacturer name `_TZ3000_0zrccfgx` was not supported yet, I added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-04 by Koenkk