Koenkk/zigbee2mqtt #13381: Sengled  E11-N1EA can't change colours after new update

greyvv, 2022-08-03:
### What happened?

After updating to the newest version the Sengled E11-N1EA bulbs no longer have the option to change colours

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

sonoff zigbee usb

elderec, 2022-08-03:
Same issue. Upgrading Home Assistant Addon for Zigbe2MQTT v1.26.0-1 to 1.27.0-1 loses color control on Sengled E11-N1EA bulbs. 

Reverting back to 1.26.0-1 restores color functionality.

**Coordinator type**
zStack3x0

**Coordinator revision**
20210708

nathan-curtis, 2022-08-03:
Confirm - Same behavior. 

Reverting restores function and S.O's sanity.

Coordinator / Adapter type
Conbee II

Danit2, 2022-08-03:
Read this: https://github.com/Koenkk/zigbee2mqtt/issues/13365#issue-1325012957

closed 2022-08-03 by greyvv