Koenkk/zigbee2mqtt #13382: Zigbee2mqtt devices becomes unavailable in HA in version 1.27

redpixy, 2022-08-03:
### What happened?

Zigbee2mqtt devices becomes unavailable in HA after short period (around 15 - 30 mins).
zigbee2mqtt/bridge/state show offline.

Restarting the Zigbee2mqtt temporary restores the function but it becomes unavailable shortly after. 
HA Controls to switches also experience long delays before action is registered.

I have downgrade back to version 1.26. The issues doesn't seem to occur anymore.

### What did you expect to happen?

Zigbee2mqtt device remains available.

### How to reproduce it (minimal and precise)

Upgrade to version 1.27 from 1.26.

### Zigbee2MQTT version

1.27

### Adapter firmware version

20220219

### Adapter

Sonoff Itead 3.0

Schubser, 2022-08-03:
I got the same Problem after upgrading from 1.26 to 1.27. Problem is gone after going back to 1.26.

jb123567, 2022-08-03:
Same problem here, can any advise how to downgrade??

Timminater, 2022-08-03:
2 different systems, 2 different dongles. Both gaeetingdevices offline after several hours. Both systems where fine on 1.26, problems started with 1.27

redpixy, 2022-08-03:
> Same problem here, can any advise how to downgrade??

if you are using docker, grab the previous image by adding :1.26

jb123567, 2022-08-03:
> > Same problem here, can any advise how to downgrade??
> 
> 
> 
> if you are using docker, grab the previous image by adding :1.26

Running as an Add-On inside Home Assistant OS. 

Looking further at this, Zigbee2MQTT is still functioning correctly, clearly the connection to HA that is failing.

Koenkk, 2022-08-03:
Please provide the debug log of this.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Grandma-Betty, 2022-08-03:
Same issue here:
```
[21:21:59] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Zigbee2MQTT:info  2022-08-03 21:22:06: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-03.21-22-06' filename: log.txt
Zigbee2MQTT:info  2022-08-03 21:22:06: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-03 21:22:06: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:error 2022-08-03 21:22:07: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-08-03 21:22:07: Failed to start zigbee
Zigbee2MQTT:error 2022-08-03 21:22:07: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-08-03 21:22:07: Exiting...
Zigbee2MQTT:error 2022-08-03 21:22:07: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
```
I also have the "Bad gateway" thing going on as described here:
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/366

b-reich, 2022-08-03:
Mhmm I have the same problem. 
Providing logs tomorrow.

The devices gets unavailable for me when turning them on. (Bulbs)
Even the MQTT shows they are on.

Minoux78, 2022-08-04:
Hello,
Same issue for me, all devices become unavalaible under ha even if they are functional under zigbee2mqtt

b-reich, 2022-08-04:
Nut sure if this a HomeAssistant Problem

redpixy, 2022-08-04:
Not sure if its HA or Z2M but I did notice, If you restart HA while on Z2M 1.27, all the Z2M devices would be shown as unavailable. When execute zigbee2mqtt/bridge/state in MQTT it will show up as offline despite Z2M is running. 

The other issue is the long delays before an action is registered. example turning On a light or smartplugs.

Minoux78, 2022-08-04:
There was no issue with 1.26, it's clearly a Z2M 1.27 problem

Koenkk, 2022-08-04:
@Grandma-Betty this is clearly an issue on your side. The adapter port is incorrect: `Zigbee2MQTT:error 2022-08-03 21:22:07: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'`

@ all others, please provide the debug log, without it I cannot do anything.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Timminater, 2022-08-04:
Already had 3 home-assistant installations with this problem. But after a complete reboot and linking the dongle to the VM again the problem has not returned.
If the problem returns I will try to grab a debug log @Koenkk

jb123567, 2022-08-04:
I host reboot seemed to have sorted my issue

Koenkk, 2022-08-04:
@b-reich 

- which lamps aren't working?
- does it keep working when disabling the availability feature?

b-reich, 2022-08-04:
@Koenkk 
"Nachttisch Lampen" wont work as a group
"Nachttisch Links" and "Nachttisch Links" can be switched individually without problems

Koendejongh, 2022-08-05:
@Koenkk 

I also loose connection between HA and Zigbee2MQTT every day. So I turned on the debug and I believe around 7:03 this morning it happened again. Here is my log
[log - geen ww.txt](https://github.com/Koenkk/zigbee2mqtt/files/9265600/log.-.geen.ww.txt)

Timminater, 2022-08-05:
Happening to me right now, slowly all devices are losing connection

changed log settings de debug and this is de log file:
[z2m_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9265682/z2m_log.txt)

Koenkk, 2022-08-05:
@Timminater in you case it seems that there is some issue on the coordinator side. What coordinator are you using? 

```
warn  2022-08-05 07:44:18: Failed to ping 'Portable_Airco' (attempt 2/2, Read 0xa4c1386cc4469f1b/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no resources' (26)))
```

@Koendejongh  in your case it appears that the MQTT broker is down

```
error 2022-08-05 07:17:43: Not connected to MQTT server!
```

@b-reich so if I understand correctly, the bulbs `Nachttisch Links` and `Nachttisch Links` are in group called `Nachttisch Lampen`. When turning the group OFF the bulbs respond, when turning `ON` nothing happens. If yes, can you try to repair those bulbs and re-add them to the group?

Minoux78, 2022-08-06:
@ Koenkk
Hello,
I loose again connection between HA and Zigbee2MQTT this night (it happens every day).
This happens since 1.27 update.
It seems to be around 04:18
If I restart Zigbee2MQTT, devices become available again.
Here is the log file: 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9274698/log.txt)

And screenshot from mosquitto broker log : 
[image]

Thanks in advance for your help.

navycrow, 2022-08-06:
Same here 
1.27.0-1 release doesn't work at all. It seems there is an issue with mqtt server.

rotilho, 2022-08-06:
Similar issue. After upgrade to 1.27 zigbee2mqtt stopped posting events to mqtt.

I couldn't find anything in the logs, I'll keep an eye and share it as soon as it happen again.

Timminater, 2022-08-06:
> @Timminater in you case it seems that there is some issue on the coordinator side. What coordinator/firmware are you using?

I'm using a cc2538+cc2592 with firmware version 20201010

Koenkk, 2022-08-07:
@Minoux78 thanks for the logging, it seems that z2m indeed disconnects from the broker. I've added some additional logging in the dev branch. Can you update to the latest **edge** addon  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and provide me the herdsman debug logging when z2m disconnects from the broker?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Minoux78, 2022-08-07:
@Koenkk thanks for your feedback. I installed the latest edge addon at around 21:50 (french time).
Devices disconnect at around 21:58
Here the debug logging :
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9277565/log.txt)

fairwood136, 2022-08-08:
@Koenkk I was having similar issues.  When I recompiled, I had ignored the Nodejs version mismatch error.  Once I updated my version from v7.x to v16.6, my issues went away.  Is there a validation check that the correct version of Node is being run?

Koenkk, 2022-08-08:
@Minoux78 I'm missing the logging, not that this is only logged to STDOUT and not to the log files. You need to execute the `docker logs` command to get the logs, see https://www.zigbee2mqtt.io/guide/usage/debug.html#home-assistant-os-supervised-addon

@fairwood136 without logs I cannot determine what the root cause is.

Minoux78, 2022-08-08:
@koenkk
I'm directly under os ha.
I tried to to grab all the logs via SSH & terminal add on by executing "docker ps"  to determine the CONTAINER_ID first but I have the following message "Bash: docker: command not found"...
Not sure it's the good process.

Koenkk, 2022-08-08:
@Minoux78 I think you need to follow https://developers.home-assistant.io/docs/operating-system/debugging/ first

jjansen85, 2022-08-10:
Possibly the same issue happens at my system (Synology NAS with Zigbee2MQTT, Mqtt and HA dockers).

It started last night after mqtt-docker was updated (via watchtower). Entities became unavailable in HA. A Zigbee2MQTT restart was required to fix the problem.

Can easily reproduce the problem by a mqtt restart. Will try to get debug logs later.

Gurgeh88, 2022-08-11:
Same problem here. Everything seems fine in Zigbee2MQTT but in Mosquitto broker, everything is reported as unavailable so remains unavailable in HA.

Koenkk, 2022-08-11:
@jjansen85 pushed a possible fix, can you check if it works? Z2M should now publish online 30 seconds after HA comes online.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jjansen85, 2022-08-11:
@Koenkk This works for my described issue. When restarting MQTT, the online status goed offline. After about 30 seconds, it turns back to online.

Koenkk, 2022-08-11:
@jjansen85 great, I found a better fix. Can you check if it works? Should immediately go to online now instead of after 30 seconds.

Changes will be available in the dev branch in 1 hour from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jjansen85, 2022-08-11:
@Koenkk Perfect! Works much faster.

b-reich, 2022-08-14:
@Koenkk after I did nothing except update HA from time to time. Now everything works again for me.

navycrow, 2022-08-15:
Do you plan to fix the stable branch ?

Koenkk, 2022-08-15:
@navycrow yes, the next release is on 1 September

@redpixy can you confirm the issue has been fixed now? Then this can be closed.

dml33, 2022-08-15:
I have the same issue. My sonoff ZBMINI and SNZB-02 devices have become Unavailable in HA. 

Zigbee2MQTT as addon with version 1.27.0-1
Home Assistant 2022.8.4
Supervisor 2022.08.3
Operating System 8.4
Frontend 20220802.0

In case this issue is fixed, could it be pushed to the stable branch before 1st of September? 15 days waiting for the fix to be propagated while the Zigbee devices are useless seems quite a lot.

Thanks!

Grandma-Betty, 2022-08-15:
> 15 days waiting for the fix to be propagated while the Zigbee devices are useless seems quite a lot.
Indeed.

jjansen85, 2022-08-15:
As a temp. 'solution', you can create a template sensor for the bridge status in combination with an automation which check bridge status and performs a restart if required.

Template sensor:
```
mqtt:
  sensor:
    # Sensor for monitoring the bridge state
    - name: Zigbee2MQTT Bridge state
      unique_id: zigbee2mqtt_bridge_state_sensor
      state_topic: "zigbee2mqtt/bridge/state"
      value_template: "{{ value_json.state }}"
      icon: mdi:router-wireless
```

Automation
```
alias: Zigbee2MQTT - Status check + restart
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.zigbee2mqtt_bridge_state
    from: online
    to: offline
    for:
      hours: 0
      minutes: 0
      seconds: 10
condition: []
action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/bridge/request/restart
  - service: notify.persistent_notification
    data:
      message: Zigbee2MQTT offline, trying restart.
  - delay:
      hours: 0
      minutes: 5
      seconds: 0
      milliseconds: 0
mode: single
```

Koenkk, 2022-08-16:
@dml33 > I have the same issue. My sonoff ZBMINI and SNZB-02 devices have become Unavailable in HA.

This sound more like range issues or issues with the device itself. If you are affected by this issue all devices will show offline, not a few.

If some more people confirm the dev branches fixes the issue I'm willing to create a hotfix release.

wixoff, 2022-08-17:
I'm experiencing this problem on 1.27.0 with a pair of AduroSmart 81813 bulbs - they appear to go offline after some fairly short period of time, but can still be operated through the Z2M front end (which causes them to come back online).

I'm testing latest-dev right now; it's my evening now so we will see how it goes tomorrow.

wixoff, 2022-08-17:
Ah, it didn't work.  Same symptoms while running 1.27.0-dev commit 040d2a1.

I need to increase my logging (I'm currently on "warn" and I don't see any messages) and I will check in with results later.

rotilho, 2022-08-18:
@wixoff that sounds an unrelated issue. This thread is about the whole Zigbee network being unavailable to home assistant not a specific device.

wixoff, 2022-08-18:
@rotilho You're right, thanks - I think I need to report this elsewhere.  Just these two bulbs are affected in my case.

kokobadu, 2022-08-19:
Same issue here.
Hotfix would be greatly appreciated instead of waiting 12 days. 
Many tnx upfront for this. 
Many tnx for all your work on this great project in general.

dsbaars, 2022-08-21:
> As a temp. 'solution', you can create a template sensor for the bridge status in combination with an automation which check bridge status and performs a restart if required.

It's really weird, this doesn't work when adding this to HA itself.
However, with node-red I did something similar to restart Z2M.

It looks like the bridge is going offline every ±15 minutes:

Aug 21 14:54:25 zigbee2mqtt/bridge/state online
Aug 21 15:10:08 zigbee2mqtt/bridge/state offline
Aug 21 15:12:36 zigbee2mqtt/bridge/state online
Aug 21 15:26:07 zigbee2mqtt/bridge/state offline

Koenkk, 2022-08-21:
@dsbaars can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

dsbaars, 2022-08-21:
@Koenkk sure, I do have ±52 devices in the log though so might be useful to apply some filtering perhaps. Is there anything specific I can look for?

m0wlheld, 2022-08-22:
Z2M, HA and Eclipse Mosquitto run in dedicated containers at my house. The issue with devices becoming unavailable in HA, but being present and controllable via Z2M UI started a couple of days **after** I updated Z2M to 1.27 and HA to 2022.8. So I considered Mosquitto to be the cause and reverted the (automatic) update of mosquitto:latest to mosquitto:1.6.10 

It's now the 3rd day, that no unavailability of Z2M devices in HA was observed. Everything works fine - as before.

jjansen85, 2022-08-22:
@m0wlheld Not sure if you got to the right conclusion ;-)

In my case, the problem looked similar to yours. After an automatic update of mosquitto (mosquitto:latest), the mqtt server will do a restart. The mosquitto docker container is updated quite often (every few days, although version stays the same...). Every automatic update will result in a restart of the mosquitto container.

In my case: every restart led to entities being offline in HA due to the Z2M bridge status not updating to online. 

You can test if your issue is within mosquitto or Z2M by stopping and starting the mosquitto container. If you experience the bug I experienced, then the entities should remain unavailable.

mlankamp, 2022-08-22:
I have exactly the same problem that @jjansen85 is describing. I'm not sure if it's just the status not being updated, or the connection to mqtt is not being restored.

dsbaars, 2022-08-22:
The MQTT broker might actually be contributing to this issue. 
I have been running HA and Z2M with an external MQTT broker (on a physically different server, in the same LAN, also mosquitto though).
After the Z2M upgrade to 1.27 the issues described appeared. An upgrade to the Z2M Edge did not solve it.

I did notice some errors in the HA logs that the MQTT server was unreachable, so I installed the Mosquitto broker add-on to run a broker locally. It configured it to replicate to the external MQTT broker as well. 

Since I switched from the external to internal broker, the issue seems to have disappeared. 
Curious if anyone else having this issue is running a non-local MQTT broker as well?

mlankamp, 2022-08-22:
I did some testing where I run the broker on the same box (pi4), an other pi in the network or a VM in cloud. As soon as the connection is lost (for instance because of an container update) the status of the bridge stays offline

dsbaars, 2022-08-22:
> 

> As soon as the connection is lost (for instance because of an container update) the status of the bridge stays offline

I think this is the desired behaviour. Did you enable the watchdog functionality which restarts the bridge after it detects it's not running?

The issue here is that the bridge seems to go offline without losing the connection with the MQTT broker. 
In my case node-red is still able to use Z2M but in HA all devices show as "Unavailable".

jjansen85, 2022-08-22:
> I think this is the desired behaviour

It also stays offline after mqtt has reconnected. That's not desired behaviour in my opinion. 
This was fixed in the current dev version

dsbaars, 2022-08-22:
> It also stays offline after mqtt has reconnected. That's not desired behaviour in my opinion.

Can you check your Home Assistant logs? It might be that Home Assistant is not noticing the MQTT server comes back again. I had about 66 repeated log lines about this.

mlankamp, 2022-08-22:
I'm just running Z2M in a docker container (so no home assistant watchdog enabled). Z2M detects that the mqtt broker is going down (because of a gracefully restart), and it reconnects successfully (topics are being updated). But the state of the bridge is offline.

[32 more replies]

closed 2022-11-06 by github-actions[bot]