Koenkk/zigbee2mqtt #13385: After upgrading from 1.26 to 1.27 zigbee2mqtt which restarts about 1 min

Krzysiekmkg9, 2022-08-03:
### What happened?

After upgrading from 1.26 to 1.27 zigbee2mqtt which restarts about 1 min

### Zigbee2MQTT version

1.27

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB, Egony Stick V4 (Ebyte ver.)

### Debug log

[08:15:35] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Error: Command 0x5c0272fffe8eedbf/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":112,"datatype":0,"data":[5,30,24,8,0,23,11,30,24,12,30,24,17,30,25,22,35,21]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
[08:16:32] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js

Koenkk, 2022-08-03:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Krzysiekmkg9, 2022-08-03:
2022-08-03T18:55:11.347Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":64776,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":29,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[1,5,36,8,0,98,234,196,120,98,234,224,152]}}
2022-08-03T18:55:11.348Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,8,253,1,1,0,239,29,0,30,13,1,5,36,8,0,98,234,196,120,98,234,224,152,202]
2022-08-03T18:55:11.356Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-08-03T18:55:11.356Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-08-03T18:55:11.356Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-08-03T18:55:11.357Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-08-03T18:55:11.357Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-03T18:55:11.559Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x5c0272fffe16b765:2014/1 (0,4,22)
2022-08-03T18:55:11.560Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2014,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":30,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[1,8,36,8,0,98,234,196,120,98,234,224,152]}}
2022-08-03T18:55:11.560Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,222,7,1,1,0,239,30,0,30,13,1,8,36,8,0,98,234,196,120,98,234,224,152,232]
2022-08-03T18:55:11.568Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-08-03T18:55:11.568Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-08-03T18:55:11.568Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-08-03T18:55:11.569Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-08-03T18:55:11.569Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-03T18:55:11.986Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,27,16]
2022-08-03T18:55:11.986Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,27,16]
2022-08-03T18:55:11.987Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,27] - 16
2022-08-03T18:55:11.987Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":27}
2022-08-03T18:55:11.987Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-03T18:55:11.987Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x5c0272fffe8eedbf:48131,205,0)
2022-08-03T18:55:11.987Z zigbee-herdsman:adapter:zStack:adapter Wait 2000ms
2022-08-03T18:55:12.256Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,28,23]
2022-08-03T18:55:12.257Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,28,23]
2022-08-03T18:55:12.257Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,28] - 23
2022-08-03T18:55:12.258Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":28}
2022-08-03T18:55:12.258Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-03T18:55:12.258Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x5c0272fffe90385b:8766,205,4)
2022-08-03T18:55:12.259Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x5c0272fffe90385b:8766/1 (0,0,21)
2022-08-03T18:55:12.259Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":8766,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":31,"options":0,"radius":30,"len":27,"data":{"type":"Buffer","data":[17,12,0,3,0,113,0,0,18,6,0,21,8,0,21,11,30,20,12,30,20,15,30,21,22,0,15]}}
2022-08-03T18:55:12.260Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,37,36,1,62,34,1,1,0,239,31,0,30,27,17,12,0,3,0,113,0,0,18,6,0,21,8,0,21,11,30,20,12,30,20,15,30,21,22,0,15,128]
2022-08-03T18:55:12.260Z zigbee-herdsman:controller:endpoint Command 0x5c0272fffe90385b/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[98,234,196,120,98,234,224,152]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
2022-08-03T18:55:12.261Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,29,22]
2022-08-03T18:55:12.261Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,29,22]
2022-08-03T18:55:12.262Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,29] - 22
2022-08-03T18:55:12.262Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":29}
2022-08-03T18:55:12.262Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-03T18:55:12.262Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x5c0272fffec88f5e:64776,205,4)
2022-08-03T18:55:12.263Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x5c0272fffec88f5e:64776/1 (0,0,20)
2022-08-03T18:55:12.263Z zigbee-herdsman:controller:endpoint Command 0x5c0272fffec88f5e/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[98,234,196,120,98,234,224,152]}, {"sendWhen":"immediate","timeout":10000,"disableResp [... cut ...]
2022-08-03T18:55:13.990Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,37,36,1,3,188,1,1,0,239,34,0,30,27,17,10,0,1,0,112,0,0,18,5,30,24,8,0,23,11,30,24,12,30,24,17,30,25,22,35,21,34]
2022-08-03T18:55:13.999Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-08-03T18:55:13.999Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-08-03T18:55:14.000Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-08-03T18:55:14.000Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-08-03T18:55:14.000Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Krzysiekmkg9, 2022-08-03:
Koenkk, thank you for your interest in the problem.
Between reboots devices can be controlled and sensor data peered

Koenkk, 2022-08-04:
What device is `0x5c0272fffe8eedbf`?

Krzysiekmkg9, 2022-08-04:
Typ urządzenia
EndDevice
Model Zigbee
TS0601
Producent Zigbee
_TZE200_ckud7u2l
Opis
Radiator valve with thermostat
Status wsparcia
Obsługiwane

Adres IEEE
0x5c0272fffe8eedbf
Adres sieciowy
0xBC03
Producent
[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
[TS0601_thermostat](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#tuya-ts0601_thermostat)

Koenkk, 2022-08-04:
Should be fixed now, please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Krzysiekmkg9, 2022-08-05:
@Koenkk my zigbee network started working normally at night while I was sleeping. Now I can't reboot the zigbee network. Should I keep checking the developer branch?

Since zigbee2mqtt update, I have updated home assistant to core 2022.8.0 and 2022.8.1 and restarted home assistant and host many times and it did nothing.

Koenkk, 2022-08-06:
> Now I can't reboot the zigbee network. 

What do you mean?

Krzysiekmkg9, 2022-08-06:
I can't reproduce the situation when zigbee2mq starts every 1 minute or so

Koenkk, 2022-08-06:
I see, so the issue is solved?

Krzysiekmkg9, 2022-08-06:
it has solved itself, but I don't know the reason :)

Krzysiekmkg9, 2022-08-06:
Close issue?

Koenkk, 2022-08-06:
I see, let me know if it comes back.

closed 2022-08-06 by Koenkk