Koenkk/zigbee2mqtt #13388: Tuya TS0601, too high PM2.5 values

jeroenhubgit, 2022-08-03:
### What happened?

My new Tuya Smart air house TS0601 sensor, has measured PM2.5 values which are unrealistically high. I have the sensor indoor and since today I have it close to a open window (outside the PM2.5 is now 8 µg/m³ as reference)
Is this the software, or would my sensor be broken on arrival?

[image]

### What did you expect to happen?

Lower PM 2.5 values between 0 and 25

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

6.7.8.0 build 373

### Adapter

ezsp via tcp - Sonoff zigbee bridge

Koenkk, 2022-08-03:
Does the device report correct values when paired to the TuYa gateway?

jeroenhubgit, 2022-08-03:
Thank you for your message. I would like to know, but I don't have the TuYa gateway. Is there a other way I could check the raw data?

Koenkk, 2022-08-04:
@jeroenhubgit yes, by enabling the debug logging: 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

closed 2022-10-12 by github-actions[bot]