Koenkk/zigbee2mqtt #13389: [New device support]: TS0202 _TZ3000_lltemgsf (Motion Sensor HAOZSE)

d51x, 2022-08-03:
### Link

https://aliexpress.ru/item/1005002625236425.html

### Database entry

{"id":32,"type":"EndDevice","ieeeAddr":"0xa4c138627348c2a9","nwkAddr":56127,"manufId":4417,"manufName":"_TZ3000_lltemgsf","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0xd0cf5efffe893c14","zoneState":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0xd0cf5efffe893c14","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1659521632160,"defaultSendRequestWhen":"immediate"}

### Comments

Just another Tuya-based Motion Sensor from Aliexpress https://aliexpress.ru/item/1005002625236425.html
without battery level and tamper

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            modelID: 'TS0202',
            manufacturerName: '_TZ3000_lltemgsf'
        },
    ],
    model: 'TS0202', // Vendor model number, look on the device for a model number
    vendor: '_TZ3000_lltemgsf', // Vendor of the device (only used for documentation and startup logging)
    description: 'Tuya Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
        fz.ias_occupancy_alarm_1_with_timeout,
        fz.battery
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ], // Should be empty, unless device can be controlled (e.g. lights, switches).
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.occupancy(), e.battery_low(), e.linkquality()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the fronte$};
```

Koenkk, 2022-08-03:
If the code work, please create a PR by clicking the pencil icon here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

Raimundex, 2022-08-07:
Warning 2022-08-06 22:44:30Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3000_lltemgsf'
Warning 2022-08-06 22:44:30Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

d51x, 2022-08-11:
made a PR [d51x:TS0202-_TZ3000_lltemgsf](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4523)

closed 2022-09-18 by github-actions[bot]