Koenkk/zigbee2mqtt #13393: Old coordinator doesnt work as router after flashing router FW

malosaa, 2022-08-03:
### What happened?

I did read many topics, but still don't know how to fix this issue.

Well i updated to a much better stick.
So i did not change anything at all just changed the stick and the network was up again.
cc2531 -> cc2652R

So i went to reflash cc2531 as router firmware and change the second  iEEE.
I did repair it and Z2M does re-pair it as an coordinator under the old IEEE.. i can't change it.

Any idea how to solve this, or do i need to sell this stick because there is no way to change the first IEEE ? 

### Zigbee2MQTT version

1.27

### Adapter firmware version

Router FW cc2531

### Adapter

CC2531

malosaa, 2022-08-03:
i did change the panid and it worked as expected

closed 2022-08-03 by malosaa

lslomczynski, 2022-11-09:
@malosaa 

Hello! Would you be so kind and elaborate your solution a little bit more? :-) For me and for future readers?
I ran into very same scenario with my setup.
I've been using CC2531 as Coordinator and for a long run it was a pain at after some time (weks / months) it was becoming unresponsive and I needed to re-flash it over and over again.
After an upgrade to SONOFF CC2652P, despite deleting /data/coordinator_backup.json  and /data/database.db the new dongle always came back to those files with IEEE of CC2531. Same address was also visible in Z2M webpanel (Settings -> About).

It was fine up until yesterday wen I decided to give the CC2531 a second chance as a router. 
After re-flashing with router firmware and pairing to my network z2m went nuts as it suddenly had 2 devices with same IEEE.
The router was not visible ot the Z2M map, all items were shaking and flying around the screen like they were a drillhammer operators... Also the CC2531 did not show up on the Devices list...

Getting to the point:
1) I cant change the IEEE of my CC2531 as I do not have the CC Debugger BOX - I've been using the "alternartive flash method" all the time.
2) I was able to read my SONOFF IEEE using TI Flash Programmer 2.
3) In the /data/coordinator_backup.json I can see my old IEEE assigned under following variables:
"coordinator_ieee": "00124b0014d9a936",
"extended_pan_id": "00124b0014d9a936",

Should I replace both values with my SONOFF IEEE? Or maybe the extended_pan_id only?
Is it the correct file to edit?
Did you had to re-pair all the devices again?

Just out of curiosity, I wonder where the hack this old IEEE is being kept and why Z2M does not use the new one, even after deletion of abovementioned files... I even ran a grep through my entire 16GB SD card looking for that IEEE hidden somewhere and despite obvious locations like syslog or kernel log (plugging / unplugging) it didn't show up anywhere...

I am looking forward to your reply!

malosaa, 2022-11-12:
Sorry i did not explain a thing what i did.
Well the problem was it got an new IEEE, so because of that all devices stopped working.

Its a while back, so i need to think again what i did.

what i remember was, i did go to the coordinator_backup.json change some settings, 

i have it like this:
 ```
 "coordinator_ieee": "00124bXXXXX0649d",
  "pan_id": "02c0",
  "extended_pan_id": "00124bXXXXX44275",
```
i think the extended_pan_id was from my old device and the coordinator_ieee is the new one.
and i'm not sure if i did change this "pan_id": "02c0", aswell.

but than it worked again.

just try it.

lslomczynski, 2022-11-14:
Hello,

Thank you very much for your time and your reply! :-)

ATM I am a little pinned-down with other stuff, but I hope to get back to this somewhere around December and I will give it a try according to your suggestions.

malosaa, 2022-11-14:
> Hello,
> 
> Thank you very much for your time and your reply! :-)
> 
> ATM I am a little pinned-down with other stuff, but I hope to get back to this somewhere around December and I will give it a try according to your suggestions.

No worries, i'm glad i could help you.
Just ping me if you need help.

regards

lslomczynski, 2022-11-16:
@malosaa 

Here is a complete list of steps that I made - maybe it will be helpful for future readers :-)

1) stop Z2M service
2) delete /data/coordinator_backup.json and /data/database.db
3) edit configuration.yaml:
- pan_id: GENERATE
- network_key: GENERATE
- channel: 20 (I've run away outside of EU 2.4GHz WiFi "official" range as I live in a block of flats and there are litellary DOZENS  of WiFi networks visible in the WiFiAnalyzer android app)
4) flash Sonoff 3.0 dongle with newest coordinator firmware (I had an older one)
5) re-plug the dongle
6) start Z2M, repair devices (as I havent deleted them from configuration.yaml, all friendly names were reassigned and my Node-RED flows were not harmed)
7) restart Z2M to save new config
8) flash CC2531 with router firmware
9) enable join in Z2M and re-plug the CC2531, it pairs straight away :-)

malosaa, 2022-11-22:
Good that you have done it :)  @lslomczynski 

I indeed need to save your steps, if i in the future have the same problem.
As i don't even remember how i did get it to work haha.

Sorry for the late reply, i was really busy.