Koenkk/zigbee2mqtt #13397: Tuya IH-K009 is being identified as Tuya TS0201

thierrybla, 2022-08-04:
### What happened?

I paired 7 Tuya IH-K009's and one of them thinks it is a Tuya TS0201.
I repaired about 50 times but no luck so far...

### What did you expect to happen?

I expected it to be identified as a Tuya IH-K009 like the other 6.

### How to reproduce it (minimal and precise)

Reset the Tuya IH-K009 and turn on joining and it will identity as Tuya TS0201.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Info 2022-08-04 16:41:48Device '0xa4c138d89334ee16' joined
Info 2022-08-04 16:41:48MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d89334ee16","ieee_address":"0xa4c138d89334ee16"},"type":"device_joined"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0xa4c138d89334ee16 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"temperature","enabled_by_default":true,"name":"0xa4c138d89334ee16 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"humidity","enabled_by_default":true,"name":"0xa4c138d89334ee16 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"0xa4c138d89334ee16 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xa4c138d89334ee16 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"0xa4c138d89334ee16 last seen","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Info 2022-08-04 16:41:48MQTT publish: topic 'zigbee2mqtt/0xa4c138d89334ee16', payload '{"battery":null,"humidity":null,"last_seen":"2022-08-04T16:41:48+02:00","linkquality":72,"temperature":null,"voltage":null}'
Info 2022-08-04 16:41:48Starting interview of '0xa4c138d89334ee16'
Info 2022-08-04 16:41:48MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d89334ee16","ieee_address":"0xa4c138d89334ee16","status":"started"},"type":"device_interview"}'
Debug 2022-08-04 16:41:48Received MQTT message on 'homeassistant/sensor/0xa4c138d89334ee16/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0xa4c138d89334ee16 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Debug 2022-08-04 16:41:48Received MQTT message on 'homeassistant/sensor/0xa4c138d89334ee16/temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"temperature","enabled_by_default":true,"name":"0xa4c138d89334ee16 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Debug 2022-08-04 16:41:48Received MQTT message on 'homeassistant/sensor/0xa4c138d89334ee16/humidity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"humidity","enabled_by_default":true,"name":"0xa4c138d89334ee16 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_humidity_zi [... cut ...]
Info 2022-08-04 16:41:57MQTT publish: topic 'zigbee2mqtt/0xa4c138d89334ee16', payload '{"battery":null,"humidity":null,"last_seen":"2022-08-04T16:41:57+02:00","linkquality":72,"temperature":29.54,"voltage":null}'
Debug 2022-08-04 16:41:57Received Zigbee message from '0xa4c138d89334ee16', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2954}' from endpoint 1 with groupID 0
Info 2022-08-04 16:41:58MQTT publish: topic 'zigbee2mqtt/0xa4c138d89334ee16', payload '{"battery":null,"humidity":null,"last_seen":"2022-08-04T16:41:58+02:00","linkquality":72,"temperature":29.54,"voltage":null}'

thierrybla, 2022-08-04:
[image]
[image]

Koenkk, 2022-08-04:
> I paired 7 Tuya IH-K009's and one of them thinks it is a Tuya TS0201.

Can you provide the data/database.db entry of the IH-K009 that is detected as a TS0202

thierrybla, 2022-08-04:
{"id":19,"type":"EndDevice","ieeeAddr":"0xa4c138d89334ee16","nwkAddr":12926,"manufId":4417,"manufName":"_TZ3000_dowj6gyi","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2814}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"msRelativeHumidity":{"attributes":{"measuredValue":5049}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659649371549,"defaultSendRequestWhen":"immediate"}

thierrybla, 2022-08-04:
> > I paired 7 Tuya IH-K009's and one of them thinks it is a Tuya TS0201.
> 
> Can you provide the data/database.db entry of the IH-K009 that is detected as a TS0202

{"id":19,"type":"EndDevice","ieeeAddr":"0xa4c138d89334ee16","nwkAddr":12926,"manufId":4417,"manufName":"_TZ3000_dowj6gyi","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2814}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"msRelativeHumidity":{"attributes":{"measuredValue":5049}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659649371549,"defaultSendRequestWhen":"immediate"}

I hope I'm doing it right..

timnolte, 2022-08-05:
@Koenkk have you ever attempted to get another TS0201 device working with your Z2M? Did you happen to attempt to manually add a TS0201 device to `config/zigbee2mqtt/configuration.yaml`? Can you look in that file to confirm that there doesn't exist an entry in there that matches your IH-K009 device ID and/or `_TZ3000_dowj6gyi` manufacturer name? I just fixed the integration for that device as it was previously identified as a WSD500A. TS0201 is a rather generic Model ID that many devices carry.

The device you are showing is setup through this code:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L1256-L1270

Which is a pretty generic identifier being used.

timnolte, 2022-08-05:
@thierrybla one other question do you have the Legacy options still turned on in Z2M?
[image]

thierrybla, 2022-08-05:
> @thierrybla one other question do you have the Legacy options still turned on in Z2M?
> [image]
> 

No, it's off.

timnolte, 2022-08-05:
I have 2 more of these in the way so I'll be curious if I see this happen. You paired over 50 of them you say so I wonder if that is somehow related.

timnolte, 2022-08-05:
@thierrybla have you tried repairing any of the other devices that were identified correctly?

thierrybla, 2022-08-05:
> @thierrybla have you ever attempted to get another TS0201 device working with your Z2M? Did you happen to attempt to manually add a TS0201 device to `config/zigbee2mqtt/configuration.yaml`? Can you look in that file to confirm that there doesn't exist an entry in there that matches your IH-K009 device ID and/or `_TZ3000_dowj6gyi` manufacturer name? I just fixed the integration for that device as it was previously identified as a WSD500A. TS0201 is a rather generic Model ID that many devices carry.
> 
> The device you are showing is setup through this code:
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L1256-L1270
> 
> Which is a pretty generic identifier being used.

  '0xa4c138940e6165f3':
    friendly_name: Bedroom temperature and humidity sensor
    
    This is one that is correct in config/zigbee2mqtt/configuration.yaml
    
      '0xa4c138d89334ee16':
    friendly_name: '0xa4c138d89334ee16'
    
    This is the one that is wrong.
    
    I have never manually edited any of these files, only used the UI. I also never owned a TS0201

thierrybla, 2022-08-05:
> @thierrybla have you tried repairing any of the other devices that were identified correctly?

I just tried to remove and repair another one, it shows up correctly but keeps saying it is offline? Weird..
[image]

Koenkk, 2022-08-05:
@thierrybla I cannot spot the issue yet, can you provide your complete data/database.db?

thierrybla, 2022-08-05:
> @thierrybla I cannot spot the issue yet, can you provide your complete data/database.db?
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/9270222/database.zip)

Or in text form:
[database db.txt](https://github.com/Koenkk/zigbee2mqtt/files/9270224/database.db.txt)

Koenkk, 2022-08-06:
Could you try to:
- Force remove the device via the z2m frontend
- Restart z2m
- Repair the device
- If that doesn't help; do the same and provide me the herdsman debug logging of it
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

thierrybla, 2022-08-06:
> Could you try to:
>
> * Force remove the device via the z2m frontend
> * Restart z2m
> * Repair the device
> * If that doesn't help; do the same and provide me the herdsman debug logging of it
>   See [...]

I force removed the device, restarted and repaired the device but still came in wrong.
Here are the logs you asked for:
`Zigbee2MQTT:info  2022-08-06 19:04:16: Device '0xa4c138d89334ee16' joined
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d89334ee16","ieee_address":"0xa4c138d89334ee16"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0xa4c138d89334ee16 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"temperature","enabled_by_default":true,"name":"0xa4c138d89334ee16 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"humidity","enabled_by_default":true,"name":"0xa4c138d89334ee16 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"0xa4c138d89334ee16 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xa4c138d89334ee16 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa4c138d89334ee16_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-08-06 19:04:16: MQTT publish: topic 'homeassistant/sensor/0xa4c138d89334ee16/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c138d89334ee16/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138d89334ee16"],"manufacturer":"TuYa","model":"Temperature & humidity sensor with display (TS0201)","name":"0xa4c138d89334ee16"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"0xa4c138d89334ee16 last seen","state_topic":"zigbee2mqtt/0xa4c138d89334ee16","unique_id":"0xa [... cut ...]
Zigbee2MQTT:info  2022-08-06 19:04:24: MQTT publish: topic 'zigbee2mqtt/0xa4c138d89334ee16', payload '{"battery":null,"humidity":54.08,"last_seen":"2022-08-06T19:04:24+02:00","linkquality":66,"temperature":27.9,"voltage":null}'
Zigbee2MQTT:debug 2022-08-06 19:04:26: Received Zigbee message from '0xa4c138d89334ee16', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-08-06 19:04:26: MQTT publish: topic 'zigbee2mqtt/0xa4c138d89334ee16', payload '{"battery":null,"humidity":54.08,"last_seen":"2022-08-06T19:04:26+02:00","linkquality":63,"temperature":27.9,"voltage":null}'`

thierrybla, 2022-08-06:
@Koenkk Did you change something? Suddenly it is listed correctly

[image]

timnolte, 2022-08-06:
@thierrybla unless you applied an update nothing @Koenkk might have done would have fixed anything. I'm wondering if your device just isn't fully being queried at it's first pairing. Also if the device is not very close to your coordinator it could be having problems communicating back to Z2M.

timnolte, 2022-08-06:
@thierrybla it's also very common to need to do a second pairing, without removing the device, you just put it back into pairing mode while it is already registered in Z2M, to get all of the details registered in Z2M.

thierrybla, 2022-08-06:
> @thierrybla it's also very common to need to do a second pairing, without removing the device, you just put it back into pairing mode while it is already registered in Z2M, to get all of the details registered in Z2M.

I tried this but it never worked even after trying 10-15 times

thierrybla, 2022-08-06:
> @thierrybla unless you applied an update nothing @Koenkk might have done would have fixed anything. I'm wondering if your device just isn't fully being queried at it's first pairing. Also if the device is not very close to your coordinator it could be having problems communicating with your coordinator and registering correctly in Z2M.

The devices was right next to it since I was testing it, but now after enabling the debug it suddenly works correctly.. very weird.

closed 2022-08-06 by thierrybla

timnolte, 2022-08-07:
@thierrybla hmm, that is weird indeed. I'll be interested to see if I run into this when my next couple of them arrive.

mrmaximas, 2023-03-27:
i have two same devices _TZ3000_dowj6gyi but strange things are happening.
Updates come at the same time with different values, I think this is the previous and current values

Info 2023-03-27 11:24:48MQTT publish: topic 'zigbee2mqtt/Test HT 2', payload '{"battery":100,"humidity":40.5,"linkquality":80,"temperature":22.5,"voltage":3000}'
Info 2023-03-27 11:24:48MQTT publish: topic 'zigbee2mqtt/Test HT 2', payload '{"battery":100,"humidity":41.1,"linkquality":80,"temperature":22.5,"voltage":3000}'

mrmaximas, 2023-03-27:
> The devices was right next to it since I was testing it, but now after enabling the debug it suddenly works correctly.. very weird.

Does your data also update no more than every five minutes? And with a sudden change in temperature or humidity nothing happens?

MilesTEG1, 2024-04-17:
Hello here,
I was about to ask here for help, because I bought 4 Tuya IH-K009 devices, and the first one I paired to my Z2M was recognized as a Tuya TS0201.
I tried several unpairing / repairing, without success.
And, I don't know why, but I reboot Z2M... And then, after restarting, the Tuya IH-K009 device was really a Tuya IH-K009 device in Z2M 😃 
[image]
But still there is this:
[image]

For now, I think it's normal.