Koenkk/zigbee2mqtt #13401: Zigbee2mqtt 1.27.0  update seems to drain Aqara batteries

kwetnico, 2022-08-04:
### What happened?

I installed the latest version of zigbee2mqtt (1.27.0) addon
This only affects the aqara mark.  A large part of my equipment has fallen.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26680700

### Adapter

Conbee2

kwetnico, 2022-08-04:
https://github.com/Koenkk/zigbee2mqtt/issues/8499

Koenkk, 2022-08-04:
The battery % calculation has been changed (https://github.com/Koenkk/zigbee2mqtt/issues/8499 as @kwetnico mentioned). It doesn't affect the actual battery.

closed 2022-08-04 by Koenkk

kwetnico, 2022-08-04:
> The battery % calculation has been changed (#8499 as @kwetnico mentioned). It doesn't affect the actual battery.

Yes I read on the other hand I wonder if the calculation is really good, my new sensor indicates a battery at 50%.... Like all my equipment

Koenkk, 2022-08-04:
What is the voltage?

kwetnico, 2022-08-04:
> What is the voltage?

3025 v
[image]

Koenkk, 2022-08-04:
I see, https://github.com/Koenkk/zigbee2mqtt/issues/8499#issuecomment-1205696747 would fix that

jdbrookes, 2022-08-05:
I have a number of aqara devices that I've been running for several years, and some of them simply don't respond an accurate voltage.

For example just yesterday I had an aqara climate sensor drop off my network. The battery reading was at that point just fine and the last voltage reported was around 2.95 or similar. Not full, but not empty either.

Checking the actual battery cell with a multimeter showed a much lower voltage of something like 2.7 or even less.

I've seen this with a number of older aqara devices in the last year or two. The battery life is great, but the hardware doesn't report the battery life to an accurate level and doesn't usually indicate that the battery is about to fail.

kwetnico, 2022-08-05:
add your interesting comment on the subject: https://github.com/Koenkk/zigbee2mqtt/issues/8499#issuecomment-1205696747