Koenkk/zigbee2mqtt #13403: Centralite 3157100: local_temperature_calibration now showing "unknown" in Home Assistant

beauwest, 2022-08-04:
### What happened?

Up until recently, the device's `local_temperature_calibration` showed a value in Home Assistant. Now the value only shows as "unknown".

I recently updated both Home Assistant, and Zigbee2MQTT, so I'm honestly not sure which caused the change in behavior. If this is obviously not something Zigbee2MQTT related, please feel free to close the issue. My gut feeling is that this is Home Assistant related, and I will be pursing that as well.

### What did you expect to happen?

I expect to see the `local_temperature_calibration` value in Home Assistant.

### How to reproduce it (minimal and precise)

Happens just by having this device connected.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Here is the most recent JSON MQTT message.

```
{
  "battery": 100,
  "fan_mode": "auto",
  "fan_state": "ON",
  "linkquality": 116,
  "local_temperature": 23.87,
  "local_temperature_calibration": 0,
  "occupied_cooling_setpoint": 23.89,
  "occupied_heating_setpoint": 21.67,
  "running_state": "cool",
  "system_mode": "cool",
  "temperature_setpoint_hold": false,
  "voltage": 2800
}
```

Koenkk, 2022-08-05:
Given that `"local_temperature_calibration": 0,` is in the JSON payload, this looks to be a HA issue.

beauwest, 2022-08-05:
Thank you, I figured it would be, but I wanted to rule out any metadata changes that Z2M might be passing along.

Thank you.

closed 2022-08-05 by beauwest