Koenkk/zigbee2mqtt #13419: Ikea Tradfri Motion : Unsupported

lowrisk75, 2022-08-05:
### What happened?

I'm try to add a few  Ikea Tradfri Motion, no mater which one I'm paring (pressing 4 times the paring button, they added on Z2M with the message "Unsupported"

### What did you expect to happen?

Added on Z2m and recognised as  Ikea Tradfri Motion

### How to reproduce it (minimal and precise)

Press 4 times in the link button on the  Ikea Tradfri Motion and enable paring mode in z2m

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB

### Debug log

Zigbee2MQTT:info  2022-08-05 18:17:28: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":45,"power":155,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:17:33: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":45,"power":170,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:17:38: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":42,"power":149,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:17:43: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.568,"linkquality":87,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:17:43: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":45,"power":155,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:17:48: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":36,"power":143,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:17:53: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":42,"power":149,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:17:58: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":48,"power":146,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
zlinky config: historique, single_phase, false, HC..
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.568,"linkquality":87,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.568,"linkquality":87,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.568,"linkquality":84,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.568,"linkquality":84,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.575,"linkquality":84,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.575,"linkquality":84,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":590,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":20479.253,"current_tier2_summ_delivered":16637.575,"linkquality":87,"meter_serial_number":"031662997326","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":null,"update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'
Zigbee2MQTT:info  2022-08-05 18:18:03: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":42,"power":158,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:18:08: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":48,"power":143,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:18:13: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":48,"power":140,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:18:16: MQTT publish: topic 'zigbee2mqtt/Sensor kitchen ', payload '{"battery":100,"illuminance":19609,"illuminance_lux":91,"led_indication":null,"linkquality":102,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":24.95,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-08-05 18:18:18: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":45,"power":155,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:18:19: MQTT publish: topic 'zigbee2mqtt/Garage  sensor road', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"led_indication":null,"linkquality":87,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":23.39,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-08-05 18:18:23: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":48,"power":154,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:18:24: MQTT publish: topic 'zigbee2mqtt/Sensor living', payload '{"battery":100,"illuminance":21171,"illuminance_lux":131,"led_indication":null,"linkquality":60,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":24.98,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-08-05 18:18:28: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":45,"power":155,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:18:36: Device '0xcc86ecfffea39a45' joined
zlinky config: historique, single_phase, false, HC..
[... cut ...] ilable":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:21:22: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":48,"power":146,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2022-08-05 18:21:27: MQTT publish: topic 'zigbee2mqtt/Office plug', payload '{"action":null,"current":null,"linkquality":42,"power":147,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'

clarsen, 2022-08-16:
I have same issue with Tradfi motion, v1.27.0-dev and Sonoff Zigbee 3.0 USB (EZSP v8) coordinator.  This was what was written to database.db.  The model info wasn't detected:
```json
{"id":4,
"type":"EndDevice",
"ieeeAddr":"0x84ba20fffecae9cc",
"nwkAddr":11842,
"manufId":4476,
"manufName":"IKEA of Sweden",
"powerSource":"Battery",
"epList":[1],
"endpoints":{
  "1":{
    "profId":260,"epId":1,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,4096],
    "clusters":{
      "genBasic":{
        "attributes":{
          "manufacturerName":"IKEA of Sweden",
          "powerSource":3,
          "zclVersion":3,
          "appVersion":32,
          "stackVersion":98,
          "hwVersion":1,
          "dateCode":"20190308",
          "swBuildId":"2.0.022"
        }
      },
      "genPollCtrl":{
          "attributes":{
            "checkinInterval":172800
          }
        }
      },
      "binds":[
        {"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffefcf2d2","endpointID":1}
      ],
      "configuredReportings":[],
      "meta":{}
    }
  },
  "appVersion":32,
  "stackVersion":98,
  "hwVersion":1,
  "dateCode":"20190308",
  "swBuildId":"2.0.022",
  "zclVersion":3,
  "interviewCompleted":true,
  "meta":{},
  "lastSeen":1660583551619,
  "defaultSendRequestWhen":"active"
}
```

the devices table in the UI showed manufacture "Unsupported", and model blank.

The device itself claimed that "interview completed" was true.

Then I tried invoking read genBasic directly by using the dev console, I could get the model info just fine:
2022-08-15 20:37:32Read result of 'genBasic': {"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI motion sensor"}

and then the database.db had updated details, specifically the modelId was "TRADFRI motion sensor".

So maybe the interview process completed partially but didn't get all the necessary data, such as the modelId.

cbourne142857, 2022-09-02:
Did you get any further with this?

I'm finally getting around to migrating to Z2M from ZHA - currently using a CC2531 stick with ZHA, and planning to shift to the Sonoff 3.0 dongle plus with Z2M.

I've got them both running concurrently at the moment and have been planning to repair devices - I've only tried Ikea ones so far (a bulb and an outlet switch) and they're both failing the interview / showing as unsupported:

```
s6-rc: info: service z2m: starting
s6-rc: info: service socat: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service z2m successfully started
s6-rc: info: service socat successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:12:23] INFO: Socat not enabled
[16:12:24] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:debug 2022-09-02 16:12:28: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-09-02 16:12:28: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-02.16-12-28' filename: log.txt
Zigbee2MQTT:debug 2022-09-02 16:12:28: Removing old log directory '/config/zigbee2mqtt/log/2022-09-02.11-49-06'
Zigbee2MQTT:info  2022-09-02 16:12:28: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-02 16:12:28: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:debug 2022-09-02 16:12:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:info  2022-09-02 16:12:29: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-09-02 16:12:29: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:debug 2022-09-02 16:12:29: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b001cce0ef2","panID":6754}
Zigbee2MQTT:info  2022-09-02 16:12:29: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-09-02 16:12:29: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-09-02 16:12:29: Connecting to MQTT server at mqtt://core-mosquitto
Zigbee2MQTT:debug 2022-09-02 16:12:29: Using MQTT login with username: mqtt-user
Zigbee2MQTT:info  2022-09-02 16:12:30: Connected to MQTT server
Zigbee2MQTT:info  2022-09-02 16:12:30: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-09-02 16:12:30: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:debug 2022-09-02 16:12:30: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b001cce0ef2","panID":6754},"permit_join":false,"version":"1.27.2"}'
Zigbee2MQTT:info  2022-09-02 16:12:30: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b001cce0ef2","panID":6754},"permit_join":false,"version":"1.27.2"}'
Zigbee2MQTT:info  2022-09-02 16:12:30: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:debug 2022-09-02 16:12:49: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"n8v60-2","value":true}'
Zigbee2MQTT:info  2022-09-02 16:12:49: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-09-02 16:12:49: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"n8v60-2"}'
Zigbee2MQTT:info  2022-09-02 16:13:34: Device '0x804b50fffe21f578' joined
Zigbee2MQTT:info  2022-09-02 16:13:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe21f578","ieee_address":"0x804b50fffe21f578"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-09-02 16:13:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x804b50fffe21f578"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-09-02 16:13:34: Starting interview of '0x804b50fffe21f578'
Zigbee2MQTT:info  2022-09-02 16:13:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe21f578","ieee_address":"0x804b50fffe21f578","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-09-02 16:13:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x804b50fffe21f578"},"type":"pairing"}'
Zigbee2MQTT:debug 2022-09-02 16:15:41: Received MQTT message on 'zigbee2mqtt/0x804b50fffe21f578/set' with data '{"read":{"attributes":["powerSource"],"cluster":"genBasic","options":{}}}'
[... cut ...]
Zigbee2MQTT:info  2022-09-02 16:17:30: Starting interview of '0x804b50fffe21f578'
Zigbee2MQTT:info  2022-09-02 16:17:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe21f578","ieee_address":"0x804b50fffe21f578","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-09-02 16:17:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x804b50fffe21f578"},"type":"pairing"}'

```

closed 2022-10-15 by github-actions[bot]