Koenkk/zigbee2mqtt #13421: LIDL 14149506L light colors not functioning properly

crooy, 2022-08-05:
### What happened?

I connected the LIDL 14149506L(https://www.zigbee2mqtt.io/devices/14149505L_14149506L.html#lidl-14149505l%252F14149506l), which went smoothly. 

However as described in the discussion (https://github.com/Koenkk/zigbee2mqtt/discussions/5944), the color only appear to work when cycling colors. 

### What did you expect to happen?

I expected realtime colors.

### How to reproduce it (minimal and precise)

I have 2 of these lights and they both show the same problem.

### Zigbee2MQTT version

    1.27.0 commit: a9b8808

### Adapter firmware version

20210120

### Adapter

Slaesh CC2652RB stick

Lynetk, 2022-08-06:
Description of the problem from linked issue pasted for convenience.

> It works "kinda". RGB only functions after power cycling the device. As soon as I try to change brightness or colour temperature the white LEDs come on and never get switched off, no matter what setting I try.

Koenkk, 2022-08-06:
Seems this device doesn't properly support the standard Zigbee commands so some reverse engineering is necessary. Can someone make a sniff of:
- Pairing the device to the TuYa hub
- Changing the color
- Changing the color temp
- Changing the color

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

mtrr, 2022-08-12:
Unfortunately I don't have the needed equipment, if somebody in Germany is capable of doing the legwork, I'd be willing to donate my lamp (or donate 20,- EUR so that one could be bought from lidl)

crooy, 2022-08-26:
> Unfortunately I don't have the needed equipment, if somebody in Germany is capable of doing the legwork, I'd be willing to donate my lamp (or donate 20,- EUR so that one could be bought from lidl)

Same here, happy to share that donation!

mtrr, 2022-09-28:
next week lidl Germany has them for € 17,99, so maybe ...

oli-f, 2022-10-02:
Seems like deconz has the same problem with the device: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5900

As can be seen in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5900#issuecomment-1077763268, the device has cold-white+warm-white LEDs (temperature mode) as well as independant RGB LEDs (color mode). Currently when changing the color, the cold-white+warm-white LEDs stay on.

Changing betwen the modes can be done via the `tuyaRgbMode`:
```
await entity.command('lightingColorCtrl', 'tuyaRgbMode', {enable: 1}, {}, {disableDefaultResponse: true});
...
await entity.command('lightingColorCtrl', 'tuyaRgbMode', {enable: 1}, {}, {disableDefaultResponse: true});
```

I achived minor improvement by using the tuya_led_control, however the color temperature is incorrect and in color mode the brighness can't be changed:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0505A', manufacturerName: '_TZ3000_gek6snaj'},
        {modelID: 'TS0505B', manufacturerName: '_TZ3210_iystcadi'}
    ],
    model: '14149505L/14149506L',
    vendor: 'Lidl',
    description: 'Livarno Lux light bar RGB+CCT (black/white) CUSTOM',

    toZigbee: [tz.on_off, tz.tuya_led_control],
    fromZigbee: [fz.on_off, fz.tuya_led_controller, fz.brightness, fz.ignore_basic_report],
    exposes: [e.light_brightness_colortemp_colorhs([153, 500]).removeFeature('color_temp_startup')],
};

module.exports = definition;
```

It seems like the device does not interpret the brightness from `tuyaMoveToHueAndSaturationBrightness`. The `genLevelCtrl/moveToLevel` switches the device to temperature mode and `tuyaBrightness` is not supported - so not sure how to set the brightness in color mode.

lSh4dowl, 2022-10-07:
I also have this device and this problem and another problem might be the fact, that the RGB Leds are super weak on their own. 
I don't know how this device normally works since I only have home assistant to connect it to, but if I get the color set correctly (which feels very random) the light is super dimm.
On the other Hand, when the light is bright, only reddish colors work and blue for example results in pink, green results in orange.

Any progress ?

steffen890, 2022-11-08:
not stale

alterschlabberanesi, 2022-11-20:
Please See #12947 

Looking for someone to sniff the traffic when controlling this device via the TuYa gateway

ALotOfMooh, 2022-11-29:
Playing around with scenes in Home assistant sometimes the white LEDs turn off and color mode kind of works. I can change colors afterwards. Changing brightness, turn on white LEDs again.
Maybe that helps?

KentuckyMC, 2022-12-02:
Got the same led here, and indeed the colors are very weird. Sometimes blue = blue, but changing brightness shows purple suddenly. Not sure what is going on with this one. With the chargeable LED light of Lidl those colors seems to work fine.

stefan131, 2022-12-14:
I just got my Sonoff ZigBee stick and set it up with Zigbee2MQTT in Home Assistant.
You can actually get it to display just an RGB color, without W/WW
- Set the light to any RGB color
- Physically turn of the light
- Wait a few seconds
- Turn it back on, whoop only RGB

Now if you toggle it off and on in HASS, it will include white again.
A workaround for this is to turn off `color_sync`:
[image]

If you toggle the light with `color_sync` disabled, it will be green every time.
It does remember the last state, so you'll have to switch to a different color before being able to set the same color again.
Using a color temperature with this light means you'll have to physically turn if off and on again in order to get just RGB

[deCONZ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5900) and [ZHA](https://github.com/zigpy/zha-device-handlers/issues/1567) seem to be having the same issue

cbcotton, 2022-12-20:
It wont set color_xy after adjusting brightness in the interface. It will let you change color after adjusting color_temp first. Setting the color_temp also seems to override the brightness settings sometimes. Ive tried reproduce this accurately, sometimes it wont behave and wont dim the whiteleds when setting color_xy and sometimes it dims all leds to the lowest brightness.
- Set brightness to 254
- Set color_temp to warmest
- Set color_temp to warm
- Set color_temp to warmest
- Set color_xy to choice

ALotOfMooh, 2023-01-21:
not stale. As i stated scenes work 50% with setting a specific color. This might help someone

RunnicFusion, 2023-01-23:
Any update on this issue?
Is disabeling `color_sync` the way to go right now?

AndreKR, 2023-01-25:
I still have the issue with `color_sync` disabled.

AndreKR, 2023-01-25:
I have ordered a hub, will provide a packet capture when it arrives.

RunnicFusion, 2023-01-25:
> I still have the issue even with `color_sync` disabled.

Same over here

AndreKR, 2023-01-29:
Ok, so here is the dump. I collected a couple more actions.

| Time| Action |
| --- | --- |
| | |
| | |
| | |
| | |
| | |
| | |
| | |

Times are relative to start of dump. I don't know if the dump file contains the already decrypted data, so just in case, the network key is `7b4a2de2a74b61ff05ef13eff654a9c7`.

Koenkk, 2023-01-29:
Thanks for the sniff. 

As a first step (probably will not fix everything), can you see what does work after using https://gist.github.com/Koenkk/16d12d4d97a92fe3687118ee39693e49 ?

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

AndreKR, 2023-01-29:
```
/app/node_modules/zigbee-herdsman-converters/index.js:78
    definition.toZigbee.push(tz.scene_store, tz.scene_recall, tz.scene_add, tz.scene_remove, tz.scene_remove_all, tz.read, tz.write,
                        ^
TypeError: Cannot read properties of undefined (reading 'push')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:78:25)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

AndreKR, 2023-01-29:
It's pretty much the same as before, the white LEDs stay on when switching to a color and the color LEDs are not affected by the brightness setting. (The very first time I switched to color today the white turned off, but it doesn't do that anymore.)

Koenkk, 2023-01-29:
> The very first time I switched to color today the white turned off, but it doesn't do that anymore.

Could you attempt to reproduce this? Because this is what I expected after my change.

AndreKR, 2023-01-29:
I think I found the pattern.
When I physically switch of the lamp (remove power) and turn it back on, I can switch between white (by setting color temperature) and color *until* I use the brightness slider, after that the white stays on.

Koenkk, 2023-01-30:
I updated https://gist.github.com/Koenkk/16d12d4d97a92fe3687118ee39693e49, can you check again?

AndreKR, 2023-01-30:
* Selecting a color now successfully switches to color mode with white LEDs turned off 👍 but always on the lowest brightness.
* Switching from color to while will now also successfully turn off the color LEDs.
* Trying to set the brightness while in color mode will switch to white with the brightness that was previously set in white mode. Setting the brightness *again* will then actually change the brightness.
* If I turn the lamp off while it is in color mode and then on again, it will always be green and I have to set the color *twice* to set the color, for example if I first click red and then purple the click on red is ignored and with the click on purple it will turn purple.

Koenkk, 2023-01-31:
> Trying to set the brightness while in color mode will switch to white with the brightness that was previously set in white mode. Setting the brightness again will then actually change the brightness.

Could you provide a sniff of doing just this via the TuYa gateway?

AndreKR, 2023-01-31:
It's the step at 57 s in the file above.

mayongze, 2023-02-03:
@Koenkk I'll try to analyze it, but I've been a little busy lately, and I need some time.

AndreKR, 2023-02-03:
By moving the sliders in the app I discovered that the structure is actually quite simple.

The interesting one is in your second screenshot, `Seq: 61`. The payload is simply a little-endian HSL value. For example (from the screenshot): `65 01 e8 03 f4 01` That's:
* 0165 hex = 357 dec = 357° on the color circle = red (red is at 0°, green at 120°, blue at 240°)
* 03e8 hex = 1000 dec = 100% saturation (in tenth of a percent)
* 01f4 hex = 500 dec = 50% brightness (in tenth of a percent)

The packet before that (`Seq: 60` in the screenshot) always has a payload of `01`. 🤷 

The packet before that (`Seq: 59`) contains basically the same numbers, but written in big-endian order and instead of binary they're ascii spelling out the hex numbers. What a weird protocol. Those packets are continuously generated while dragging the slider in the app. If I just tap the slider at the brightness I want instead of sliding it, **those packets are not generated at all, so I believe we can simply omit them**.

AndreKR, 2023-02-04:
By the way, the lamp supports one more feature: You can set up a sequence of colors that the lamp then cycles through on its own. It's a bit of a gimmick, but it can also be used to switch between colors *without a transition*, which is useful (even necessary) to use the lamp as a status indicator or something like that. Normally when you set the color, the lamp slowly fades to the new color and I haven't found a way around that.

I don't know if Z2M could implement this, maybe just by allowing to send arbitrary bytes as scene configuration?

This "scene mode" is configured like this:

<details><summary>Screenshot</summary>

[image]
</details>

You set a cycle speed and choose between instant and "breathe" transition between colors. Then you add up to 8 colors. For each color you specify hue, saturation and brightness, just like usual. A "color" can also be set to use the white LEDs in which case you set color temperature and brightness.

To activate the sequence, a command `0xf1` packet is sent over Zigbee:

[image]

The meaning of the payload is as follows:

* Byte 1: always 06
* Byte 2:
  * if there is a single color, then always 00
  * if the color transition is set to instant, then always 00
  * if the color transition is set to "breathe", then this is a value with as many bits set as there are colors, i.e. 2 colors = 00000011 = 03, 3 colors = 00000111 = 07, etc.

After that there are 8 bytes per color:
* Byte 1: Color cycle speed between 28 hex (slowest) and 64 hex (fastest). Also, if the color transition is set to instant, the MSB is always set. All configured colors share the same value.
* Byte 2: The speed value again. Also, the MSB is essentially the 9th bit of the hue value (see below).
* Byte 3: The hue value in degrees on the color circle. For values between 255 and 360 that cannot be represented by a single byte, the MSB of byte 2 (see above) is set instead. 00 if this "color" is shown using the white LEDs.
* Byte 4: Extra bits for the other values: `ssbbwwtt` (see below)
* Byte 5: The saturation in tenths of a percent. The two extra bits required to reach 03E8 (1000 dec) are the ones I marked "bb" in Byte 4. 00 if this "color" is shown using the white LEDs.
* Byte 6: The brightness in tenths of a percent. The two extra bits required to reach 03E8 (1000 dec) are the ones I marked "bb" in Byte 4. 00 if this "color" is shown using the white LEDs.
* Byte 7: The brightness value in tenths of a percent for the white LEDs. The two extra bits required to reach 03E8 (1000 dec) are the ones I marked "ww" in Byte 4.
* Byte 8: The color temperature from 0 (warm white) to 1000 (cold white). The two extra bits required to reach 03E8 (1000 dec) are the ones I marked "tt" in Byte 4.

Koenkk, 2023-02-04:
@AndreKR awesome! I've added this command now `tuyaMoveToHueAndSaturationBrightness2` and adapted `tz.tuya_led_control` to use this. Can you check what works with the following converter?  https://gist.github.com/Koenkk/16d12d4d97a92fe3687118ee39693e49

Feel free to adapter where necessary! This converter requires the latest dev branch: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

AndreKR, 2023-02-05:
`constants` and `libColor` are undefined. I replaced `constants.colorMode[0]` and `constants.colorMode[2]` with 0 and 2 respectively, but I don't know where to get `libColor` from.

Koenkk, 2023-02-05:
I updated https://gist.github.com/Koenkk/16d12d4d97a92fe3687118ee39693e49

AndreKR, 2023-02-06:
* On/off works.
* In white mode, color temperature seems to work, but cool and warm are swapped and I have the impression the range isn't as wide as with the hub.
* In white mode, brightness works. The lowest setting is pretty bright, but I'd have to check with the hub again if it's the same.
* Whenever I tried to change to color mode, I had a number of different things happen, but before I try to further analyze this I think we should fix the following point.
* If I'm already in color mode (= previously clicked a color) then trying to change the color always sends `17 00 15 00 xx 00` in the command 0xe1 packet, i.e.:
  * The color is wrong, it's always 23° and 2.1% saturation, no matter which color I click.
  * The brightness value needs to be multiplied by 10.
* When in color mode, for some reason setting a brightness value of 32 and above enables the white LEDs. Setting the brightness lower than that disables the white LEDs again.

Also, at one point I got this error:
```
Publish 'set' 'color_temp' to '0xb4e3f9fffeba8034' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 0xb4e3f9fffeba8034/1 genLevelCtrl.moveToLevel({"level":1000,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 1000)'
```

I might be able to help with debugging, but I have some questions:

* There isn't any way to change the data that is sent in the packets without restarting Z2M every time I make a change, is there?
* Can I make log output in the converter?
* I'm guessing returning `{state: ...}` from `convertSet()` store this as the current state of the device and it will come back as `meta.state` on the next call? What's `globalStore` and how is it different from the state?

From how I understand [these lines](https://gist.github.com/Koenkk/16d12d4d97a92fe3687118ee39693e49#file-ext-js-L112-L114) the command 0xf0 packet should come first and then the command 0xe1, the same order as with the hub. (Does the `await` wait for an acknowledgement?) But that's not always the case, look...

[image]

Also, I don't know if it makes any difference but unlike with the hub, the "disable default response" flag isn't set on the packet even though the code suggests that it should be set.

RunnicFusion, 2023-02-07:
> * mperature seems to work, but cool and warm are swapped and I have the impression the range isn't as wide as with the hub.

hi, I would like to help during testing. How can I run/use your provided fix?

Koenkk, 2023-02-07:
> There isn't any way to change the data that is sent in the packets without restarting Z2M every time I make a change, is there?

No there isn't indeed

> Can I make log output in the converter?

Yes, you can use a `console.log('LOG HERE')` which will be logged to the STDOUT

> I'm guessing returning {state: ...} from convertSet() stores this as the current state of the device and it will come back as meta.state on the next call? What's globalStore and how is it different from the state?

True, `globalStore` are for things you do NOT wan't to put in the state (usually timers, helper things)

AndreKR, 2023-02-07:
> hi, I would like to help during testing. How can I run/use your provided fix?

@RunnicFusion As described in https://github.com/Koenkk/zigbee2mqtt/issues/13421#issuecomment-1407640757

AndreKR, 2023-02-10:
So I did some debugging. Unfortunately I don't feel confident to provide a working implementation because there are some things I don't really understand about the converter. For example the code of `convertSet()` seems to expect `meta.message` to contain *multiple* keys but there's also a `key` parameter that gets treated special? Like what's the significance of those cases that have their own `if` blocks. I also don't understand the whole `syncColorState` and `readAfterWriteTime` stuff in the last line.

Anyway, I think I have identified a couple of issues that currently prevent it from working properly:

* Clicking a color in the frontend "sends" a message with `{ color: { hex: "#..." } }` but I can find no code to handle that. Home Assistant sends `{ color: { h: ..., s: ... } }` though.
* I'm a bit unclear about the ranges of the numbers. Do we store the maximum brightness and saturation in the state as 1000 (what the device uses) or as 100? Or as 254? Unless we actually *want* to store it as 254 I think almost all of those `mapNumerRange()` calls need to be removed.
* When starting with an empty state (not sure if that can happen in real life, I reset the status manually), accessing `meta.state.color.h` in line 71 will throw an error.

Koenkk, 2023-02-10:
@AndreKR 

- Since this device handles multiple keys (brightness/colortemp/color) you can ignore `key` and should only act on `meta.message`
- `readAfterWriteTime` ignore this (not used)
- `syncColorState` this has no impact on the device, what it does is update e.g. the color temp to closely match the color when you set the color

> Clicking a color in the frontend "sends" a message with { color: { hex: "#..." } } but I can find no code to handle that. Home Assistant sends { color: { h: ..., s: ... } } though.

The color conversion can be done through the already existing `libColor`, you can find an example [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/300d75007b6e2d2bd559bf1767eb34fd3956573d/converters/toZigbee.js#L1076). To use it add `const libColor = require('zigbee-herdsman-converters/lib/color');` to the top of your converter. When hex is send, `newColor.isRGB()` is true and you can convert it to HSV by calling `newColor.rgb.gammaCorrected().toXY().toHSV()`

> I'm a bit unclear about the ranges of the numbers. Do we store the maximum brightness and saturation in the state as 1000 (what the device uses) or as 100? Or as 254? Unless we actually want to store it as 254 I think almost all of those mapNumerRange() calls need to be removed.

This mapping is needed since the Z2M scale is 0..254 and the device scale is 0..1000, when you would e.g. send 254 to the device it will turn on at roughly 25% of the max brightness (while this should be 100%)

> When starting with an empty state (not sure if that can happen in real life, I reset the status manually), accessing meta.state.color.h in line 71 will throw an error.

In this case a default should be used, e.g. 0

AndreKR, 2023-02-22:
I have made quite a few changes, but now everything seems to work. @Koenkk Can you give it a review?
https://gist.github.com/AndreKR/e2fe383cab53ea7438693ce854514fab

Couple of notes:
* I decided to ignore the brightness of the color and instead use the brightness of the lamp even in color mode. That's also how it works with my Tradfri bulbs.
* When I set the Tradfri bulbs to black, they turn off. The LIDL lamp with my converter doesn't do that. I don't know which is the expected behavior.
* I noticed that when I use the color picker in the UI and I drag the mouse cursor across the color gradient, a lot of messages are generated. It seems they are queued and the lamp needs a minute or so to catch up. I don't know if that is expected or if it is a result of how I wrote the converter.
* The `'tuyaRgbMode'` command is still sent with the "Disable Default Response" set to *false* despite the `{disableDefaultResponse: true}` in the code. I don't know if that matters.

Koenkk, 2023-02-22:
I reviewed your code, it's very clean! 😄 

Do you want to make a PR? (otherwise I can also integrate it)

> When I set the Tradfri bulbs to black, they turn off. The LIDL lamp with my converter doesn't do that. I don't know which is the expected behavior.

The Tradfri behaviour is a bit unexpected, so your converter seems fine

> I noticed that when I use the color picker in the UI and I drag the mouse cursor across the color gradient, a lot of messages are generated. It seems they are queued and the lamp needs a minute or so to catch up. I don't know if that is expected or if it is a result of how I wrote the converter.

This is an issue of the UI (it should do some debounce). @nurikk could you check that?

> The 'tuyaRgbMode' command is still sent with the "Disable Default Response" set to false despite the {disableDefaultResponse: true} in the code. I don't know if that matters.

Do you see the device responding with a default response message?

AndreKR, 2023-02-23:
> Do you want to make a PR?

https://github.com/Koenkk/zigbee-herdsman-converters/pull/5514

The final `lidl.js` file with my changes is not tested because I don't know if that's possible with a Docker installation of Z2M.

I only really know about the `convertSet()` function, I copied the `options`, `fromZigbee` and `exposes` settings from your file, so please check them (especially the `color_sync()` thing is now wrong?).

I "fixed" the transition at least for brightness and color temperature in white mode like [this](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5514/files#diff-9f95626d459809420576fa9d9e45a723238c91c0b9c2d86a9ff3573d3559e3abR396-R401). Am I missing something or is the device just weird?

Also I have only tested the settings on the "Exposes" tab and in Home Assistant. In particular I did not test any reading/writing using the "Dev console" tab and I also don't know what `fromZigbee` and `convertGet()` do.

> Do you see the device responding with a default response message?

I think so:

[image]

KentuckyMC, 2023-02-23:
Just want to say, thanks for your effort to fix this issue! Can't wait till it's in z2m :) Have a nice day!

Koenkk, 2023-02-23:
I've merged the changes from @AndreKR.

Changes will be available in the dev branch in a few hours from now (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and in the 1 February release.

@AndreKR 

- Could you do a final check with the latest dev to confirm everything  works? Don't forget to remove your external converter otherwise that will be used instead.
- Could you check if the color sync works by changing  `return {state: newState};` in your converter to `return {state: libColor.syncColorState(newState, meta.state, entity, meta.options, meta.logger)};`. To test if it works, change e.g. the color temp and observe that the HS color also changes (and the other way around). 

> I "fixed" the transition at least for brightness and color temperature in white mode l
- The link links to an unrelated location, can you give me the correct link?

> Also I have only tested the settings on the "Exposes" tab and in Home Assistant. In particular I did not test any reading/writing using the "Dev console" tab and I also don't know what fromZigbee and convertGet() do.

- These are needed when another device can also change the led state or when the device has button on it to turn it on/off. I've looked at the code and this should work.

> Do you see the device responding with a default response message?

As controlling works now, we can keep it like this. The default response gives Zigbee2MQTT guarantees that the message arrived at the controller.

AndreKR, 2023-02-25:
@Koenkk 

> Could you do a final check with the latest dev to confirm everything works? Don't forget to remove your external converter otherwise that will be used instead.

There is an issue. Now that `color_mode` is a number (by the way, the string was nicer for the UI) my haphazard use of `||` as null coalescing operator no longer works because 0 is falsy. [Here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5523)'s a PR with a fix. Unfortunately eslint will not accept `??` because `ecmaVersion` in `.eslintrc.json` is set to `2018` and `2020` would be needed for `??`. I have to think about a better solution.

There is another issue, and I don't know why this didn't happen in my previous tests - I'm assuming it's a quirk of the light: When setting the color temperature while in color mode, the first time it switches to white mode but at the _previous_ color temperature. I have to set the color temperature a second time for the light to actually accept the new color temperature. To fix this we *could* send the `moveToColorTemp` command twice, but we can also send the `tuyaRgbMode` command with `enable: 0` just like the hub does. The PR contains a fix for that as well.

Another issue is that the brightness isn't synced between color mode and white mode. (The light keeps track of two separate brightness values while Z2M only has one.) I won't fix that today, so I'll send another PR later.

> Could you check if the color sync works [...]
> change e.g. the color temp and observe that the HS color also changes (and the other way around).

Oooh, *that's* what color sync does.
But no, it doesn't work:

```
console.log(newState);
console.log(libColor.syncColorState(newState, meta.state, entity, meta.options, meta.logger));
```

```
{ color_temp: 500 }
{ color_mode: 2 }
```

> As controlling works now, we can keep it like this. The default response gives Zigbee2MQTT guarantees that the message arrived at the controller.

Ok, I was just worried because it indicates that `entity.command('...', '...', {...}, {}, {disableDefaultResponse: true})` doesn't work as expected.

Koenkk, 2023-02-25:
@AndreKR the `??` operator is fine to use, min supported z2m version is Node 14 which supports it.

rouvenst, 2023-02-26:
Hi all,

i just integrated two Light bars (14149506L) via Home Assistant and z2m edge. As I've read the last comment from @AndreKR there's an open issue 
> Another issue is that the brightness isn't synced between color mode and white mode. (The light keeps track of two separate brightness values while Z2M only has one.) I won't fix that today, so I'll send another PR later.

Maybe I've come along with another issue:
1. Select color in HA
2. In HA, Brightness is shown "as 0" + color of light icon is not updated

That's strange because z2m logs always has a brightness and color set:
```
Zigbee2MQTT:info  2023-02-26 11:21:21: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeeaccef', payload '{"brightness":221,"color":{"h":60,"s":71.42857142857142},"color_mode":0,"color_temp":305,"linkquality":255,"state":"ON"}'
Zigbee2MQTT:info  2023-02-26 11:23:00: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeeaccef', payload '{"brightness":221,"color":{"h":315,"s":100},"color_mode":0,"color_temp":305,"linkquality":255,"state":"ON"}'
Zigbee2MQTT:info  2023-02-26 11:23:02: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeeaccef', payload '{"brightness":97,"color":{"h":315,"s":100},"color_mode":0,"color_temp":305,"linkquality":255,"state":"ON"}'
Zigbee2MQTT:info  2023-02-26 11:23:03: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeeaccef', payload '{"brightness":97,"color":{"h":0,"s":85.71428571428571},"color_mode":0,"color_temp":305,"linkquality":255,"state":"ON"}'
```

That's how it looks after a page refresh 
[image]

Is this related?

AndreKR, 2023-02-27:
@Koenkk The "color of light icon is not updated" issue that @rouvenst reports is because with the new color mode constants Z2M publishes `"color_mode":0` but Home Assistant needs `"color_mode":"hs"`.

@rouvenst I can't reproduce the issue with the brightness being shown as 0. When exactly does that happen? When you try to set a brightness is stays at 0? Or when you select a color is jumps back to 0?

rouvenst, 2023-02-28:
@AndreKR It happens when I select a color, but not when I select a color temperature.

What wonders me that the brightness attribute is missing on my light entity in HA, as seen on the screenshot.
Could it be some misconfiguration?
[image]

Just for completeness what I'm running:
```
Home Assistant 2023.2.5
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230202.0 - latest

Zigbee2MQTT Edge - Current version: edge (just updated)
```

Koenkk, 2023-02-28:
@AndreKR fixed the `color_mode`!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

AndreKR, 2023-02-28:
@Koenkk Ok, that helped, but now I'm having another issue. The `color` attribute is somehow frozen:

The state currently is;
```
{
    "color": {
        "h": 122,
        "hue": 122,
        "s": 100,
        "saturation": 100
    },
    "color_mode": "hs",
    "color_temp": 153,
    "linkquality": 36,
    "state": "ON"
}
```

In my converter I have for testing:
```
            newState.color.foo = 'bar';
            console.log(newState);
            return {state: newState};
```

The log output:
```
{ color: { h: 122.53888502559742, s: 99.99994617226982, foo: 'bar' } }
{ color: { h: 240, s: 100, foo: 'bar' } }
```
The new state (which also gets published via MQTT):
```
{
    "color": {
        "foo": "bar",
        "h": 122,
        "hue": 122,
        "s": 100,
        "saturation": 100
    },
    "color_mode": "hs",
    "color_temp": 153,
    "linkquality": 36,
    "state": "ON"
}
```

So whatever I return for `color.h` and `color.s`, those values don't change on the "State" tab. As you can see, I can add and update other fields like `color.foo`.

I was able to (temporarily?) fix it by stopping Z2M, removing the device from `state.json` and restarting Z2M. @rouvenst That workaround will probably work for you as well.

rouvenst, 2023-03-01:
Just updated to the newest Z2M Edge version and my problems are fixed. It was not needed to edit `state.json`.
Thank you very much @AndreKR and @Koenkk!

Koenkk, 2023-03-01:
@AndreKR 

- Does this also happen without the external converter?
- Will e.g. `newState.color.h= 900;` get published?

rotilho, 2023-03-01:
Thanks for all the work guys! Very appreciated :D

AndreKR, 2023-03-01:
@Koenkk

Yes, in fact I (and possibly you as well) can reproduce the issue without the external converter, just by adding this to `state.json` (while Z2M is stopped):
```
    "[...]": {
        "color": {
            "h": 122,
            "hue": 122,
            "s": 100,
            "saturation": 100
        },
        "color_mode": "hs",
        "color_temp": 153,
        "linkquality": 36,
        "state": "OFF"
    }
```

Koenkk, 2023-03-02:
@AndreKR what about:

- Will e.g. newState.color.h= 900; get published?

AndreKR, 2023-03-02:
@Koenkk I'm not sure I understand what part exactly you're asking about.

If I add `newState.color.h=900;` to `convertSet()` and then I click a color in the UI the function will return 
```
{ state: { color: { h: 900, s: 99.99994617226982 } } }
```
but the state is not updated (as shown on the "State" tab) and the `900` will also not be published via MQTT but instead the current state is published:
```
Zigbee2MQTT:info  2023-03-02 19:49:02: MQTT publish: topic 'zigbee2mqtt/Light Bar', payload '{"color":{"h":122,"hue":122,"s":100,"saturation":100},"color_mode":"hs","color_temp":153,"linkquality":32,"state":"ON"}'
```

Koenkk, 2023-03-04:
In the external converter, can you replace `return {state: newState};` with:

```js
meta.logger.debug(`GOING TO PUBLISH: ${JSON.stringify(newState)}`);
return {state: newState};
```

Is the expected new state logged?

AndreKR, 2023-03-04:
Yes, the expected new state is logged, but then the old state is published.
```
Zigbee2MQTT:debug 2023-03-04 17:15:38: GOING TO PUBLISH: {"color":{"h":359.4519599116082,"s":100}}
Zigbee2MQTT:info  2023-03-04 17:15:38: MQTT publish: topic 'zigbee2mqtt/Light Bar', payload '{"color":{"h":122,"hue":122,"s":100,"saturation":100},"color_mode":"hs","color_temp":153,"linkquality":21,"state":"ON"}'
```

[7 more replies]

closed 2023-04-17 by Koenkk