Koenkk/zigbee2mqtt #13425: [New device support]: Gledopto GL-S-006P GU10 lamp

lysanev, 2022-08-06:
### Link

https://www.gledopto.com/h-col-402.html

### Database entry

{"id":39,"type":"Router","ieeeAddr":"0xa4c1386231a77113","nwkAddr":2034,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-S-006P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-S-006P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20220421","swBuildId":"10551203"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20220421","swBuildId":"10551203","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659720867787,"defaultSendRequestWhen":"immediate"}

### Comments

i tried to follow support new device, but got stuck with OtaUpgrade

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GL-S-006P'],
    model: 'GL-S-006P',
    vendor: 'Gledopto',
    description: 'Gledopto GL-S-006P GU10 lamp',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

### Supported color modes

color tempertarure, color (RGB + CCT)

Koenkk, 2022-08-06:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

> i tried to follow support new device, but got stuck with OtaUpgrade

This is not required, we can only do this if Gledopto provides us an OTA file.

lysanev, 2022-08-06:
{"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}

Koenkk, 2022-08-07:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-07 by Koenkk