Koenkk/zigbee2mqtt #13429: Xiaomi WXKG02LM Aqara double key wireless wall switch - Previously requested problem correction

husnainali09, 2022-08-06:
### What happened?

Single press work but double press and press hold do not work

Previously discussed in https://github.com/Koenkk/zigbee2mqtt/issues/5425

Still an issue in main branch 

### What did you expect to happen?

Problem corrected as known fix

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

Conbee ii

### Debug log

Previously known issue but fix not in main branch

husnainali09, 2022-08-09:
Anyone have a suggestion ?

Koenkk, 2022-08-09:
How is it recognized, as `WXKG02LM_rev1` or `WXKG02LM_rev2`? 
- rev2 supports single, hold and double (https://www.zigbee2mqtt.io/devices/WXKG02LM_rev2.html)
- rev1 only supports single (https://www.zigbee2mqtt.io/devices/WXKG02LM_rev1.html#action-enum)

husnainali09, 2022-08-09:
Fudge, it's rev 1, WXKG02LM_rev1.. my apologies.

Koenkk, 2022-08-09:
rev1 doesn't support it, so not seeing that action is expected behaviour.

closed 2022-08-09 by Koenkk

SergioRius, 2022-09-28:
Hi @Koenkk I have the same problem. I have 6 of those switches. Underneath the same exact model, revision and build date. But half of them are recognized as rev2 and the other half as rev1.
Shouldn't be recognized all the same? Is there a way to make z2m threat all of them as rev2?

Koenkk, 2022-09-28:
@SergioRius the firmware on them is definitely different since they use a different Zigbee model ID.

You can try to hack your way around by stopping z2m, opening the data/database.db and changing the modelID of those devices from  `lumi.sensor_86sw2` OR `lumi.sensor_86sw2.es1` to `lumi.remote.b286acn01` which will get them recognized as `WXKG02LM_rev2` but I expect it won't give you the additional actions.

SergioRius, 2022-09-28:
Thanks @Koenkk 
I was hoping something more like a firmware upgrade. I suppose i should also change the endpoints descriptors.
I'm asking because i had one of those devices installed that worked with double and hold actions that suddenly stopped working.

Koenkk, 2022-09-28:
These devices don't have firmware upgrades. You can try the db hack I mentioned earlier.