Koenkk/zigbee2mqtt #13433: [New device support]: TS0601 Based Blind Motor "OZ ZIGBEE SMART BLIND MOTOR"

skandiah94, 2022-08-06:
### Link

https://www.ozsmartthings.com.au/collections/zigbee/products/zigbee-smart-blind-motor

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0x94deb8fffeee9a25","nwkAddr":39398,"manufId":4098,"manufName":"_TZE200_pw7mji0l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���*\u0013%*\u0012&*\u0012","65506":31,"65508":0,"appVersion":72,"stackVersion":0,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0bdff","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1659792384544,"defaultSendRequestWhen":"immediate"}

### Comments

I tried using some of the existing tuya blind motor converters and their configs but got stuck trying to debug the datapoints.. :( 

The manufacturer does provide a device handler text file but I have no idea of what I'm looking at:

https://drive.google.com/file/d/1J5Gj608_IUEA9dD6U2ZeVAmMGdYN66S0/view 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_pw7mji0l'
        },
    ],
    model: 'ZM85EL-1Z', // Vendor model number, look on the device for a model number
    vendor: 'Oz Zigbee Blind', // Vendor of the device (only used for documentation and startup logging)
    description: 'Zigbee Blind Motor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        exposes.composite('options', 'options')
            .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(255)
                .withDescription('Motor speed'))],
};

module.exports = definition;
```

skandiah94, 2022-08-12:
I've added the contents of tuya,dump.txt below. The unit itself has 3 buttons. Happy to press more buttons on the remote to add those if it helps

```
1659792336724 0x94deb8fffeee9a25 100 00 07 04 00
1659792339070 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792339172 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
1659792344705 0x94deb8fffeee9a25 100 00 07 04 00
1659792345685 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792345815 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
1659792346850 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792346971 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
1659792347889 0x94deb8fffeee9a25 100 00 07 04 00
1659792348537 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792348657 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
1659792348951 0x94deb8fffeee9a25 100 00 07 04 00
1659792349577 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792349696 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
1659792383101 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792383225 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
1659792384415 0x94deb8fffeee9a25 100 00 03 02 00 00 00 00
1659792384546 0x94deb8fffeee9a25 100 00 0d 02 00 00 00 64
```

jscooksey, 2022-08-28:
I have the same blind/cover controller.  It seems Zemismart sell the same thing as well, so maybe some similarities that can help.

skandiah94, 2022-08-28:
Oh cool! Sorry I actually reached out to ozsmartthings directly and they said their Dev will look at integrating support for it :)

I'll leave this issue open until it's resolve so it's tracked

skandiah94, 2022-09-09:
Reaching out to the dev his response was "why don't you integrate directly with ZHA?"

I've poked again to see if I can get a response.

Interestingly in the ZHA integration the blind comes up as zemismart. maybe it's possible to modify the zigpy implementation for z2m?

Koenkk, 2022-09-10:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_pw7mji0l'}],
    model: 'ZM25RX-08/30',
    vendor: 'Zemismart',
    description: 'Tubular motor',
    fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options, tz.tuya_data_point_test],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works

skandiah94, 2022-09-10:
Core functionality works as expected (open, close, stop, position 0-100) there is one data type that comes through as unknown though, however it doesn't break functionality.

`zigbee2mqtt    | Zigbee2MQTT:info  2022-09-10 22:18:19: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"linkquality":109,"position":100,"running":false,"state":"OPEN"}'
zigbee2mqtt    | Zigbee2MQTT:warn  2022-09-10 22:18:19: TuYa_cover_control: Unhandled DP #13 for _TZE200_pw7mji0l:
zigbee2mqtt    |                 {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
zigbee2mqtt    | Zigbee2MQTT:info  2022-09-10 22:18:25: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"linkquality":109,"position":80,"running":false,"state":"OPEN"}'
zigbee2mqtt    | Zigbee2MQTT:warn  2022-09-10 22:18:25: TuYa_cover_control: Unhandled DP #13 for _TZE200_pw7mji0l:
zigbee2mqtt    |                 {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,95]}}
zigbee2mqtt    | Zigbee2MQTT:info  2022-09-10 22:18:42: MQTT publish: topic 'zigbee2mqtt/Study Motion Sensor', payload '{"battery":87,"illuminance_above_threshold":false,"linkquality":109,"occupancy":true,"requested_brightness_level":254,"requested_brightness_percent":100,"update":{"state":"idle"},"update_available":null}'
zigbee2mqtt    | Zigbee2MQTT:info  2022-09-10 22:18:46: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"linkquality":112,"position":80,"running":false,"state":"OPEN"}'
zigbee2mqtt    | Zigbee2MQTT:warn  2022-09-10 22:18:46: TuYa_cover_control: Unhandled DP #13 for _TZE200_pw7mji0l:
zigbee2mqtt    |                 {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
zigbee2mqtt    | Zigbee2MQTT:info  2022-09-10 22:18:52: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"linkquality":112,"position":100,"running":false,"state":"OPEN"}'
zigbee2mqtt    | Zigbee2MQTT:warn  2022-09-10 22:18:52: TuYa_cover_control: Unhandled DP #13 for _TZE200_pw7mji0l:
zigbee2mqtt    |                 {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,90]}}`

Koenkk, 2022-09-11:
Added, to find what this datapoint means see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

closed 2022-10-19 by github-actions[bot]