Koenkk/zigbee2mqtt #13438: [Feature request]: Support effect / effect_list in groups exposed to Home Assistant

deviantintegral, 2022-08-06:
### Is your feature request related to a problem? Please describe

I noticed that I was unable to trigger effects like "okay" and "blink" on my Philips Hue bulbs that are in a group. Targeting individual bulbs worked fine, and sending an effect to a group over MQTT worked as well.

The hint I was missing is that Home Assistant wasn't showing effects in the entity card to control the group:

[image]

When I manually added `effect` and `effect_list` to the device configuration topic, it showed up in the UI and effects worked as expected.

[image]

### Describe the solution you'd like

I think groups should be exposed to Home Assistant with the _union_ of the capabilities of their underlying bulbs. I believe this is the behaviour for attributes like colour and temperature.

### Describe alternatives you've considered

It looks like the topic has already reverted and I don't see a way to ensure it stays configured in the long term.

### Additional context

Here's a working topic:

```
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state"
    },
    {
      "topic": "zigbee2mqtt/Kid's Bathroom/availability"
    }
  ],
  "availability_mode": "all",
  "brightness": true,
  "brightness_scale": 254,
  "color_mode": true,
  "command_topic": "zigbee2mqtt/Kid's Bathroom/set",
  "device": {
    "identifiers": [
      "zigbee2mqtt_1221051039810110150109113116116_5"
    ],
    "name": "Kid's Bathroom",
    "sw_version": "Zigbee2MQTT 1.27.0"
  },
  "effect": true,
  "effect_list": [
    "okay",
    "breathe",
    "blink",
    "channel_change",
    "finish_effect",
    "stop_effect"
  ],
  "max_mireds": 500,
  "min_mireds": 150,
  "name": "Kid's Bathroom",
  "schema": "json",
  "state_topic": "zigbee2mqtt/Kid's Bathroom",
  "supported_color_modes": [
    "xy",
    "color_temp"
  ],
  "unique_id": "5_light_zigbee2mqtt"
}
```

deviantintegral, 2022-09-06:
not stale

deviantintegral, 2022-10-07:
not stale!

Koenkk, 2022-10-08:
not stale but closed! implemented 😄 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-08 by Koenkk

deviantintegral, 2022-10-08:
This is working well for me. I even have a group of Hue and non-Hue bulbs that don't support effects, and I can send an effect to the group and if the bulb supports it, it executes. Thank you!